lvgl/lvgl #1703: [suggestion] add lv_obj_is_instanceof(lv_obj* obj, const char* typename)

incity, 2020-08-04:
[image]
the code here can be optimized.

kisvegabor, 2020-08-04:
I can't see what's the point here. Please describe the issue or your idea in more detail.

embeddedt, 2020-08-04:
All three of them loop through the `lv_obj_type_t` struct looking for a particular string. That loop could potentially be optimized into a common function.

incity, 2020-08-04:
@kisvegabor the code is used to check if an object is an instance of a type. So you can offer a common function named instanceof likes the java keyword instanceof.

kisvegabor, 2020-08-04:
Ah, got it. Sounds reasonable. 
I also like the function name you suggest.

@incity 
Are you interested in adding it?

incity, 2020-08-05:
@kisvegabor yes I'd like to try

incity, 2020-08-05:
@kisvegabor i found a function lv_debug_check_obj_type(...) in lv_obj.c

kisvegabor, 2020-08-06:
Really, it does the same. 

I suggest implementing `lv_obj_is_instanceof` and using it in `lv_debug_check_obj_type`.

incity, 2020-08-07:
how about move lv_debug_check_obj_type and lv_debug_check_obj_valid from lv_obj.c to lv_debug.c

kisvegabor, 2020-08-10:
`lv_debug.c` is in the `lv_misc` folder and I like to keep these parts free from objects, styles, widgets, etc. These are just helper stuff, independent of the core lvgl.

closed 2020-09-07 by stale[bot]