lvgl/lvgl #1704: Text rendering (in lv_obj_set_style_local_value_str?) is inconsistent

upbeat27, 2020-08-04:
**Describe the bug**

The text rendering can look different, when everything is "the same". Some text looks a little bolder, less _clean_.
This seems to only apply to text rendered from ```lv_obj_set_style_local_value_str```, as using a label instead seems to fix the inconsistency.
It only happens with some text. I don't know what the link is.

**To Reproduce**

Use ```lv_demo_widgets.c``` as is. Release version 7.2.0.
For the sake of the screenshots below, I renamed some of the text so they would be the same. This makes it very clear that they look different even though they should not.

**Expected behavior**

The text should look the same.

**Screenshots**

[image]

[image]

**Additional context**

None.

embeddedt, 2020-08-04:
I've noticed this before but thought it was just inaccuracies of integer math or something like that. The darkness suggests that the text might be getting drawn twice?

upbeat27, 2020-08-04:
Hmm... even rendering twice should draw the exact same pixels the second time around, unless the second pass is at different subpixel coordinates.
I would start with how text from ```lv_obj_set_style_local_value_str``` is different than text from a label. I have not seen a label render strangely yet.

kisvegabor, 2020-08-04:
Hi,

Thanks for the report. It was really drawn twice. 

I fixed it in the commit above.

closed 2020-08-04 by kisvegabor

embeddedt, 2020-08-10:
The Dark switch label still seems to be darker than the tab labels with the latest `dev` version.

kisvegabor, 2020-08-11:
Good catch!
Fixed here: 2068ab32

commit b4955f0b (2020-08-04): Fix drawing value string twice
Fixes #1704