lvgl/lvgl #1705: Dynamic font loading

fabiowguerra, 2020-08-05:
Hello,

Thanks for this great project!

At [Muxi](http://www.muxi.com/) we are considering using LVGL on one of our products and one requirement is to be able to load new fonts without changing to application binary.

I created a fork on https://github.com/fabiowguerra/lvgl/tree/feat/dynamic-font-loading. With this fork I can load fonts generated by https://github.com/lvgl/lv_font_conv.

I followed the spec on https://github.com/lvgl/lv_font_conv/blob/master/doc/font_spec.md to load the font file and generated a `lv_font_t` object with the same structure as `.c` file the generator creates.

But I was not able to test some possible formats described:
- cmaps table format type 1
- kern format type 0
- advance_width_format, I don't know how to interpret the `advance_width_format`. All the fonts I generated has the same format, and for now I am ignoring this parameter.

Can you help me with this understandings? Or maybe on how to generate fonts to test this formats?

Do you think this fork worth a PR, even not supporting all the possible formats?

In advance, thank you for your assistance. We look forward to hearing from you.

puzrin, 2020-08-05:
>  * cmaps table format type 1

List of formats was done to guaranty possibility to cover all theoretic cases. But in real world, something can never happen (but without guarantee). I don't remember details, but some format may be impossible to generate from real data.

>  * kern format type 0

This can save ~ 100 bytes for small sets (english letters only). In practice - tables are preferred, because much faster. And i would recommend force tables (type 1) via convertor options.

To reproduce - create font with range 0x20-07F, type 0 may be used by default with high probability.

>  * advance_width_format, I don't know how to interpret the `advance_width_format`. All the fonts I generated has the same format, and for now I am ignoring this parameter.

https://github.com/lvgl/lv_font_conv/blob/master/lib/font/font.js#L44 fonts without kerning (monospace) don't need fractional part in width. You can drop kerning via convertor option to reproduce. Or try to convert monospace font (it should not have kerning, but check in editor).

kisvegabor, 2020-08-06:
First of all, thank you for sharing this feature. It was already asked several times by the users. :clap: :clap: 

I've looked into your implementation and it looks great! Do you plan to send a pull request?

fabiowguerra, 2020-08-06:
I'll try to test a little bit more, then I will open the pull request.

fabiowguerra, 2020-08-07:
@puzrin thanks for your information. I tested with monospaced fonts and I had some problems. The advanced_width_bits is 0, I expected to use the default adv_w, and not writing to the adv_w field, but it is not happening, and I don't know how many bits to read. Maybe I am miss-understanding something.

@kisvegabor it's ok to open the Pull Request with some limitations? Like not working with monospace fonts? And not supporting some internal formats?

I created a simple test that compares the generated font with the `.c` the generator creates. But I don't know if it is a good idea to commit the binary file, since the fonts are already commited on `scripts/built_in_font` and could be generated (but we can't assume everyone will have all dependencies needed to run the generator). What do you think about that?

puzrin, 2020-08-07:
@fabiowguerra sorry, i don't understand. I don't keep all details in head. If you found a bug - please provide a reproducible sample. Or point to concrete line of convertor source.

fabiowguerra, 2020-08-07:
https://github.com/lvgl/lv_font_conv/blob/master/lib/font/table_head.js : 84
```
if (f.monospaced) buf.writeUInt8(0, O_ADVANCE_WIDTH_BITS);
else buf.writeUInt8(f.advanceWidthBits, O_ADVANCE_WIDTH_BITS);
```

https://github.com/lvgl/lv_font_conv/blob/master/lib/font/table_glyf.js : 40
```
// Store Width
if (f.advanceWidthBits > 0) {
  let w = f.widthToInt(glyph.advanceWidth);
  bs.writeBits(w, f.advanceWidthBits);
}
```
 When font is monospaced, 0 is written on the header, but when writting the glyph on the binary file, it does not test if the font is monospace. Since the number of bits is not stored on the header, I cannot

fabiowguerra, 2020-08-07:
On https://github.com/lvgl/lv_font_conv/blob/master/lib/font/table_head.js : 84
```
if (f.monospaced) buf.writeUInt8(0, O_ADVANCE_WIDTH_BITS);
else buf.writeUInt8(f.advanceWidthBits, O_ADVANCE_WIDTH_BITS);
```

On https://github.com/lvgl/lv_font_conv/blob/master/lib/font/table_glyf.js : 40
```
// Store Width
if (f.advanceWidthBits > 0) {
  let w = f.widthToInt(glyph.advanceWidth);
  bs.writeBits(w, f.advanceWidthBits);
}
```

If the font is monospaced, `0` is written on `O_ADVANCE_WIDTH_BITS`, but when writing the glyph, it uses `f.advanceWidthBits`, without a monospace test.
When parsing I don't know how many bits to read.

puzrin, 2020-08-08:
@fabiowguerra You are right. I guess, nobody tested well binary output + monospace fonts are very rare (used only with code editors).

https://github.com/lvgl/lv_font_conv/tree/dev - try fix in dev branch. Kick me when no more fixes needed, and i will release.

---

About your font loader, let me explain situation in general.

- Font convertor is designed to be universal (not pinned to lvgl) and effective (fast data extract from bin format).
- At current moment main use of convertor is lvgl. Core devs insisted on text format, and i had to give up and add secondary writer. But text format is for lvgl only and a bit limited.

So, IMO, if you need binary format support, ideal solution would be to create alternate "font engine" instead of convert binary data to objects. With term "font engine" i mean something, providing API like this https://github.com/lvgl/lvgl/blob/master/src/lv_font/lv_font.c

From one hand, your code allows get result fast for your concrete need (dynamic font load). But i think, lvgl also need work with binary fonts directly, and that's a bit different thing. You don't need that, but you asked opinion about code merge an here is my one.

@kisvegabor how difficult will be to provide explicit abstraction layer for font engine? I mean, font loader should create some generic struct (im memory) + custom extensions, on init. Then lvgl should select appropriate font engine and methods to access data.

puzrin, 2020-08-08:
Ah, found example in freetype demo https://github.com/lvgl/lv_lib_freetype. The same approach can be used for binary loader.

kisvegabor, 2020-08-10:
@Fabien-Chouteau 
>  it's ok to open the Pull Request with some limitations? Like not working with monospace fonts? And not supporting some internal formats?

If the current implementation works in the typical cases then there is no problem with having a limitation for special, rarely used cases. Just need to add these limitations to the docs. 

I think it's better to add the test C and bin files explicitly to make running tests easier. These are small files anyway.

@puzrin 
>  how difficult will be to provide explicit abstraction layer for pluggable "font engine"?
> Ah, found example in freetype demo https://github.com/lvgl/lv_lib_freetype. 

Yes, that's it. The documentation is here: https://docs.lvgl.io/latest/en/html/overview/font.html#add-a-new-font-engine

The intermediate step in @Fabien-Chouteau's implementations (convert bin to C array)  is not strictly required as most of the data can be read directly from the bin file too. But I see that it's more simple. It might be worth considering to add a switch whether to load the bitmaps data with malloc or use them from the bin file. It requires a custom `lv_font_get_bitmap_fmt_txt` function though.

fabiowguerra, 2020-08-12:
Hello @kisvegabor , appears you select the wrong name on the auto-complete for my user :)

I created the Pull Request at https://github.com/lvgl/lvgl/pull/1717. I implemented the missing formats I mentioned on my last post.

The monospace fonts will work on the next version of the converter.

kisvegabor, 2020-08-12:
> Hello @kisvegabor , appears you select the wrong name on the auto-complete for my user :)

Ups, updates :) Somehow github offers that user for "f". 

> I created the Pull Request at #1717. I implemented the missing formats I mentioned on my last post.

I'll review it today. Thank you very much!

kisvegabor, 2020-09-01:
The PR is merged so I close this issue. Thank you very much for this great feature!

closed 2020-09-01 by kisvegabor