lvgl/lvgl #1708: Support asynchronous GPU operation

rabbitsaviola, 2020-08-06:
Request to add asynchronous GPU suppot as described by this [thread](https://forum.lvgl.io/t/add-asynchronous-gpu-interface/2838)

kisvegabor, 2020-08-08:
Thanks for opening this issue.

I didn't consider this when used `_lv_refr_get_disp_refreshing` during drawing. So it might work, but it's pure luck. :)

I'd prefer adding a dedicated function for this. E.g. `lv_gpu_wait(disp_refr->driver)`.

What do you think?

rabbitsaviola, 2020-08-10:
@kisvegabor it's ok to add `lv_gpu_wait`,  but where will this function be called?

kisvegabor, 2020-08-11:
> it's ok to add lv_gpu_wait, but where will this function be called?

That's the difficult part. :slightly_smiling_face: 

As it's only about rendering we need to ensure that when all GPU operations are finished while the next drawing occurs.

My initial guess it to add in after the initial `return`s in `_lv_blend_fill` and `_lv_blend_map`. If you agree can you test it?

rabbitsaviola, 2020-08-11:
Is it necessary to call lv_gpu_wait also in lv_refr_vdb_flush before executing `flush_cb` to make sure gpu is done?

kisvegabor, 2020-08-12:
That's correct! It's required there too.

rabbitsaviola, 2020-08-13:
> That's correct! It's required there too.

ok， I'll try it.

rabbitsaviola, 2020-08-19:
@kisvegabor I created a PR  #1736, could you review it?

kisvegabor, 2020-08-28:
I close it as the PR is merged.

closed 2020-08-28 by kisvegabor