lvgl/lvgl #1709: Also zoom in and rotate the canvas to show the error

lilengqiang, 2020-08-07:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

Also zoom in and rotate the canvas to show the error

**To Reproduce**

    const uint16_t canvas_w = 500;
    const uint16_t canvas_h = 322;
    
    lv_draw_img_dsc_t img_dsc;
    lv_draw_img_dsc_init(&img_dsc);
    
    const int bufsize = LV_CANVAS_BUF_SIZE_TRUE_COLOR(canvas_w, canvas_h);
    static lv_color_t *cbuf = (lv_color_t *) malloc(bufsize);
    
    lv_obj_t * canvas = lv_canvas_create(lv_scr_act(), NULL);
    lv_obj_set_pos(canvas , 100 ,100);
    lv_canvas_set_buffer(canvas, cbuf, canvas_w, canvas_h, LV_IMG_CF_TRUE_COLOR);
    lv_canvas_fill_bg(canvas, LV_COLOR_WHITE, LV_OPA_TRANSP);
    LV_IMG_DECLARE(test_picture);

    lv_canvas_draw_img(canvas, 0, 0, &test_picture, &img_dsc);

    lv_img_set_pivot(canvas, 0, 0);
    lv_img_set_zoom(canvas, 260);
    lv_img_set_angle(canvas, 450);
    lv_obj_set_drag(canvas, true);

**Expected behavior**

**Screenshots**
[image]

**Additional context**

Both zoom and rotate appear at the same time, and only one of the items is set to display properly.

kisvegabor, 2020-08-10:
Hi,

To reproduce I need to know the dimensions of `test_picture`. Or even better if you could attach it. :slightly_smiling_face: (it should be zip-ed to upload to GitHub)

lilengqiang, 2020-08-10:
[test_picture.zip](https://github.com/lvgl/lvgl/files/5049005/test_picture.zip)
@kisvegabor

kisvegabor, 2020-08-11:
I've investigated it and fixed the issue in the master branch

It was an integer overflow when large images was zoomed and rotated. 

See the fix here f066f0b8

lilengqiang, 2020-08-11:
ok.   Thanks

closed 2020-08-11 by lilengqiang