lvgl/lvgl #1710: About imgbtn cannot use radius and color picker size attributes to set the same simulator black screen.

Accidentaly-across, 2020-08-07:
Hello,

I found two problems during use.

Question 1:  Can't imgbtn set radius property? At present, I have found that setting the radius property does not work. I can't set the radius attribute like a button.

Question 2: color picker Set the rectangle type. If your size is the same, the simulator will have a black screen.

Looking forward to your answer.

embeddedt, 2020-08-07:
> Can't imgbtn set radius property? At present, I have found that setting the radius property does not work. I can't set the radius attribute like a button.

I think this is a known limitation. A workaround is to apply the radius to your images before converting them.

kisvegabor, 2020-08-10:
> I think this is a known limitation. A workaround is to apply the radius to your images before converting them.

Actually normal `lv_img` uses the radius and if clip_corner == true˙ it adds a mask before drawing the image to clip the corners. 
I see no problem with doing the same for image buttons too so I added it here: ebaeb1bf
@Accidentaly-across can you test it?

> color picker Set the rectangle type. If your size is the same, the simulator will have a black screen.

Please send a code snippet to reproduce.

Accidentaly-across, 2020-08-10:
color picker Set the rectangle type. If your size is the same, the simulator will have a black screen.
`void lv_ex_cpicker_2(void)
{
    lv_obj_t * cpicker;

    cpicker = lv_cpicker_create(lv_scr_act(), NULL);
    lv_cpicker_set_type(cpicker, LV_CPICKER_TYPE_RECT);
    lv_obj_set_size(cpicker, 100, 100);
    lv_obj_align(cpicker, NULL, LV_ALIGN_CENTER, 0, 0);
}
`
[image]

Accidentaly-across, 2020-08-10:
> > I think this is a known limitation. A workaround is to apply the radius to your images before converting them.
> 
> Actually normal `lv_img` uses the radius and if clip_corner == true˙ it adds a mask before drawing the image to clip the corners.
> I see no problem with doing the same for image buttons too so I added it here: [ebaeb1b](https://github.com/lvgl/lvgl/commit/ebaeb1bf7cd88d05cc14757365aabb952108afca)
> @Accidentaly-across can you test it?
Sorry, I didn't find a test sample. Can you send it to me?

embeddedt, 2020-08-10:
@Accidentaly-across Please update LVGL to the latest version on `master` and test that your code now works.

closed 2020-08-11 by Accidentaly-across