lvgl/lvgl #1712: [7.4] Empty space on a tabview is now clickable

embeddedt, 2020-08-10:
**Describe the bug**

Clicking the empty space to the left of the tabs in `lv_demo_widgets` results in Controls being clicked.

**To Reproduce**

Run `lv_demo_widgets` on any device. Click the area to the left of Controls or even to the left of the Dark/Light switch.

**Expected behavior**

Empty space should not be clickable.

kisvegabor, 2020-08-11:
Actually it's a feature :slightly_smiling_face: 

To avoid having inactive areas in the button matrix the space around the buttons are mapped to the nearest button.

In the demo, the tabs have large padding to flush the button to the right an the "padding" is mapped to the first button. 

A possible solution would be to use "margin"  instead. Although it's not handled by the tabview it seems easy to add it.

What do you think?

embeddedt, 2020-08-11:
Margin makes more sense, and doesn't add to the object's click area, so I agree with using that.

kisvegabor, 2020-08-13:
I implemented it but I'm that happy with the result:
[image]

The gray separator line below the tabs is part of the button matrix. This way it doesn't fill the whole width.

Another approach could be to limit the distance in which the clicks are mapped to buttons. E.g. 0.5 inches can be a reasonable choice.

embeddedt, 2020-08-13:
I'm looking at the old implementation and it appears that the padding is being set on the tab background (`LV_TABVIEW_PART_TAB_BG`). Why does this padding affect button clicks? Shouldn't it be padding on `LV_TABVIEW_PART_TAB_BTN` that affects tab clicks? I think that would solve the problem.

kisvegabor, 2020-08-13:
Padding on `LV_TABVIEW_PART_TAB_BTN` makes the buttons larger but there are gaps between the buttons. The gaps are "assigned" to the buttons to be sure you don't click and "inactive" area.

embeddedt, 2020-08-14:
> there are gaps between the buttons. The gaps are "assigned" to the buttons to be sure you don't click and "inactive" area.

Okay, so it's a specific workaround for touchscreens then. I guess in that case having the assignment limited to a certain range would be an appropriate fix.

kisvegabor, 2020-08-14:
> Okay, so it's a specific workaround for touchscreens then. I guess in that case having the assignment limited to a certain range would be an appropriate fix.

I've fixed it according to this.

commit 1815ff4b (2020-08-14): Limit extra button click area of button matrix's buttons. Fixes #1712
closed 2020-08-14 by kisvegabor