lvgl/lvgl #1713: The efficiency of animation

daiyinger, 2020-08-11:
# All enhancement requests or questions should be directed to the Forum.
`if(a->exec_cb) a->exec_cb(a->var, new_value);`
Can change to
`if(new_value != a->current)
{
       a->current = new_value;
       /*Apply the calculated value*/
       if(a->exec_cb) a->exec_cb(a->var, new_value);
}`

closed 2020-08-11 by daiyinger

kisvegabor, 2020-08-11:
Why have you closed it?

daiyinger, 2020-08-11:
> Why have you closed it?

I was going to make a suggestion on how to improve the efficiency of animation, but I found it a bit disorganized

kisvegabor, 2020-08-12:
The idea was correct any way.

Just format the code like this
````
```c
your
code 
here
```
````

daiyinger, 2020-08-14:
My idea is to add a variable to the lv_anim_t structure that holds the current value.
If the path evaluates to the current value, the animation callback function is no longer called.
`
typedef struct _lv_anim_t {
    void * var;                  /**<Variable to animate*/
    lv_anim_exec_xcb_t exec_cb;   /**< Function to execute to animate*/
    lv_anim_start_cb_t start_cb; /**< Call it when the animation is starts (considering `delay`)*/
    lv_anim_ready_cb_t ready_cb; /**< Call it when the animation is ready*/
    lv_anim_path_t path;         /**< Describe the path (curve) of animations*/
    int32_t start;               /**< Start value*/
    int32_t current;		     /**< Current value */
    int32_t end;                 /**< End value*/
    int32_t time;               /**< Animation time in ms*/
    int32_t act_time;            /**< Current time in animation. Set to negative to make delay.*/
    uint32_t playback_delay;     /**< Wait before play back*/
    uint32_t playback_time;      /**< Duration of playback animation*/
    uint32_t repeat_delay;       /**< Wait before repeat*/
    uint16_t repeat_cnt;         /**< Repeat count for the animation*/
    uint8_t early_apply  : 1;    /**< 1: Apply start value immediately even is there is `delay` */
#if LV_USE_USER_DATA
    lv_anim_user_data_t user_data; /**< Custom user data*/
#endif

    /*Animation system use these - user shouldn't set*/
    uint32_t time_orig;
    uint8_t playback_now : 1; /**< Play back is in progress*/
    uint32_t has_run : 1;     /**< Indicates the animation has run in this round*/
} lv_anim_t;

static inline void lv_anim_set_values(lv_anim_t * a, lv_anim_value_t start, lv_anim_value_t end)
{
    a->start = start;
    a->current = start;
    a->end   = end;
}

if(new_value != a->current)
{
    a->current = new_value;
    /*Apply the calculated value*/
   if(a->exec_cb) a->exec_cb(a->var, new_value);
}
`

daiyinger, 2020-08-14:
My idea is to add a variable to the lv_anim_t structure that holds the current value.
If the path evaluates to the current value, the animation callback function is no longer called.
typedef struct _lv_anim_t { 
...
int32_t start;               /**< Start value*/
int32_t current;		     /**< Current value */
int32_t end;                 /**< End value*/
...
} lv_anim_t;

static inline void lv_anim_set_values(lv_anim_t * a, lv_anim_value_t start, lv_anim_value_t end)
{
a->start = start;
a->current = start;
a->end = end;
}

if(new_value != a->current)
{
a->current = new_value;
/Apply the calculated value/
if(a->exec_cb) a->exec_cb(a->var, new_value);
}
`

daiyinger, 2020-08-14:
My idea is to add a variable to the lv_anim_t structure that holds the current value.
If the path evaluates to the current value, the animation callback function is no longer called.
`
typedef struct _lv_anim_t {
...
int32_t start; /< Start value*/
int32_t current; /< Current value /
int32_t end; /**< End value/
...
} lv_anim_t;

static inline void lv_anim_set_values(lv_anim_t * a, lv_anim_value_t start, lv_anim_value_t end)
{
a->start = start;
a->current = start;
a->end = end;
}

if(new_value != a->current)
{
a->current = new_value;
/Apply the calculated value/
if(a->exec_cb) a->exec_cb(a->var, new_value);
}
`

kisvegabor, 2020-08-17:
I like the idea. Can you send a pull request?

For better readability - as I mentioned above - please put the code blocks in the comments into:
````
```c
code
here
```
````