lvgl/lvgl #1718: lvgl crashes when creating a container as a child of a tab of tabview

mahyar1284, 2020-08-12:
when I create a container as a child of a tab in a tabview, lvgl crashes. but when I create it as a child of lv_scr_act() it's ok.

    lv_obj_t* tabview;
    tabview = lv_tabview_create(lv_scr_act(), NULL);
    lv_obj_align(tabview, title_txt, LV_ALIGN_OUT_BOTTOM_MID, 0, 0);
    /*Add 3 tabs (the tabs are page (lv_page) and can be scrolled*/
    tab1 = lv_tabview_add_tab(tabview, "Transducer 1");

        lv_obj_t* t4_btns_container = lv_cont_create(tab1, NULL);
        lv_obj_set_auto_realign(t4_btns_container, true);
        lv_cont_set_fit(t4_btns_container, LV_FIT_TIGHT);
        lv_cont_set_layout(t4_btns_container, LV_LAYOUT_ROW_MID);
        lv_obj_align(t4_btns_container, NULL, LV_ALIGN_IN_BOTTOM_MID, 0, 0);
        lv_obj_set_style_local_bg_opa(t4_btns_container, LV_CONT_PART_MAIN, LV_STATE_DEFAULT, 0);
        lv_obj_set_style_local_border_opa(t4_btns_container, LV_CONT_PART_MAIN, LV_STATE_DEFAULT, 0);

        lv_obj_t* t4_prg_btn_label;
        lv_obj_t* t4_prg_btn = lv_btn_create(t4_btns_container, NULL);
        //lv_obj_set_event_cb(stBtn, stBtn_onClick);
        t4_prg_btn_label = lv_label_create(t4_prg_btn, NULL);
        lv_label_set_text(t4_prg_btn_label, "Reset All");

        lv_obj_t* t4_ply_btn_label;
        lv_obj_t* t4_ply_btn = lv_btn_create(t4_btns_container, NULL);
        //lv_obj_set_event_cb(auBtn, auBtn_onClick);
        t4_ply_btn_label = lv_label_create(t4_ply_btn, NULL);
        lv_label_set_text(t4_ply_btn_label, LV_SYMBOL_SETTINGS);
        lv_obj_set_width(t4_ply_btn, lv_obj_get_height(t4_ply_btn));

I want to have 2 buttons in a container in a tab of a tabview.

[image]

this is the visual studio error in lvgl simulator

kisvegabor, 2020-08-12:
Hi,

It results in an infinite recursive cycle. If you align an object to very bottom but the parent has some padding and "auto-fit" it will become larger to involve the children  + padding. But after this, the child is not on the very bottom, so parent resites again, and so on.

Try something like this:
```c
  lv_obj_t * tab1 = lv_tabview_add_tab(tabview, "Transducer 1");  
  lv_page_set_scrollable_fit(tab1, LV_FIT_PARENT); /*Size to the parent (tabview), no auto sizing based o nthe children*/

  lv_obj_t* t4_btns_container = lv_cont_create(tab1, NULL);
  lv_cont_set_fit(t4_btns_container, LV_FIT_TIGHT);
  lv_cont_set_layout(t4_btns_container, LV_LAYOUT_ROW_MID);
  lv_obj_set_auto_realign(t4_btns_container, true);
  lv_obj_align(t4_btns_container, NULL, LV_ALIGN_IN_BOTTOM_MID, 0, 0);
//  lv_obj_set_style_local_bg_opa(t4_btns_container, LV_CONT_PART_MAIN, LV_STATE_DEFAULT, 0);
//  lv_obj_set_style_local_border_opa(t4_btns_container, LV_CONT_PART_MAIN, LV_STATE_DEFAULT, 0);
```

mahyar1284, 2020-08-12:
Thanks for your reply.

closed 2020-08-12 by mahyar1284