lvgl/lvgl #1720: Use image for gauge background

david-res, 2020-08-12:
Raising this here as I have not signed up to the forum yet, and my use of LVGL would be dependant on the implementation of this feature.

Would it be possible to draw a gauge background from an image, and not just the gauge needle? Reading through the guide I did not see such an option.
The gauges currently available are nice, but still rather simple to say the least. I have a gauge template that has a range of -30 to +30, where -30 to 0 start at 270° point and ends at 180°, and the 0 to 30 range starts from 180° all the way through 360°

If this would be possible, it would allow me to port my entire project to LVGL (which I'd really like to do)

kisvegabor, 2020-08-13:
Hi,

It's pretty simple to do. Just make the gauge's background transparent and create it on an image.

```c
  lv_obj_t * gauge = lv_gauge_create(bg_image, NULL);
  lv_obj_clean_style_list(gauge, LV_GAUGE_PART_MAIN);   //The default is transparent
```

I'd be very curious about from which platform do you plan to move ot LVGL. Just to know what are the good/weak points of LVGL in your eyes. Can you tell me about it?

david-res, 2020-08-14:
@kisvegabor thank you for the snippet! I will give it a try as soon as I have some spare time to test it out.

I'm currently using a Teensy 4.0 and an HX8357 display over SPI to present a gauge and some numeric figures.
It works very well and smooth as I'm using a frame buffer, a clip rectangle (to update changed areas only) and slightly overclocked SPI bus.
[Here](https://www.youtube.com/watch?v=RSyV6P0iJ90) is a video of it

I want to add more functionality to my project (data logging to SD) and measurement unit changing in the UI which will requires menus, setting screens, toggle buttons etc.
Initially I has seen a demo video of Qt for MCU's which was really amazing, but I think it will be a long time until it supports Arduino framework (plus it's not actually free to use). So I kept searching and found LVGL which seems to have improved A LOT since I last saw it over a year ago (kudos on the upgrades, its really impressive)

So to be honest, I think LVGL will defiantly answer my needs and hopefully perform with the same speed I have now (I do plan to upgrade to the Teensy 4.1 with additional 8Mbit PSRAM and an ILI9488 display), but If LVGL would some day have the animation and transitions capabilities I saw on Qt, it would make it even more appealing to myself and others to allow our projects to look super impressive -industry grade looks and performance

david-res, 2020-08-16:
I've built a small demo just to see how it displays on the simulator

This is my gauge object:
`#include "../../lv_examples.h"
#include "guage_bg_bar.c" //Image generated on LVGL image creator

LV_IMG_DECLARE(guage_bg_bar);

void lv_ex_get_started_guage(void)
{

	static lv_color_t colors[1];
	colors[0] = LV_COLOR_BLUE;

	lv_obj_t * guage_bg = lv_img_create(lv_scr_act(), NULL);
	lv_img_set_src(guage_bg, &guage_bg_bar);
    lv_obj_t * gauge1 = lv_gauge_create(guage_bg, NULL);
    lv_obj_clean_style_list(gauge1, LV_GAUGE_PART_MAIN);   //The default is transparent
    lv_gauge_set_needle_count(gauge1, 0, colors);
    lv_obj_set_size(gauge1, 240, 240);
    lv_obj_align(gauge1, NULL, LV_ALIGN_CENTER, 0, 0);

    lv_gauge_set_value(gauge1, 0, 40);
}`

It compiles but the build ends with an error in Eclipse:
`duplicate symbol '_guage_bg_bar' in:
    ./lv_examples/src/lv_ex_get_started/guage_bg_bar.o
    ./lv_examples/src/lv_ex_get_started/lv_guage_bar.o
duplicate symbol '_guage_bg_bar_map' in:
    ./lv_examples/src/lv_ex_get_started/guage_bg_bar.o
    ./lv_examples/src/lv_ex_get_started/lv_guage_bar.o
ld: 2 duplicate symbols for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [lv_gauge_demo] Error 1
"make all" terminated with exit code 2. Build might be incomplete.`

Not sure how to fix though...

embeddedt, 2020-08-16:
You don't need to `#include` your image file; it should be built as a separate file. You just need the `LV_IMG_DECLARE` line.

david-res, 2020-08-16:
Thanks. I removed the image include and now it runs, but not looking like what I expected..
[image]

The text should be white with white ticks, and no range from LVGL. Here's and example of the transparent image:
[image]

embeddedt, 2020-08-16:
@david-res What settings did you use to convert the image?

david-res, 2020-08-16:
True color, C array.

I tried Raw, C array and just got a white screen, no needle - it might be displaying fine, but not seeing the needle worries me.

kisvegabor, 2020-08-17:
If the image has transparency you should use "true-color with alpha"

david-res, 2020-08-17:
Okay, resized the image as I had reduced the screen size but not the image size. Its looking much better but still have default LVGL ticks and range
[image]
Also, is there a way to change the length of the needle?

kisvegabor, 2020-08-18:
You just need to clean another part of the gauge as well. 
```c
lv_obj_clean_style_list(gauge, LV_GAUGE_PART_MAJOR);  
```

david-res, 2020-08-18:
@kisvegabor thanks for all the help, that removed the ticks.
I added `lv_gauge_set_scale(gauge1, 270, 0, 0);` to remove the number labels
 and then `lv_gauge_set_angle_offset(gauge1, -45);` to fix the offset of the scale

I'll keep tinkering with it and see what else I can do.

I must admit though, the documentation is very hard for me to interpret, so I guess I will sign up to the forum and use that space for guidance - you can close this issue

kisvegabor, 2020-08-18:
> I must admit though, the documentation is very hard for me to interpret

Could you elaborate on this? I'd be very curious to hear what caused difficulties for you in order to improve those parts.

closed 2020-08-18 by kisvegabor

david-res, 2020-08-18:
> > I must admit though, the documentation is very hard for me to interpret
> 
> Could you elaborate on this? I'd be very curious to hear what caused difficulties for you in order to improve those parts.

I think it mainly how it's structured. It's unclear to me what the main object is - is it a generic object, a page, a window etc.
After that, the styling is confusing, for example, how to apply a background color to a window(or should it be a page?), or even what elements can be styled and how - like the gauge needle - I saw some APIs suggested online which don't seem to be documented in the user guide

Now it might be just me, as I do not have a very big programming background, I'm just a hobbyist. It looks like lvgl can do some amazing things on MCUs, It's perhaps just a tad too complex for me to understand at first glance.

I think what might be nice is to provide a template with a page/window, multiple containers with some generic objects like a button, list, LED etc. Seeing what something simple looks like helps build knowledge from trial and error.

Thanks again for all the assistance - defiantly going to give this a try, it just might take me a while to understand the structure

kisvegabor, 2020-08-20:
Thanks, David. I started a [discussionħ(https://github.com/lvgl/lvgl/issues/1719#issuecomment-675935063) to see how we can simplify stuff to make them easier to digest