lvgl/lvgl #1721: Tabview bug:scrllable radius

lilengqiang, 2020-08-13:
**Describe the bug**

when I set the radius of the scrllable, partial rounded corners have no effect

**To Reproduce**

#include "../../../lv_examples.h"
#if LV_USE_TABVIEW

void lv_ex_tabview_1(void)
{
    /*Create a Tab view object*/
    lv_obj_set_style_local_bg_color(lv_scr_act(), LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, LV_COLOR_GREEN);

    lv_obj_t *tabview;
    tabview = lv_tabview_create(lv_scr_act(), NULL);

    lv_obj_set_size(tabview, LV_HOR_RES - 20, LV_VER_RES -20);
    lv_obj_set_pos(tabview,10,10);

    lv_obj_set_style_local_bg_opa(tabview, LV_TABVIEW_PART_BG, LV_STATE_DEFAULT, LV_OPA_TRANSP);
    lv_obj_set_style_local_radius(tabview, LV_TABVIEW_PART_BG_SCRLLABLE, LV_STATE_DEFAULT, LV_DPX(20));
    lv_obj_set_style_local_bg_opa(tabview, LV_TABVIEW_PART_BG_SCRLLABLE, LV_STATE_DEFAULT, LV_OPA_COVER);

    /*Add 3 tabs (the tabs are page (lv_page) and can be scrolled*/
    lv_obj_t *tab1 = lv_tabview_add_tab(tabview, "Tab 1");
    lv_obj_t *tab2 = lv_tabview_add_tab(tabview, "Tab 2");
    lv_obj_t *tab3 = lv_tabview_add_tab(tabview, "Tab 3");

    /*Add content to the tabs*/
    lv_obj_t * label = lv_label_create(tab1, NULL);
    lv_label_set_text(label, "This the first tab\n\n"
                             "If the content\n"
                             "of a tab\n"
                             "become too long\n"
                             "the it \n"
                             "automatically\n"
                             "become\n"
                             "scrollable.");

    label = lv_label_create(tab2, NULL);
    lv_label_set_text(label, "Second tab");

    label = lv_label_create(tab3, NULL);
    lv_label_set_text(label, "Third tab");
}
#endif

**Expected behavior**

**Screenshots**

[image]

**Additional context**

The right two corners of the first tab don't work, and the four corners of the second tab don't work

embeddedt, 2020-08-13:
That's the expected behavior. The tabview is implemented as a large scrollable object with several pages inside. What you've done is to set the radius on the scrollable itself - that means you're only looking at the left and middle of it.

If you want every page to have rounded corners I think you need to set the radius on the tabs themselves, make sure the tabs each have a background color, and make sure that the scrollable's background is transparent.

lilengqiang, 2020-08-14:
ok thanks

closed 2020-08-14 by lilengqiang