lvgl/lvgl #1722: 24 bit RGB888 support

kisvegabor, 2020-08-13:
I implemented a minimal and hacky RGB888 support in `feat/rgb888` branch.

It is useful because:
- usually display doesn't have alpha channel and using ARGB888 makes display buffers 25% larger
- ILI9488 doesn't support RGB565 only RGB666 format packet into 3 bytes. RGB888 can be used directly in this case. (Now a conversation is required)

It can be tested by setting `LV_COLOR_DEPTH 24` in `lv_conf.h`

I tested it in the simulator but feedback would be very welcome from real hardware too. (E.g.  with ILI9488)

There is one question regarding the data format. `lv_color24_t` is defined like [this](https://github.com/lvgl/lvgl/blob/07382b5a4b87a401aa4cd3317d37333aed337e03/src/lv_misc/lv_color.h#L250-L254).

The problem is it's no guaranteed that `lv_color24_t x[2]` is 6 bytes long because there could be paddings.  
I opened this [StackOverflow question](https://stackoverflow.com/questions/62957621/padding-of-a-struct-in-c-with-3-bytes?noredirect=1#comment111361674_62957621) and although packing is not guaranteed we couldn't find ANY compiler that adds extra bytes. What do you think about this case?

embeddedt, 2020-08-13:
> What do you think about this case?

I'd recommend adding the ability for someone to specify their compiler's packed attribute - that should address this problem.

kisvegabor, 2020-08-13:
> I'd recommend adding the ability for someone to specify their compiler's packed attribute - that should address this problem.

At first look, an `LV_ATTRIBUTE_COLOR` or `LV_ATTRIBUTE_PACKED` option doesn't bother too much. But it needs to be applied every place where a color array or color variable is defined. 

I think it's very unlikely that it causes issues because only a few people use RGB888, and almost all compiler behaves well. 
I suggest keeping this as a backup plan until someone will have an issue with it.

embeddedt, 2020-08-13:
> But it needs to be applied every place where a color array or color variable is defined.

I think it only needs to be applied to the `struct` definition.

kisvegabor, 2020-08-13:
Hmm, I thought it's compiler-specific exactly where the attributes need to be placed.  Or is it general?
```c
typedef struct __attribute__((packed, aligned(4))) {
    uint8_t x[3];
} foo_t;
```

embeddedt, 2020-08-14:
> I thought it's compiler-specific exactly where the attributes need to be placed. Or is it general?

First off, I'm no expert on all the compilers out there. :wink: I think this kind of thing is technically compiler-specific (since attributes are not standardized), but I also think having some attribute (even with fixed placement) is better than no attribute, especially since newer GCC versions tend to add optimizations that break this kind of thing.

I'm not aware of any compiler that wouldn't support the placement you gave in that example.

kisvegabor, 2020-08-14:
Okay, let's do this way then :slightly_frowning_face:

turoksama, 2020-08-28:
@kisvegabor 
Hello there,
I think we should be careful to define data structures. More effecient, less RAM consumption  in proper order.
The compiler is not smart enough to identify access to 32-bit data and then do some optimizations.

> Hmm, I thought it's compiler-specific exactly where the attributes need to be placed. Or is it general?
> 
> ```c
> typedef struct __attribute__((packed, aligned(4))) {
>     uint8_t x[3];
> } foo_t;
> ```
Is it effective in 32-bit data types only?

kisvegabor, 2020-08-28:
Hi Samuel,

Sorry for the confusion, by `typedef struct __attribute__((packed, aligned(4))) {` I just want to ask about the `attribute` placement.
Finally, it should look like:
```c
typedef struct __attribute__((packed)) {
    uint8_t x[3];
} foo_t;
```

I believe it's really slower than 32 bit color format. But with this at least you have the option to choose between less RAM (24 bit) and better performance (32 bit). I'm not sure what is the real performance difference though. Considering the following cases it shouldn't be too much:
1. Set color channels (r,g,b): byte access in both 32-bit in 24-bit case
2. Fill a large array with a color: can be optimized by packing 4 x 24 bits (12 bytes)  to use word write where possible
3. Copy an image: no color format is concerned, uses simple `memcpy`

What do you think?

embeddedt, 2020-08-28:
> Copy an image: no color format is concerned, uses simple `memcpy`

There's a potential performance loss if one of the pointers is unaligned (I think it could happen with 24-bit, since the pixels are only 3 bytes long). I know our `memcpy` implementation does some magic with unaligned pointers, but it's not smart enough to handle all these cases. For instance, if the source image is 4-byte aligned, but the destination is 3-byte aligned, it appears that it will just do a bytewise copy.

turoksama, 2020-08-28:
Hi kisvegabor
Definitely I agree with you, sorry for the confusion about what i said.
I meant that alignment(4) is prefered for 32-bit data. 
It is challenging to deal with 24-bit RGBs, and 
```
typedef struct __packed() {
    uint8_t x[3];
} foo_t;
```
is a way of compromise.

@embeddedt 
memcpy seems fine to me, but let's do some magic fast 4-byte aligned copy as much as possible.

kisvegabor, 2020-08-31:
@embeddedt 
> There's a potential performance loss if one of the pointers is unaligned

Ah, that's true. And theoretically, it will be unaligned in 3/4 of cases. 

@turoksama 
> is a way of compromise.

Yes, that's why RGB888 wasn't supported so far because I also see it a little bit hacky. But it's better to have once it needs 25% less memory. 150kB difference on a 480x320 image. 
BTW, it also helps to save some memory bandwidth on external SRAM.

kisvegabor, 2020-09-23:
It's added this to the [ROADMAP](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md) so I close this issue. We can reopen it if it becomes active again.

closed 2020-10-07 by kisvegabor

EagleLiao, 2022-05-19:
I tried LVGL on my board which has a 24bit RGB888 screen. Yes it is not supported. I have to config LVGL to 32bit and convert the format in lv_drivers/fbdev_flush, and then display normally. But the animation seems not correct at all. Anyone know why?

kisvegabor, 2022-05-22:
Converting the pixels in should work in "normal mode" too. Can you show you `flush_cb`?

EagleLiao, 2022-05-26:
> Converting the pixels in should work in "normal mode" too. Can you show you `flush_cb`?

`

#define PIXEL8888ALPHA(pixelval)	(((pixelval) >> 24) & 0xff)
#define PIXEL8888RED(pixelval)  	(((pixelval) >> 16) & 0xff)
#define PIXEL8888GREEN(pixelval)	(((pixelval) >> 8) & 0xff)
#define PIXEL8888BLUE(pixelval) 	((pixelval) & 0xff)
/**
 * Flush a buffer to the marked area
 * @param drv pointer to driver where this function belongs
 * @param area an area where to copy `color_p`
 * @param color_p an array of pixels to copy to the `area` part of the screen
 */
void fbdev_flush(lv_disp_drv_t * drv, const lv_area_t * area, lv_color_t * color_p)
{
    if(fbp == NULL ||
            area->x2 < 0 ||
            area->y2 < 0 ||
            area->x1 > (int32_t)vinfo.xres - 1 ||
            area->y1 > (int32_t)vinfo.yres - 1) {
        lv_disp_flush_ready(drv);
        return;
    }

    /*Truncate the area to the screen*/
    int32_t act_x1 = area->x1 < 0 ? 0 : area->x1;
    int32_t act_y1 = area->y1 < 0 ? 0 : area->y1;
    int32_t act_x2 = area->x2 > (int32_t)vinfo.xres - 1 ? (int32_t)vinfo.xres - 1 : area->x2;
    int32_t act_y2 = area->y2 > (int32_t)vinfo.yres - 1 ? (int32_t)vinfo.yres - 1 : area->y2;

    lv_coord_t w = (act_x2 - act_x1 + 1);
    long int location = 0;
    long int byte_location = 0;
    unsigned char bit_location = 0;

    /*32 or 24 bit per pixel*/
    if(vinfo.bits_per_pixel == 32 || vinfo.bits_per_pixel == 24) {
        uint8_t * fbp8 = (uint8_t *)fbp;
        int32_t x;
        int32_t y;
        for(y = act_y1; y <= act_y2; y++)
        {
            location = (act_x1 + vinfo.xoffset) + (y + vinfo.yoffset) * finfo.line_length / 4;
            for (x = act_x1; x <= act_x2; x++)
            {
                uint32_t rgba = (*(color_p + (x - act_x1))).full;

                *fbp8++ = PIXEL8888BLUE(rgba);
                *fbp8++ = PIXEL8888GREEN(rgba);
                *fbp8++ = PIXEL8888RED(rgba);
            }
            color_p += w;
        }
    }
	
    //May be some direct update command is required
    //ret = ioctl(state->fd, FBIO_UPDATE, (unsigned long)((uintptr_t)rect));

    lv_disp_flush_ready(drv);
}
`

kisvegabor, 2022-05-26:
> I don't know how to attach a good formatted code segment here. 

Code blocks (not inline code) can be formatted like:
````c
```c
if(a > b) {
  a = 5;
}
```
````

I edited your post to see it in the practice.

In the flush_cb you should do something like [this](https://github.com/lvgl/lv_drivers/blob/0b01584e299e9a7bb880678c33d4da6fe472fc9e/sdl/sdl.c#L165-L169).

The gist is that if LVGL refreshes only a small area, you need to copy the rendered image to the same area on the frame buffer.

EagleLiao, 2022-05-27:
> Code blocks (not inline code) can be formatted like:
>
> ```c
> ```c
> if(a > b) {
>   a = 5;
> }
> ```
> ```
>
> I edited your post to see it in the practice.
>
> In the flush_cb you should do something like [...]

Yes. That's why my screen only works after I config it to 'direct_mode and full_refresh' mode. The draw_buf is same size with screen, the small area means the full screen.

FASTSHIFT, 2022-10-12:
@kisvegabor 
Hi, can we restart this discussion? the 24bpp color is great for saving precious SRAM.

kisvegabor, 2022-10-12:
I'v already implemented it in https://github.com/lvgl/lvgl/tree/arch/drivers

I try my best to merge the new driver architecture in a week. It will bring simplified driver API, new color format management, and direct RGB888 support.

FASTSHIFT, 2022-10-12:
> I'v already implemented it in https://github.com/lvgl/lvgl/tree/arch/drivers
> 
> I try my best to merge the new driver architecture in a week. It will bring simplified driver API, new color format management, and direct RGB888 support.

Great! I'll assist with testing.