lvgl/lvgl #1724: Animation curruption in the lastest master branch/printer demo

barbiani, 2020-08-14:
I have just pulled the last commit and found that the animated icons are showing some corruption when clicked or fading in/out.

Not show when it happened because it has been a while since I tested it in a VS/SDL environment.

It is the plain printer demo with the windows SDL driver code from the other repository.

Anyone else saw these?

Fading in...
[image]

Clicked...
[image]

kisvegabor, 2020-08-14:
With the Eclipse SDL simulator it works fine.
Can someone confirm that it's working in Eclipse?

embeddedt, 2020-08-14:
I will test it in a few hours.

barbiani, 2020-08-17:
Might be only here but:

Setting to
#define LV_COLOR_DEPTH     32
#define LV_COLOR_SCREEN_TRANSP    1

Turns to a semi transparent red area during the animation. At least no more corruption.

I am using SDL (800x600) with two 32 lines buffers.

[image]

sveniv, 2020-08-19:
I had the same problem, but reversing the change made by commit 28807b6a77b96ffcb144bc433f932a2f0b67a36e fixed it for me. [Link to commit](https://github.com/lvgl/lvgl/commit/28807b6a77b96ffcb144bc433f932a2f0b67a36e)
(I also tried commenting out the line, but that also gave the same glitches.)

I'm running this on Linux and `sdl2-config --version` returns `2.0.8` if that helps.

sveniv, 2020-08-20:
I removed the old problematic `lv_sim_eclipse_sdl` project from eclipse, cloned a fresh copy from github and imported that into Eclipse. The printer demo now works for me without any changes other than un-commenting/commenting the function calls for the examples in `main.c`. I suspect that I had made some changes to the project configuration in Eclipse that caused it.

kisvegabor, 2020-08-20:
Hmm, very strange. Anyway, good to hear it works with a fresh copy. 

I close this issue then.

tjstyle, 2020-08-30:
[image]

I also have the same problem with the latest master branch using STM32 hardware and reversing the https://github.com/lvgl/lvgl/commit/28807b6a77b96ffcb144bc433f932a2f0b67a36e commit the problem is gone.

kisvegabor, 2020-08-31:
It seems LVGL thinks that the images cover the area while they are transparent. However, it's checked [here](https://github.com/lvgl/lvgl/blob/694d9eff199ab9725b325e752003a72bb911e936/src/lv_widgets/lv_img.c#L605).

Can you try what happens if you add `return LV_DESIGN_RES_NOT_COVER;` immediately in `if(mode == LV_DESIGN_COVER_CHK)` .

tjstyle, 2020-08-31:
confirmed, using `return LV_DESIGN_RES_NOT_COVER;` problem is gone

kisvegabor, 2020-09-01:
Thanks for testing!

I've pushed a fix to master. Can you try it?

tjstyle, 2020-09-01:
Tested, the latest master solve the problem, thanks!

kisvegabor, 2020-09-01:
Gret! Thank you for the feedback!

closed 2020-09-01 by kisvegabor