lvgl/lvgl #1728: [7.4] Tabview button does not show press overlay

embeddedt, 2020-08-15:
**Describe the bug**

The tab buttons on the tabview do not show a press overlay when clicked. I am using the light material theme.

**To Reproduce**

Run `lv_demo_widgets`. Press one of the tab buttons. Observe that the press overlay does not appear.

**Expected behavior**

A gray overlay should appear when the button is clicked.

**Additional context**

If you switch the theme to dark using the switch, the press overlay appears correctly. If you then switch the theme back to light, the press overlay still appears correctly.

kisvegabor, 2020-08-17:
It works well for me with the latest master.

embeddedt, 2020-08-17:
I'm using `dev`.

kisvegabor, 2020-08-18:
Thanks, I just fixed it.

embeddedt, 2020-08-18:
Thanks for the quick fix!

commit 677ce48b (2020-08-18): tabview: fix press state style update
fixes: #1728