lvgl/lvgl #1731: add some new  feature widgets

brilliantsong, 2020-08-17:
I hope the lvgl upstream can support some new widgets as show :
Radio Buttons
Animated Digital Clock
Animated Analog Clock
Text Progress bar
Image Progress bar
Date text box
Repeat Button

kisvegabor, 2020-08-17:
Hi, 

Thanks for the ideas.

Please post the future proposals into the forums [Feature request](https://forum.lvgl.io/c/feature-request/9) category (one idea per topic) and explain them in more detail. Expected behavior, example images, etc.

closed 2020-08-17 by kisvegabor

embeddedt, 2020-08-17:
@brilliantsong Have you looked at some of our examples? Some of your suggestions are already implemented or can be implemented easily with a little bit of external event logic.

> Radio Buttons

These can already be implemented with checkboxes and some logic in an event handler.

> Text Progress bar

This can already be done in 7.0+. See the `lv_demo_widgets` example.

> Repeat Button

Not sure what you mean by this... are you suggesting that the button should auto-repeat clicks? If so, this is already supported; you just need to listen to the `LV_EVENT_LONG_PRESSED_REPEAT` event.