lvgl/lvgl #1735: Plenty of objs in page slow down system speed.

iDalink, 2020-08-19:
# All enhancement requests or questions should be directed to the Forum.

I append lots of objs in page. These objs lay as list style, and accept  vertical scroll. But it response slowly when I drag it.
Does there are any solution to fix the slow effect? 
I realize that iOS`s `UITableView` use `resuable view` to improve performance.

We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

embeddedt, 2020-08-19:
> I append lots of objs in page. These objs lay as list style, and accept vertical scroll. But it response slowly when I drag it.
Does there are any solution to fix the slow effect?

How many objects are you using? Can you provide a code sample that can be used for benchmarking?

turoksama, 2020-08-28:
Well, I think it is reasonable when you have more jobs to do.
It is a sufficient way that the library do so far, cpu needs more process time to deal with heavy loads.
At the moment, FYI, less content(visual effect, change) updates, use gpu acceleration, increase cpu freq, increase dram freq, use more advance cpu platform.
What do you think?

kisvegabor, 2020-09-01:
I close this issue in lack of feedback from the OP.

closed 2020-09-01 by kisvegabor

iDalink, 2020-09-07:
> > I append lots of objs in page. These objs lay as list style, and accept vertical scroll. But it response slowly when I drag it.
> > Does there are any solution to fix the slow effect?
> 
> How many objects are you using? Can you provide a code sample that can be used for benchmarking?

I make following example code. It spend 15 second at my single core arm-A7 board.

```C

static const char *test_textg1_1 = "gitee开源的一个lvgl的demo,包括控件使用例程以及应用，视频简单介绍了使用方法以及桌面系统的框架开发板包括控件使用例程以及应用，视频简单介绍了使用方法以及桌面系统的框架开发板包括控件使用例程以及应用，视频简单介绍了使用方法以及桌面系统的框架开发板包括控件使用例程以及应用，视频简单介绍了使用方法以及桌面系统的框架开发板";
static const char *test_textg1_2 = "I append lots of objs in page. These objs lay as list style, and accept vertical scroll. But it response slowly when I drag it.";
static const char *test_textg2_1 = "LVGL分析1 littlevgl线程littlevgl有几个线程,作用是什么?三个,主线程一个,和在主线程的hal_init函数中创建的另两个sdl线程。主线程完成一系列初始化工作后";
static const char *test_textg2_2 = "We use GitHub issues for development related discussions.";

#define COLOR_BG_BLOCK      lv_color_hex(0x121212)
#define COLOR_TXT_BRIGHT    lv_color_hex(0xD0D0D0)
extern lv_font_t arial_20;
#include <time.h>
unsigned int os_ticks_get(void)
{
    struct timespec ts;
    clock_gettime(CLOCK_MONOTONIC, &ts);
    return (ts.tv_sec * 1000 + ts.tv_nsec / 1000000);
}

lv_obj_t *create_wrapper(lv_obj_t *par, int w)
{
    lv_obj_t *cont = lv_cont_create(par, NULL);
    lv_cont_set_fit4(cont, LV_FIT_NONE, LV_FIT_NONE, LV_FIT_TIGHT, LV_FIT_TIGHT);
    lv_obj_set_size(cont, w, 0);
    lv_cont_set_layout(cont, LV_LAYOUT_COL_M);
	lv_page_glue_obj(cont, true);
	return cont;
}

lv_obj_t *create_block_wrapper(lv_obj_t *par, int w)
{
	lv_obj_t *cont = create_wrapper(par, w);
	static lv_style_t style;
	style = lv_style_pretty;
    style.body.radius = 10;
	style.body.main_color = COLOR_BG_BLOCK;
	style.body.grad_color = COLOR_BG_BLOCK;
	lv_cont_set_style(cont, LV_CONT_STYLE_MAIN, &style);
	return cont;
}

lv_obj_t *create_label(lv_obj_t *par, int w,  const char *title)
{
	static lv_style_t label_style;
	label_style = lv_style_pretty;
	label_style.body.main_color = LV_COLOR_RED;
	label_style.body.grad_color = LV_COLOR_RED;
	label_style.body.radius = 0;
	label_style.text.color = COLOR_TXT_BRIGHT;
	label_style.text.font = &arial_20;
	label_style.text.line_space = 10;

	lv_obj_t *label = lv_label_create(par, NULL);
	lv_label_set_text(label, title);
	lv_obj_set_style(label, &label_style);

	lv_label_set_long_mode(label, LV_LABEL_LONG_BREAK);
	lv_obj_set_width(label, w);
	return label;
}

lv_obj_t *wrapper_add_jack(lv_obj_t *wrapper, int jack_height)
{
	lv_obj_t *label = lv_label_create(wrapper, NULL);
	lv_label_set_text(label, "");
	lv_label_set_long_mode(label, LV_LABEL_LONG_DOT);

	lv_obj_t *jack = label;
	lv_page_glue_obj(jack, true);
	lv_obj_set_size(jack, lv_obj_get_width(wrapper), jack_height);
	return jack;
}

lv_obj_t *wrapper_add_obj(lv_obj_t *wrapper, lv_obj_t *obj, int left)
{
	lv_obj_t *item_wrapper = create_wrapper(wrapper, lv_obj_get_width(wrapper));
	lv_cont_set_style(item_wrapper, LV_CONT_STYLE_MAIN, lv_cont_get_style(wrapper, LV_CONT_STYLE_MAIN));
    lv_cont_set_fit4(item_wrapper, LV_FIT_NONE, LV_FIT_NONE, LV_FIT_TIGHT, LV_FIT_TIGHT);
	lv_cont_set_layout(item_wrapper, LV_LAYOUT_COL_L);
	lv_obj_set_parent(obj, item_wrapper);
	lv_obj_align(obj, item_wrapper, LV_ALIGN_IN_LEFT_MID, left, 0);
	return obj;
}

static void test_eff( lv_obj_t *par)
{
	//par = state->base;
	int w = lv_obj_get_width(par);

	bool use_text1 = true;
	uint32_t ticks1 = os_ticks_get();
	for (long i = 0; i < 500; i++) {
		lv_obj_t *bg = create_block_wrapper(par, w - 7*2);
		wrapper_add_jack(bg, 20);
		lv_obj_t *label1 = create_label(bg, w - 16*2, test_textg1_1);
		wrapper_add_obj(bg, label1, 9);
		wrapper_add_jack(bg, 24);
		lv_obj_t *label2 = create_label(bg, w - 16*2, test_textg1_2);
		wrapper_add_obj(bg, label2, 9);
		wrapper_add_jack(bg, 20);

		use_text1 = !use_text1;
		if (use_text1)
		{
			lv_label_set_text(label1, test_textg1_1);
			lv_label_set_text(label2, test_textg1_2);
		}
		else
		{
			lv_label_set_text(label1, test_textg2_1);
			lv_label_set_text(label2, test_textg2_2);
		}
	}
	uint32_t ticks2 = os_ticks_get();
	printf("[test]calc used:%d\n", ticks2 - ticks1);
}

/*
call test_eff as example entry.
void main()
{
    test_eff(obj);
   // stuf...
}
*/
```

kisvegabor, 2020-09-08:
Yes, the layout of v7 is slow for hundreds of objects. 

FYI, in v8 the layouts are rewritten, and special care was taken to make object creation fast. But it will be released in a few months so it doesn't help now.

A few things you can try/check to speed it up:
- be sure `LV_USE_DEBUG` is `0` in lv_conf.h.
- standard malloc ca be faster than LVGL's malloc. Try `LV_MEM_CUSTOM 1` in `lv_conf.h`
-  On `bg` set `fit` and `layout` after all children are added (it makes the layout recalculated only once)
- Instead of layout use "direct positioning" with `lv_obj_set_pos` where possible.