lvgl/lvgl #1737: Zooming Image + Setting Size doesn't work as intended

joshcamas, 2020-08-20:
**Describe the bug**

If I zoom in an image, setting size before or after has some very strange results, with the size not being the intended value. It looks like the size is scaled too, but not to scale of the scaling of image. In the example below, I scale the image up 2x, and set the size to 20x20. On screen, the image displays 60x60 of the original pixels (scaled of course).

I've also experimented with setting pivots and offsets, to no avail. I've also tried setting all of these values in different orders, and that didn't work either.

**To Reproduce**

```
    //Boring image creation
    lv_obj_t *img1 = lv_img_create(lv_scr_act(), NULL);
    lv_img_set_src(img1, &myimage);
    lv_img_set_auto_size(img1, true);
    lv_img_set_antialias(img1, false);
    
    //Zoom in double 
    lv_img_set_zoom(img1, 256*2);

    //Make size 20 x 20 pixels
    lv_obj_set_size(img1,21,21); 

    //Align on center
    lv_obj_align(img1, NULL, LV_ALIGN_CENTER, 0, 0);

```

**Expected behavior**

There should be a way to set the size of the image via its original width, and the scaling should not affect it. In other words, if I set my image size to 20x20 and then zoom it, the "slice" of the image visible should be the same "slice", just... well, zoomed in. Think of it as a crop, then zooming in on that crop.

**Screenshots**
I made some images representing what my arduino displays to explain the issue:

Here's the image with no zoom, no size:
[image]

Here's the image with no zoom, size set to 21:
[image]

Here's the image with zoom of 2, size set to 21 **(The bug!)**:
[image]

And here's the expected outcome with image with zoom of 2, size set to 21:
[image]
**Additional context**

kisvegabor, 2020-08-24:
Hi,

I've fixed it.

It can be used like this:
```c

    lv_obj_set_size(img, 20, 20);
    lv_img_set_zoom(img, 512);
    lv_img_set_pivot(img, 0, 0);  //To zoom from the left top corner
    lv_img_set_offset_x(img, -20); //Select the second image
```

kisvegabor, 2020-09-14:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2020-10-07 by kisvegabor