lvgl/lvgl #1738: repeatedly call the function of 'lv_img_set_src(obj, images[index]' simulator show with tear screen.

brilliantsong, 2020-08-20:
repeatedly call the function of 'lv_img_set_src(obj, images[index]' simulator show with tear screen.I fell what maybe multiply thread lead to conrupt,so the buffer can't clear before dirty-data.

kisvegabor, 2020-08-24:
Do you increment `index` or always set the same image?

brilliantsong, 2020-08-24:
Hi:
We use the increment index to show different picture. To be emulate the animation effect we must use the API of “lv_img_set_src”.
I can’t find any error when I checked the thread relative variable,Please tell me the reason when you solved!

Thank you very much!
Regards
hui

Caution: EXT Email

kisvegabor, 2020-08-25:
Most probably it happens because refreshing the framebuffer is not synchronized with VSYNC.

closed 2020-09-26 by stale[bot]

Xiaolin-He-private, 2020-09-28:
Has this issue been fixed, I find that set new image source with lv_img_set_src() won't clear the old image from display buffer, maybe we can add a new api to switch image source like lv_img_change_src().

kisvegabor, 2020-09-29:
It seems like a bug in `lv_img_set_src()`. Does it work well if you add `lv_obj_invalidate(img);` this line to the beginning of `lv_img_set_src`?