lvgl/lvgl #1739: Image doesn't display

zkoalexey, 2020-08-20:
lvgl vers 6.1.1
I have monochrome display.

Init Code

```
const lv_img_dsc_t logo = {
	.header.always_zero = 0,
	.header.w = 230,
	.header.h = 56,
	.data_size = 1632,
	.header.cf = LV_IMG_CF_INDEXED_1BIT,
	.data = logo_map,
};
LV_IMG_DECLARE(logo);

screen = lv_cont_create(NULL, NULL);
lv_disp_load_scr(screen);
lv_obj_set_style(screen, &lv_style_plain);

img1 = lv_img_create(screen, NULL);
lv_img_set_auto_size(img1, true);
lv_img_set_src(img1, &mImageDesc);
lv_obj_align(img1, screen, LV_ALIGN_CENTER, 0, 0);
```
Lvgl writes 'No Data' on display.

kisvegabor, 2020-08-20:
What is `mImageDesc` in `lv_img_set_src`? It seems it should be `&logo`.

zkoalexey, 2020-08-21:
It's just const reference, that was initialized by logo
`const lv_img_dsc_t &mImageDesc;`

kisvegabor, 2020-08-24:
`mImageDesc` should be a pointer not a reference:
`const lv_img_dsc_t * mImageDesc = &logo;`

zkoalexey, 2020-08-25:
The result is the same.
```
		img1 = lv_img_create(screen, NULL);
		lv_img_set_auto_size(img1, true);
		lv_img_set_src(img1, mImageDesc);
		lv_obj_align(img1, screen, LV_ALIGN_CENTER, 0, 0);

		return screen;
	}
	lv_obj_t * img1;
	const lv_img_dsc_t *mImageDesc;
```
I've checked with debugger this pointer - it points correctly to logo (I can see width and heigth)

zkoalexey, 2020-08-26:
I've found that function lv_img_decoder_open return LV_RES_INV

kisvegabor, 2020-08-26:
Please attach the `logo` image.

zkoalexey, 2020-08-27:
For example this logo. Result  is 'No data' too.
[image]

logo.c is in Archive. It was created with image convert tool.
[logo.zip](https://github.com/lvgl/lvgl/files/5135306/logo.zip)

kisvegabor, 2020-08-28:
I'm using the latest version of LVGL and with this it works well:
```c
  LV_IMG_DECLARE(logo);
  lv_obj_t * img = lv_img_create(lv_scr_act(), NULL);
  lv_img_set_src(img, &logo);
```

Please try using only this code without any other functions and UI elements.

zkoalexey, 2020-09-11:
I've found the reason:
#define LV_IMG_CF_INDEXED was 0. (it should be 1).
Thanks a lot for answers.

closed 2020-09-11 by zkoalexey