lvgl/lvgl #1742: img bug：Zoom in and the edges will be cut off

lilengqiang, 2020-08-21:
**Describe the bug**

When you zoom in an image, the edge will be cut off.

**To Reproduce**

static int zoom = 256;
static lv_obj_t * img;
static lv_obj_t * label;
static void btn_event_cb(lv_obj_t * btn, lv_event_t e)
{
    if(e == LV_EVENT_CLICKED){
        zoom++;
        char buf[64];
        sprintf(buf, "zoom : %d", zoom);
        lv_label_set_text(label, buf);

        lv_img_set_zoom(img, zoom);

    }
}

void test_zoom_in_img()
{
    img = lv_img_create(lv_scr_act(), NULL);
    LV_IMG_DECLARE(test_picture);

    lv_img_set_src(img,&test_picture);
    lv_obj_set_drag(img, true);

    lv_obj_t * btn = lv_btn_create(lv_scr_act(), NULL);
    lv_obj_set_event_cb(btn , btn_event_cb);

    label = lv_label_create(lv_scr_act(), NULL);
    char buf[64];
    sprintf(buf, "zoom : %d", zoom);
    lv_label_set_text(label, buf);
    lv_obj_align(label, btn , LV_ALIGN_OUT_RIGHT_MID, 100, 0);
    
}

**Expected behavior**

**Screenshots**
[image]
[image]

**Additional context**

kisvegabor, 2020-08-24:
Hi,

Thanks for the report. I've just fixed it the master branch.

lilengqiang, 2020-08-24:
@kisvegabor Thanks.

lilengqiang, 2020-08-24:
Bugs will still show up.

To Reproduce

    const uint16_t canvas_w = 400;
    const uint16_t canvas_h = 300;

    lv_draw_rect_dsc_t rect_dsc;
    lv_draw_rect_dsc_init(&rect_dsc);
    rect_dsc.border_color = LV_COLOR_BLACK;
    rect_dsc.border_width = 3;
    rect_dsc.border_opa   = LV_OPA_COVER;

    const int bufsize = LV_CANVAS_BUF_SIZE_TRUE_COLOR(canvas_w, canvas_h);
    static lv_color_t *cbuf = (lv_color_t *) malloc(bufsize);

    canvas = lv_canvas_create(lv_scr_act(), NULL);
    lv_obj_set_pos(canvas , 100 ,100);
    lv_canvas_set_buffer(canvas, cbuf, canvas_w, canvas_h, LV_IMG_CF_TRUE_COLOR);
    lv_canvas_fill_bg(canvas, LV_COLOR_WHITE, LV_OPA_COVER);

    lv_canvas_draw_rect(canvas, 0, 0, lv_obj_get_width(canvas), lv_obj_get_height(canvas), &rect_dsc);
    lv_img_set_pivot(canvas, 0, 0);
    lv_obj_set_drag(canvas, true);
    lv_img_set_zoom(canvas, 1000);

Screenshots

[image]

kisvegabor, 2020-08-24:
I pushed an update.

lilengqiang, 2020-08-24:
When I zoom in on larger multiples, this bug will still show up.
lv_img_set_zoom(canvas, 5000);

lilengqiang, 2020-08-24:
When I zoom in on larger multiples, this bug will still show up.

To Reproduce
const uint16_t canvas_w = 400;
const uint16_t canvas_h = 300;

lv_draw_rect_dsc_t rect_dsc;
lv_draw_rect_dsc_init(&rect_dsc);
rect_dsc.border_color = LV_COLOR_BLACK;
**rect_dsc.border_width = 1;**
rect_dsc.border_opa   = LV_OPA_COVER;

const int bufsize = LV_CANVAS_BUF_SIZE_TRUE_COLOR(canvas_w, canvas_h);
static lv_color_t *cbuf = (lv_color_t *) malloc(bufsize);

canvas = lv_canvas_create(lv_scr_act(), NULL);
lv_obj_set_pos(canvas , 100 ,100);
lv_canvas_set_buffer(canvas, cbuf, canvas_w, canvas_h, LV_IMG_CF_TRUE_COLOR);
lv_canvas_fill_bg(canvas, LV_COLOR_WHITE, LV_OPA_COVER);

lv_canvas_draw_rect(canvas, 0, 0, lv_obj_get_width(canvas), lv_obj_get_height(canvas), &rect_dsc);
lv_img_set_pivot(canvas, 0, 0);
lv_obj_set_drag(canvas, true);
**lv_img_set_zoom(canvas, 5000);**

kisvegabor, 2020-08-24:
It comes from a rounding error. I increased the precision to 1/16 which is enough for not extreme zoom values. If increasing the precision further we'd risk overflow for large images. 

IMO 1 px error on x20 zoom is acceptable. What is your use-case when such a large zoom and precision are required?

kisvegabor, 2020-08-24:
I've further improved the precision by applying some rounding. Not 100% perfect because the huge enlarged border is still a little bit smaller.

lilengqiang, 2020-08-25:
OK  Thanks.

lilengqiang, 2020-08-25:
@kisvegabor 
[image]

To Reproduce
    const uint16_t canvas_w = 400;
    const uint16_t canvas_h = 300;

    lv_draw_rect_dsc_t rect_dsc;
    lv_draw_rect_dsc_init(&rect_dsc);
    rect_dsc.border_color = LV_COLOR_BLACK;
    rect_dsc.border_width = 1;
    rect_dsc.border_opa   = LV_OPA_COVER;

    lv_draw_line_dsc_t line_dsc;
    lv_draw_line_dsc_init(&line_dsc);
    line_dsc.width = 1;
    line_dsc.color = LV_COLOR_BLUE;

    const int bufsize = LV_CANVAS_BUF_SIZE_TRUE_COLOR(canvas_w, canvas_h);
    static lv_color_t *cbuf = (lv_color_t *) malloc(bufsize);

    canvas = lv_canvas_create(lv_scr_act(), NULL);
    lv_obj_set_pos(canvas , 100 ,100);
    lv_canvas_set_buffer(canvas, cbuf, canvas_w, canvas_h, LV_IMG_CF_TRUE_COLOR);
    lv_canvas_fill_bg(canvas, LV_COLOR_WHITE, LV_OPA_COVER);

    lv_canvas_draw_rect(canvas, 0, 0, lv_obj_get_width(canvas), lv_obj_get_height(canvas), &rect_dsc);

    lv_point_t p[2];

    p[0].x = 0;
    p[0].y = 100;
    p[1].x = 100;
    p[1].y = 100;

    lv_canvas_draw_line(canvas, p, 2, &line_dsc);

    lv_img_set_pivot(canvas, 0, 0);
    lv_obj_set_drag(canvas, true);
    **lv_img_set_zoom(canvas, 172);**

kisvegabor, 2020-08-25:
I've fixed the "repeating content" issue but in some cases, to left or bottom pixels are missing.

I found the issue but it needs some reconsideration about the image transform algorithm and just a hotfix.

I've reopened this issue to show that there is a bug here but I can deal with it only a little bit later. (in a few weeks)

Until that, I suggest manually adding some transparent pixels on the edges. :slightly_frowning_face:

closed 2020-09-26 by stale[bot]

gilphilbert, 2020-12-01:
Has there been an update on this? I've just used zoom for the first time, and I'm getting the "extra" pixels (repeat) on my image. I can't really add extra transparent pixels to the image so I was wondering if there's a workaround I can use instead.

embeddedt, 2020-12-01:
How far are you trying to zoom in the image?

> I can't really add extra transparent pixels to the image

Normally, you could do this in a image editor like GIMP. Could you elaborate?

gilphilbert, 2020-12-01:
Oh, it's not that I can't do it (I certainly know how) it's more that my image size has to be pretty specific when it's not zoomed and a transparent border is going to mess things up for uses when the image is full sized.

embeddedt, 2020-12-01:
Yep; I see what's happening there.

Hmm.. how many sizes do you need? If it's just two or three you could consider converting in several different sizes and using the appropriate one where needed.

Another idea is to try downscaling - it may not suffer from the same issue.

gilphilbert, 2020-12-01:
> Hmm.. how many sizes do you need? If it's just two or three you could consider converting in several different sizes and using the appropriate one where needed.
Yeah, this is what I was thinking. And I don't always need them, so I could even exclude them from the build (MCU) when they're not needed. I was just hoping for a way to do this that didn't need me to generate every hand-drawn icon I have two times...

commit ba94b9cc (2020-08-24): fix croping of last column/row if an image is zoomed
Fixes #1742