lvgl/lvgl #1751: Cant display labels with Chinese charactors

sk123sos, 2020-08-25:
labels with chinese charactors will not be displayed or displayed strange.

**To Reproduce**
`
    static lv_style_t style1;
    lv_style_init(&style1);
    lv_style_set_text_font(&style1, LV_STATE_DEFAULT, &ch_font);
    lv_obj_t* label1 = lv_label_create(lv_scr_act(),NULL);
    lv_label_set_text(label1,"一二三四五六七八九十一二三四五六七八九十");
    lv_obj_t* label2 = lv_label_create(lv_scr_act(), label1);
    lv_obj_t* label3 = lv_label_create(lv_scr_act(), label1);
    lv_obj_add_style(label1,LV_LABEL_PART_MAIN,&style1);
    lv_obj_add_style(label2, LV_LABEL_PART_MAIN, &style1);
    lv_obj_add_style(label3, LV_LABEL_PART_MAIN, &style1);
    lv_label_set_long_mode(label1,LV_LABEL_LONG_EXPAND);
    lv_label_set_long_mode(label2, LV_LABEL_LONG_BREAK);
    lv_label_set_long_mode(label3, LV_LABEL_LONG_CROP);
`

**Expected behavior**

labels should have been displayed  in three long modes.

**Screenshots**
I cant offer a screen shot because it show nothing...

embeddedt, 2020-08-25:
Check if your C source file is using the UTF-8 encoding; other encodings has caused issues in the past.

sk123sos, 2020-08-26:
> Check if your C source file is using the UTF-8 encoding; other encodings has caused issues in the past.

Thanks for reply,I have checked, it's UTF-8 but it still not work.

kisvegabor, 2020-08-27:
Just add these lines after your code snippet:
```
  lv_obj_set_y(label2, 50);
  lv_obj_set_width(label2, 100);

  lv_obj_set_y(label3, 200);
  lv_obj_set_width(label3, 100);
```

Reason: if you copy an "expended "label the copied label's size will be equal to the text size therefore there is nothing to crop or break.
But if you set a smaller width the text won't fit into the object's size and break and crop can do something.

closed 2020-09-26 by stale[bot]

lxa520, 2020-11-09:
i have same question. Are you resolved?