lvgl/lvgl #1752: Kconfig usage

kisvegabor, 2020-08-25:
Hi,

For a few months, we are considering adding KConfig support to LVGL. However there some questions:

### Keep `lv_conf_template.h` and Kconfig synchronized
@embeddedt suggested maintaining only the KConfig file and generate `lv_conf_template.h` from it. As Kconfig has a well-defined structure it's easy to parse but not every LVGL config can be described by KConfig. E.g. `typedef`s.

So the other way (lv_conf_template -> Kconfig) seems reasonable too with a proper markup in `lv_conf_template.h`.

### Work with or without lv_conf.h and Kconfig

Any of `lv_conf.h` (in Kcofig project) or Kconfig (in case of "normal" projects) can be missing. Mixing the two also should be possible. All these can be done in [lv_conf_internal.h](https://github.com/lvgl/lvgl/blob/master/src/lv_conf_internal.h). For example, checking `LV_USE_BTN` looks this now:
```c
#ifndef LV_USE_BTN
#  define LV_USE_BTN      1
#endif
```

Instead, I suggest:
```c
#ifndef LV_USE_BTN               /*Only if not defined in lv_conf.h*/
#  ifdef CONFIG_LV_USE_BTN    /*Use KConfig value if set*/
#    define LV_USE_BTN  CONFIG_LV_USE_BTN
#  else
#    define LV_USE_BTN      1      /*Use default value*/
#  endif
#endif
```

So `lv_conf.h` has the highest priority, if it's not set, try Kconfig, if not set use default value.

Using  `lv_conf.h` is not necessary at all when Kconfig is present. However, LVGL [wants to included](https://github.com/lvgl/lvgl/blob/master/src/lv_conf_internal.h#L20) it somehow. We can add an option like `LV_CONF_SKIP` or `CONFIG_LV_CONF_IGONRE`. If it's defined do not include `lv_conf.h` but define non-Kconfig compatible values to a default value. E.g. `typedef int16_t  lv_coord_t;`

What do you think?

cc @C47D @BFFonseca

acassis, 2020-08-26:
Hi @kisvegabor I think this approach should work and will make the integration with foreign projects based on Kconfig (i.e. NuttX, ESP-IDF, Zephyr, etc) more transparent. +1

C47D, 2020-08-27:
Hi,

Current `lv_conf.h` (being used in [lv_port_esp32](https://github.com/lvgl/lv_port_esp32) is available [here](https://github.com/lvgl/lv_port_esp32/blob/master/components/lvgl/lv_conf.h), it's based on LVGL v7.0.1). Currently we have two "issues" with it:

This part of the `lv_config` code will be fixed (see [Rotate display using the lv_disp_drv_t rotated field](https://github.com/lvgl/lv_port_esp32/issues/174)):
```
/* Maximal horizontal and vertical resolution to support by the library.*/
#if defined (CONFIG_LVGL_PREDEFINED_DISPLAY_NONE) && defined (CONFIG_LVGL_DISPLAY_ORIENTATION_PORTRAIT)
    #define LV_VER_RES_MAX          (CONFIG_LVGL_DISPLAY_WIDTH)
    #define LV_HOR_RES_MAX          (CONFIG_LVGL_DISPLAY_HEIGHT)
#elif defined (CONFIG_LVGL_PREDEFINED_DISPLAY_NONE) && defined (CONFIG_LVGL_DISPLAY_ORIENTATION_PORTRAIT_INVERTED)
    #define LV_VER_RES_MAX          (CONFIG_LVGL_DISPLAY_WIDTH)
    #define LV_HOR_RES_MAX          (CONFIG_LVGL_DISPLAY_HEIGHT)
#elif defined (CONFIG_LVGL_PREDEFINED_DISPLAY_NONE) && defined (CONFIG_LVGL_DISPLAY_ORIENTATION_LANDSCAPE)
    #define LV_HOR_RES_MAX          (CONFIG_LVGL_DISPLAY_WIDTH)
    #define LV_VER_RES_MAX          (CONFIG_LVGL_DISPLAY_HEIGHT)
#elif defined (CONFIG_LVGL_PREDEFINED_DISPLAY_NONE) && defined (CONFIG_LVGL_DISPLAY_ORIENTATION_LANDSCAPE_INVERTED)
    #define LV_HOR_RES_MAX          (CONFIG_LVGL_DISPLAY_WIDTH)
    #define LV_VER_RES_MAX          (CONFIG_LVGL_DISPLAY_HEIGHT)
#else
    #define LV_HOR_RES_MAX          (CONFIG_LVGL_DISPLAY_WIDTH)
    #define LV_VER_RES_MAX          (CONFIG_LVGL_DISPLAY_HEIGHT)
#endif
```

Also is worth seeing this another issue: [Some LVGL configuration options are not available on the menuconfig](https://github.com/lvgl/lv_port_esp32/issues/178), such as:

```c
/* Type of coordinates. Should be `int16_t` (or `int32_t` for extreme cases) */
typedef int16_t lv_coord_t;
```

and

```c
/*=====================
 *  Compiler settings
 *====================*/
/* Define a custom attribute to `lv_tick_inc` function */
#define LV_ATTRIBUTE_TICK_INC

/* Define a custom attribute to `lv_task_handler` function */
#define LV_ATTRIBUTE_TASK_HANDLER

/* Define a custom attribute to `lv_disp_flush_ready` function */
#define LV_ATTRIBUTE_FLUSH_READY

/* With size optimization (-Os) the compiler might not align data to
 * 4 or 8 byte boundary. This alignment will be explicitly applied where needed.
 * E.g. __attribute__((aligned(4))) */
#define LV_ATTRIBUTE_MEM_ALIGN

/* Attribute to mark large constant arrays for example
 * font's bitmaps */
#define LV_ATTRIBUTE_LARGE_CONST

/* Prefix performance critical functions to place them into a faster memory (e.g RAM)
 * Uses 15-20 kB extra memory */
#define LV_ATTRIBUTE_FAST_MEM
```

This is because the option types available on Kconfig.

For the LV_ATTRIBUTE_x options we can do something like this, so the users can define them somewhere else:
```c
/*=====================
 *  Compiler settings
 *====================*/
/* Define a custom attribute to `lv_tick_inc` function */
#ifndef LV_ATTRIBUTE_TICK_INC
#define LV_ATTRIBUTE_TICK_INC
#endif
```

guoweilkd, 2020-08-27:
If I want to use LVGL's more advanced features, I need to configure LV_conf.h after configuring Kconfig? like `lv_obj_user_data_t`:
```
#if LV_USE_USER_DATA
/*Declare the type of the user data of object (can be e.g. `void *`, `int`, `struct`)*/
typedef void * lv_obj_user_data_t;
#endif
```

kisvegabor, 2020-08-27:
@C47D 
> This part of the lv_config code will be fixed (see Rotate display using the lv_disp_drv_t rotated field):

Great! Finally, we should have only two configs for it: `CONFIG_LV_HOR_RES_MAX` and `CONFIG_LV_VER_RES_MAX`

> typedef int16_t lv_coord_t;

That's a pity that it can't set in Kconfig.h. :slightly_frowning_face: 
By default we can run LVGL without lv_conf on some systems (e.g. ESP). In this case, the typedefs will have default values (I think it's suitable in most of the cases) If the user needs something specific, he can add `lv_conf.h` somewhere (even next to main.c) and apply the required configs.

> For the LV_ATTRIBUTE_x options we can do something like this, so the users can define them somewhere else:

It can work with the "define structure" I proposed in the first post. It's first line checks if the `define` already set. If so it ignores other config sources.

@guoweilkd 
If you need to modify the typedefs then you need to use lv_conf.h too. But it can be almost empty. You just need to add the typedefs.

kisvegabor, 2020-09-08:
I've updated [lv_conf_internal.h](https://github.com/lvgl/lvgl/blob/dev/src/lv_conf_internal.h#L97) to work with Kconfig.

It checks `LV_CONF_SKIP` or `CONFIG_LV_CONF_SKIP` is used to not include `lv_conf.h` just add the default typedefs. See [here](https://github.com/lvgl/lvgl/blob/dev/src/lv_conf_internal.h#L1903).

kisvegabor, 2020-10-02:
Not stale.

puzrin, 2020-10-24:
I like `lv_conf_internal.h`. In PIO i'll be able to define vars in `plaformio.ini`, that's cool.

C47D, 2020-10-24:
Kconfig working in v7.1, I need to get it in sync with the current master, I saw some NXP GPU related configs.

kisvegabor, 2020-10-26:
> In PIO i'll be able to define vars in plaformio.ini, that's cool.

Yes, confirmed. :slightly_smiling_face: 

> Kconfig working in v7.1, I need to get it in sync with the current master, I saw some NXP GPU related configs.

Awesome! :clap: :clap: :clap:

kisvegabor, 2020-11-23:
The KConfig file is added to the `master` branch.

If you find any issues with it, please open a new issue.

closed 2020-11-23 by kisvegabor