lvgl/lvgl #1753: Regulal meetings

kisvegabor, 2020-08-25:
Hi,

We had the first LVGL video meeting with @C47D and @BFFonseca and we were talking about the integration of LVGL as ESP component.

It was great to see people around LVGL face to face.  :slightly_smiling_face: 
We were also talking about it would be great to have regular monthly or every 2 week meetings. So we scheduled the next one to 8 September. 

Until that, a few things should turn out about the LVGL-ESP integration, and hopefully, we can move forward with this topic. Independently we can discuss other topics too. So if you are interested, feel free to comment here, send me your email address (kisvegabor@gmail.com), and suggest a topic. 

It'd be great to know you better, so feel free to join :slightly_smiling_face:

C47D, 2020-08-27:
It was great to meet you two as well, looking forward for the next meeting!

excitedbox, 2020-08-27:
Oh that is super. Where are these meetings held? Will they be open to everyone?

Blender is expanding their community engagement right now although they have done this for a while. Basically their meetings are not public obviously but the community manager does a weekly youtube stream to show new features as they are developing them and kind of show the direction blender will be going in and also works as a direct way to get feedback from the community and keep them updated.

BFFonseca, 2020-08-27:
It was great seeing and hearing you rather than just reading your comments.

Would be great to have more developers participate in these meetings and have more opinions and ideas!

> Oh that is super. Where are these meetings held? Will they be open to everyone?

We did a video-call through Google Meet. I think the idea is to include the LVGL dev community in them, but that's for @kisvegabor to decide.

pete-pjb, 2020-08-27:
Hi @kisvegabor ,

If I am available at your meeting times, I would be happy to join in.  I think you have my email address already.

Kind Regards,

Pete

kisvegabor, 2020-08-28:
@excitedbox 
>  but the community manager does a weekly youtube stream to show new features as they are developing them

That's sounds interesting. Weekly announcements seem to frequent for us but monthly blog post seems doable. (I'm afraid my English is not good enough to speak on youtube. :laughing: )

> Would be great to have more developers participate in these meetings and have more opinions and ideas!

Exactly. Not open for literally everybody but if you develop something in LVGL than it'd be great to see there and hear your feedback an opinion. :slightly_smiling_face: If there will be too many people there we will figure out something how to limit the number of participants to a reasonable number. 

I've created the next event and sent an invitation to @BFFonseca @C47D and @pete-pjb

turoksama, 2020-08-28:
Hi @kisvegabor ,

> Hi @kisvegabor ,
> 
> If I am available at your meeting times, I would be happy to join in. I think you have my email address already.
> 
> Kind Regards,
> 
> Pete

I would like to join the meeting too.

excitedbox, 2020-08-30:
I am a native English and German speaker but don't like public speaking :)

I agree weekly would be too much. Blender is a huge project so they have tons of things to show off and the stream is usually multiple hours long.

I am just figuring out the last details of my API suggestion (can't decide whether if it is better to have states with style definitions or styles with states). I think it will be good to have a chat once I post what I came up with so I can clarify any questions and get feedback on improvements. I got swamped with work the last few days doing security upgrades to our servers.

kisvegabor, 2020-08-31:
@turoksama 
> I would like to join the meeting too.

Great! Shall I use the email address shown in your profile?

@excitedbox 
Would you like to join as well to show the API suggestions?

turoksama, 2020-08-31:
@kisvegabor 
Thanks, turoksama@hotmail.com is better. 
I don't know how to join you without Google meeting, we usually use ding or wechat in our country…

kisvegabor, 2020-08-31:
@turoksama I sent the invitation
Google meets runs in your browser. So it should simply work.

turoksama, 2020-08-31:
Great👍

excitedbox, 2020-09-01:
I sent you an email. spam bots scrape the web so I don't post it publicly.

tjstyle, 2020-09-07:
Still not have any experience to fully contribute to this project, but I still follow and use lvgl.
please let me know if I allowed to participate in the meeting.
No worries if I'm not allowed to enter the meeting, my English is also not good ;-)

kisvegabor, 2020-09-08:
@tjstyle 
Sure, it'd be great to get to know each other.
Later we can make more thematic meetings were only the affected people present, so no worries.
I sent you an invitation.

kisvegabor, 2020-09-08:
On today's meeting, we have further discussed the LVGL-ESP component question and some other minor topics. 
And assigned the following tasks:
- @C47D finalize ESP files (Kconfig, CMAkelist, component.mk)
- @pete-pjb Write FreeRTOS and Xilinx tutorial to the docs site
- @kisvegabor Update LVGL to work with Kconfig
- @kisvegabor Add radio button to the [roadmap](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md). (Done)

tjstyle, 2020-09-08:
> @tjstyle
> Sure, it'd be great to get to know each other.
> Later we can make more thematic meetings were only the affected people present, so no worries.
> I sent you an invitation.

Just seeing the invitation in bulk folder, sorry I miss today’s meeting :-(

kisvegabor, 2020-09-09:
> Just seeing the invitation in bulk folder, sorry I miss today’s meeting :-(

No problem :slightly_smiling_face:

kisvegabor, 2020-10-02:
During the last meeting, @C47D and I agreed on to add the Kconfig and CMakeList.txt to the lvgl repo. (Carlos will send a PR after some tests). So we are really close to having the ESP component directly from the lvgl repo. :slightly_smiling_face:  

If it works well the similar files will be added to the lv_examples repo too.

We were also talking about that `LV_HOR/VER_RES_MAX` can be removed from `lv_conf.h` because these max values are not really used internally. The problem with having max values is that if a specific board is selected in driver repo, it needs to be sure LVGL's max resolutions are enough for this board. Without these limits during board initialization, any resolution can be set. 

To maintain compatibility, these define can remain in `lv_conf.h` in v7, but in v8 they can be replaced with `LV_HOR/VER_RES_DEF` or simply removed.

The next meeting is scheduled for 20 October 2 pm (UTC+2). Let me know if you would like to join.

Feel free to comment here if you have any remarks.

BFFonseca, 2020-10-02:
Would love to be a part of that meeting! 
Really excited for the component

kisvegabor, 2020-10-05:
@BFFonseca Cool! :)  I sent an invitation to you and Carlos.

kisvegabor, 2020-11-02:
I forgot to write that the next meeting is planned for 10 November, 12:00 (UTC+1)

Last time we are talking about the Kconfig integration and discussed some feedback on LVGL.

If you are interested, feel free to drop a line here. We can discuss any ideas, remarks, suggestions, feedback you have.

C47D, 2020-11-02:
I need to check what time is in Mexico, we recently move our clocks one hour back.

kisvegabor, 2020-11-02:
We did as well. It's the same time as before.

C47D, 2020-11-03:
Hi,
I won't be available on November 10, can we reschedule it for November 9?

kisvegabor, 2020-11-04:
Ok for me.

C47D, 2020-11-05:
Did you updated the meeting date at google meets?

kisvegabor, 2020-11-08:
Updated

kisvegabor, 2020-12-07:
The main target of the meetings was the Kconfig integration and it's successfully done. As currently there is no proposed topic for the meetings I close this issue. 

Anyway, if something comes, I'd be happy to talk about it in "live". So feel free to comment here or open a new issue for topics.

closed 2020-12-07 by kisvegabor