lvgl/lvgl #1754: Bug in function _lv_txt_ins

zhwxChos, 2020-08-26:
**Describe the bug**

There is no non-zero judgment in this loop body, which leads to a HardFault

[image]

**To Reproduce**

 txt is illegal content,when using void lv_textarea_set_text(lv_obj_t * ta, const char * txt);
[image]

**Expected behavior**
show the illegal content or no content.

Does not cause HardFault.

**Screenshots**
[image]

**Additional context**

Add any other context about the problem here.

kisvegabor, 2020-08-26:
I don't understand. Can you provide some example parameters to see when the issue happens?

zhwxChaos, 2020-08-26:
@kisvegabor 
The problem site has been destroyed。。。I want textareax to display the value obtained in eeprom. 

size_t is unsigned int . 
When i is 0, i is decremented, i becomes very large, and txt_buf is out of range。
You can find them in the last screenshort.

I think the call stack display is very clear。

The problem code is
	lv_obj_t* par = lv_scr_act();
	lv_obj_t* t_ta = lv_textarea_create(par, NULL);
	lv_textarea_set_text(t_ta,buf);

the buf addr is 0x20027170 in second screenshort.

kisvegabor, 2020-08-26:
I see, simple using `int32_t i` would solve the issue?

pete-pjb, 2020-08-26:
If buf is being loaded from EEPROM are you terminating the buffer with a NULL?  If not this will cause all sorts of problems!

embeddedt, 2020-08-26:
@kisvegabor  I don't think making `i` a signed integer will help, as it's used to index into a buffer, which means that it should never be negative in the first place.

zhwxChaos, 2020-08-26:
@kisvegabor I simply added a line of code to circumvent this problem, and then wait for you to fix it。
[image]

@pete-pjb I cannot check all the values obtained from eeprom, and add \0 to their last byte
And  You can't stop someone from doing this   _lv_txt_ins("",0,"");    ^_^ 

@embeddedt Should this be done? Or other ways 
size_t i ;
for( i = a ; i >= b; )
{
    ..//do something
   if(i > 0)
   {
       i--;
   }
}

embeddedt, 2020-08-26:
>  I cannot check all the values obtained from eeprom, and add \0 to their last byte

All C strings need `\0` as their last byte, otherwise disaster strikes. If you don't have `\0` as the last byte, you can't pass it to a function expecting a C string and expect things to work. `\0` is used to determine the length of the string.

> I simply added a line of code to circumvent this problem, and then wait for you to fix it

However, we should definitely bail out of that function early if the `ins_txt` string is determined to have 0 length.

zhwxChaos, 2020-08-26:
@kisvegabor using this code ,then ins_txt is "".

	char buf[5];
	buf[0] = 0xE9;
	buf[1] = ' ';
	buf[2] = ' ';
	buf[3] = 0xE9;
	buf[4] = '\0';
	lv_textarea_set_text(t_ta,buf);

pete-pjb, 2020-08-26:
Hi @zhwxChaos ,

With all due respect the function ```lv_textarea_set_text()``` like all C functions which deal with text strings must be called with a NULL terminated buffer, this is mandatory for the C language.

If you wish to display the contents of an EEPROM in a text box you must load the buffer and terminate it with a NULL, if you fail to terminate the buffer things may appear to be okay but at some point you program is very likely to crash.  

You can do something along the lines of:
```c
void load_buffer_from_eeprom_as_string( uint32_t eeprom_size, char *buf ) {

    your_read_eeprom( buf, eeprom_size );
    *(buf+eeprom_size) = '\0'; 

}
```

The buffer provided needs to be at least eeprom_size+1 long.

Another option would be to read the eeprom byte by byte in a loop and call the ```void lv_textarea_add_char(lv_obj_t * ta, uint32_t c)``` function to add them 1 at a time but this is clearly much less efficient.

I suspect the issue you are seeing is caused by failing to terminate the buffer with a NULL and the string length functions are returning erroneous numbers.

I agree with @embeddedt it wouldn't hurt to do the check and bail the function if a zero length is detected but under normal circumstances what we are seeing here would probably not happen IMHO.

I hope this helps.

Kind Regards,

Pete

zhwxChaos, 2020-08-26:
@pete-pjb 
Thank you so much for replying to me,and  I know everything you say. 
You can try my code above.

pete-pjb, 2020-08-26:
@zhwxChaos 

I assume the contents of the EEPROM are not text on the basis of what you are showing here.  This will also break the character handling routines.  In my experience when dumping the contents of an EEPROM you might choose to show the contents as hexadecimal maybe rather than ASCII?  Or maybe create a routine to process the buffer before it is passed to the text area to filter out the non-printable characters?

A text area can not display non-printable characters so you need to do some kind of post processing after the EEPROM read to format the text so it can be displayed in the text area.

I hope that makes sense.

Kind Regards,

Pete

zhwxChaos, 2020-08-26:
@pete-pjb how to  filter out the non-printable characters?

pete-pjb, 2020-08-26:
Hi @zhwxChaos ,

Assuming you read EEPROM into a ```char``` buffer and NULL terminate it you can use the standard c library function ```isprint()``` so applying this to your example you could do something like this:

```c
  lv_obj_t *t_ta;

  t_ta=lv_textarea_create(lv_scr_act(), NULL);
  char buf[5];
  buf[0] = 0xE9;
  buf[1] = ' ';
  buf[2] = ' ';
  buf[3] = 0xE9;
  buf[4] = '\0';
  lv_textarea_set_text( t_ta, "" );
  for(int i=0; i<strlen(buf); i++)  if( isprint(buf[i]) ) lv_textarea_add_char(t_ta, buf[i]);
```

Hope that helps.

Kind Regards,

Pete

kisvegabor, 2020-08-27:
Ah got it.
Just to confirm. Let's say to `for` loop has this values:
```c
for(i = 0, i >= 0 + 0; i --)
```
At first iteration `i = 0` and `i >= 0` so run the body. But after that `i--` makes `i = 0xffffffff` and `0xffffffff >= 0` so run the body again...

I agree to return if `ins_len == 0`.

@embeddedt, @pete-pjb ?

embeddedt, 2020-08-27:
@kisvegabor Looks like both of us agree.

kisvegabor, 2020-08-28:
I've fixed it!

closed 2020-08-28 by kisvegabor

zhwxChaos, 2020-08-28:
> I've fixed it!

QQQ

commit adf65b67 (2020-08-28): Fix _lv_txt_ins with zero length insert text
Closes #1754