lvgl/lvgl #1755: lv_keyboard wrong input

incity, 2020-08-28:
when click the text_area which locates at the bottom of screen, popup a keyboard widget, but the keyboard also been clicked.

embeddedt, 2020-08-28:
Please provide a code sample for testing.

turoksama, 2020-08-28:
I think i know what he is talking about.
It is too "sensitive" for your touch device, as far as i know, may 2 ways help:
1.  make indev read period longer -->20? 30? test with it.
```
/* Input device read period in milliseconds */
#define LV_INDEV_DEF_READ_PERIOD          10
```
2. wait indev_is_released after clicked.

kisvegabor, 2020-08-28:
I think it's something else. Try using `lv_indev_wait_release(lv_indev_act(())` when you open the keyboard in the event callback.

incity, 2020-08-28:
ok thanks solved

lv_indev_wait_release(lv_indev_get_act(())

closed 2020-08-28 by incity