lvgl/lvgl #1756: Moving the HID mouse cursor will blur the picture

zhwxChaos, 2020-08-28:
My STM32 devices supports mouse and displays cursor. 

But when the cursor moves on the imageBtn picture, it will produce smears, and the picture looks blurred.

STM32F429,  HID Mouse, littleVGL  V7.3.0

[391d9d52890ab22e5f8bfc9260784f62.zip](https://github.com/lvgl/lvgl/files/5140358/391d9d52890ab22e5f8bfc9260784f62.zip)

kisvegabor, 2020-08-28:
Can you reproduce it in a simulator too?

zhwxChaos, 2020-08-28:
> Can you reproduce it in a simulator too?

In the simulator, I don’t know how to display the cursor.

embeddedt, 2020-08-28:
The Eclipse simulator does it [out of the box](https://github.com/lvgl/lv_sim_eclipse_sdl/blob/28f8ed4e60e44cba63cfb2302550a0b7074e8cff/main.c#L108-L112); for any other simulator, just copy the implementation and image file.

turoksama, 2020-08-28:
FYI, please check if there is a lv_obj_invalidate(obj) called in VALUE_CHANGE event?

zhwxChaos, 2020-08-31:
@kisvegabor 
I can not reproduce it in a simulator.
Is the screen refresh rate in the STM32  too low?

kisvegabor, 2020-08-31:
Another issue also implies that there is some problem with image invalidation. Please try this:
https://github.com/lvgl/lvgl/issues/1724#issuecomment-683668898

turoksama, 2020-08-31:
@kisvegabor 
I noticed that some frames of animations will be lost when using true double buffer, and it's fine with one single 1/10 lines of VDB. (Version 6. 1, due to production ready reasons, not update to the latest) 
May it be in rendering free run in the background when changing buffer's address, and it's too late to show animating contents? 
but digit fonts of changing values are fine.

kisvegabor, 2020-09-01:
@turoksama 
True double buffering can be slower than single 1/10 VDB because of a lot of access to the slower external SRAM. The animations are aware of the elapsed time and calculate "where should I be at this moment?". So if an animation should last for 1 sec, it will last for 1 sec be may contain 5 or 20 frames depending on the performance.

turoksama, 2020-09-01:
Exactly.
Well, looks like lack of performance mainly.

zhwxChaos, 2020-09-01:
@kisvegabor 
I am using single buffer now. 
I think, the function of disp_drv.flush_cb is to copy color_buf to display buf. 

Can I directly pass the display buf as color_buf to the function lv_disp_buf_init?       
And In disp_drv.flush_cb, I only need to start DMA2D. This can reduce the memory and reduce the time of  memory copy .
Is this feasible?

kisvegabor, 2020-09-02:
> Can I directly pass the display buf as color_buf to the function lv_disp_buf_init?

I don't understand exactly. Can you show an example?

> And In disp_drv.flush_cb, I only need to start DMA2D. 

You can use DMA2D, but to find the issue, it'd be better to use a very simple `set_px()`-like function to be sure that part work well.

zhwxChaos, 2020-09-02:
@kisvegabor The simplified code is ,

`#define COLOR_BUF_SIZE		(LV_HOR_RES_MAX*LV_VER_RES_MAX)
static lv_color_t color_buf[COLOR_BUF_SIZE]	__attribute__((at(LCD_FRAME_BUF_ADDR))); 

in fuction LTDC_init,
void LTDC_init()
{
	LTDC_LayerCfgTypeDef pLayerCfg;
	...
	pLayerCfg.FBStartAdress = color_buf
	...
}

in fuction lv_port_disp_init
void lv_port_disp_init(void)
{
	static lv_disp_buf_t disp_buf;
	lv_disp_buf_init(&disp_buf, color_buf, NULL, COLOR_BUF_SIZE); 
	...
}

in fuction disp_flush
static void disp_flush(lv_disp_drv_t * disp_drv, const lv_area_t * area, lv_color_t * color_p)
{
	//calc color Addr and start DMA2D
        DMA2D->OMAR = addr;
        DMA2D->NLR = XXX;
	start_DMA2D(addr,offline);

	lv_disp_flush_ready(disp_drv);
}
`

kisvegabor, 2020-09-03:
Is` start_DMA2D();` blocking?
If not you need to call `lv_disp_flush_ready` when the DMA is ready.

zhwxChaos, 2020-09-03:
Block,   Is this plan feasible?

kisvegabor, 2020-09-04:
Ah see it now. The issue is here.
```
pLayerCfg.FBStartAdress = color_buf;
```

You can't use the same buffer. You need a separate screen-sized buffer as the frame buffer.

closed 2020-09-23 by kisvegabor