lvgl/lvgl #1757: Discussion of testing

no1wudi, 2020-08-28:
Hi, 
We need to improve the test coverage since only a few API were coverd now.

And for GUI library, you can't just check the return value of it's API, the image drawn on the screen is also very important.

@kisvegabor Did you have any idea or plan for it ?

embeddedt, 2020-08-28:
Related: https://github.com/lvgl/lvgl/issues/1658

This is on the [roadmap](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md#ideas), but not planned for a specific version yet.

no1wudi, 2020-08-28:
Thanks, is there any suggestion or design document for testing ?
I'm looking for a way to test LVGL (drawn image).

embeddedt, 2020-08-28:
* [Another prior discussion](https://github.com/lvgl/lvgl/issues/1243#issuecomment-569441164)
* [Blog post about taking screenshots and testing them](https://blog.lvgl.io/2019-04-25/snapshots) (this is slightly outdated but the principle still applies)

excitedbox, 2020-08-30:
How complete/up to date is the benchmarking and stressing example? What if we try and maintain a version with calls to all APIs and then we can try running it on a few systems whenever there is a release? Having a test that is always the same will also make it easier to catch sudden failures. I have an ESP32 and 3.5" display that I am not using and could leave setup for testing. That way I can flash the RC to it and play around with it to make sure things display correctly.

puzrin, 2020-08-31:
> How complete/up to date is the benchmarking and stressing example?

From my point of view, usability of current benchmarks is far from perfect. No automation, to many efforts to execute, no testing of each component, no history tracking, and so on.

https://github.com/lvgl/lvgl/issues/1660

I did proposal, but it's blocked by lack of tests (could use ~ the same tools).

> What if we try and maintain a version with calls to all APIs and then we can try running it on a few systems whenever there is a release? Having a test that is always the same will also make it easier to catch sudden failures.

IMO usually regressions are not platform-specific. Software testing via gem5 will be much more easy.

kisvegabor, 2020-08-31:
Hi,

I agree that improving test coverage would be important. The current test framework is created by me and probably it'd be better to use a well know one. 

In the current framework, there is an assert for screenshots. I've just checked and updated it. See [here](https://github.com/lvgl/lvgl/blob/91cfb708560515596187723a6c4bf80e7fa9855b/tests/lv_test_widgets/lv_test_label.c#L67).
As you can see it handles taking a screenshot and compares it with a [PNG reference image](https://github.com/lvgl/lvgl/blob/master/tests/lv_test_ref_imgs/lv_test_img32_label_1.png).

As @puzrin  mentioned benchmarking with gem5 seems like a good direction. I definitely agree to set up a gem% environment and run some stress tests automatically.

kisvegabor, 2020-09-23:
@no1wudi Is there any news from your side in this topic?

no1wudi, 2020-09-23:
> @no1wudi Is there any news from your side in this topic?

Nothing new, we are working on CSS3 compatibility currently.
As there are many challenges in testing, we decided to test final application based on LVGL currently.

kisvegabor, 2020-09-24:
I see. When v8 is released testing will get a higher priority. So at the beginning of next year, we will have much more tests. 

I close this issue then.

closed 2020-09-24 by kisvegabor