lvgl/lvgl #1760: Roller widget crashes when LV_USE_GROUP = 0

Alex-I-B, 2020-08-30:
**Describe the bug**
``
If LV_USE_GROUP is set to 0, demo examples with Roller widget (lv_ex_roller_1, lv_demo_widgets) crash.

**To Reproduce**
Set in lv_conf.h 
`#define LV_USE_GROUP            0`
and start e.g. lv_ex_roller_1() in a PC simulator (QT Creator/Linux). GUI widget does not appear. Similar situation on STM32 Nucleo F303RE board with ILI9486 driven TFT display.
`
  lv_init();

  /*Initialize the HAL (display, input devices, tick) for LVGL*/
  hal_init();

  lv_ex_roller_1();
`
**Expected behavior**

Roller widget is visible and operable.

**Screenshots**
n.a.

**Additional context**
Presumably disabled LV_USE_GROUP blocks at the line 589 of the lv_roller.c backpropagation of the signals to the ancestors. However this is necessary for LV_SIGNAL_CHILD_CHG (to automatically move newly added child to the page's scrollable container) and LV_SIGNAL_GET_TYPE signals.

kisvegabor, 2020-08-31:
Thank you for the report.
I commented on the PR.

kisvegabor, 2020-09-01:
The PR is merged so I close this issue.

closed 2020-09-01 by kisvegabor