lvgl/lvgl #1762: multiline textarea raise unexpected LV_EVENT_DEFOCUSED event

incity, 2020-08-31:
[image]

[image]

when click area 2 in the pic will raise a LV_EVENT_DEFOCUSED event

kisvegabor, 2020-08-31:
As far as I remember it's already solved. Do you use the latest version of LVGL?

incity, 2020-09-01:
yes, has pull the lastest version of lvgl and lvgl examples

kisvegabor, 2020-09-01:
And is it working now or you already used it?

incity, 2020-09-02:
it's not sloved

incity, 2020-09-03:
[image]

when click area 2 print LV_EVENT_DEFOCUSED and LV_EVENT_FOCUSED at the same time as follows:
[image]

kisvegabor, 2020-09-04:
I pushed a fix, it should work well now.

closed 2020-09-04 by kisvegabor

incity, 2020-09-05:
@kisvegabor still has problem

incity, 2020-09-05:
and for single line textarea, there is no LV_EVENT_FOCUSED  now

incity, 2020-09-05:
[image]

kisvegabor, 2020-09-07:
Thanks for the feedback!
I've pushed a fix.

commit 804c3c79 (2020-09-04): fix unexpeted DEFOCUS on lv_page
Fixes #1762