lvgl/lvgl #1763: [v8] Review LVGL API for Micropython Binding compatibility

amirgon, 2020-08-31:
For Micropython (or other future) binding to work correctly, **certain API naming conventions must be followed**.
LVGL API in general follows these conventions. However, in certain cases these conventions were missed.

Examples for such problems:
- #1750
- [lv_list_get_* issues](https://forum.lvgl.io/t/help-lv-list-maybe-have-bugs/3111)

Changing the API breaks it, so we cannot fix that for v7 (although there are workarounds)

Therefore - **I suggest reviewing LVGL API before releasing v8**, and checking that the following conventions are followed:

 - [ ] Widget member function must start with lv_\<widgetName\> 
 - [ ] Widget member function must receive `lv_obj_t *` as its first argument, and this argument must be a pointer to the Widget objects itself.
 - [ ] Member functions of structs should follow the Widget conventions.
They should receive a pointer to the struct as the first argument, and the prefix of the struct name should be used as the prefix of the function name (for example **lv_indev_drv**_register(**lv_indev_drv**_t * driver) )
 - [ ] [Callback conventions](https://github.com/lvgl/lv_binding_micropython#callbacks) must be followed:
   - There's a struct that contains a field called `void * user_data`.
   - A pointer to that struct is provided as the first argument of a callback registration function.
   - A pointer to that struct is provided as the first argument of the callback itself.  
OR  
   - A parameter called `void * user_data` is provided to the registration function as the last argument.
   - The callback itself recieves `void * user_data` as the last argument
 - [ ] Callbacks that are not available for the Binding must include the string `xcb`
 - [ ] All global pointers to dynamically allocated RAM must be defined as [GC roots](https://github.com/lvgl/lvgl/blob/d8a37e8fd8552c7ae8cfadde152a548051d6359d/src/lv_misc/lv_gc.h#L28-L45) and accessed with `LV_GC_ROOT` macro.
 - [ ] Functions which are not part of the public API must begin with underscore, to mark them as "private".
 - [ ] Strings must be passed as `const char *`
 - [ ] Arrays must be passed with square brackets and **not** as pointers. For example: `lv_draw_triangle(const lv_point_t points[]`
 - [ ] Argument must be named also on H files (For example `void lv_style_list_remove_style(lv_style_list_t * list, ...` but not  `void lv_style_list_remove_style(lv_style_list_t *, ...`)
 - [ ] Constants should be declared as enums instead of macros, when possible. (enums are visible to the bindings, macros are not)
 - [ ] C Macros are not available in the binding. Add static inline functions to encapsulate macros for macros that should be part of the API. ([For example, `lv_color_hex`](https://github.com/lvgl/lvgl/blob/652809d0059e57320c2b58102c47dbde3622d6b0/lv_misc/lv_color.h#L405-L411))
 - [ ] All function prototypes must have implementation
 - [ ] Function prototypes must not be duplicated

kisvegabor, 2020-08-31:
Thanks, for the detailed list. We should definitely check it before releasing v8.

Do you see a way to run a test to see if there are any issues?

amirgon, 2020-08-31:
> Do you see a way to run a test to see if there are any issues?

Some things like missing function implementation / duplicate definition would cause the binding not to compile.
Other things like global pointer not GC might cause an application to crash in certain cases.
The others would just cause the Micropython API to be wrong. It might be possible to write a script to check some of these cases.

But if we take for example #1750 - a script cannot detect that problem.  
There is no simple way for a script to "automatically" understand that the first argument of `lv_page_glue_obj`, which is an `lv_obj_t *`, is expected to be **an object on a page** instead of **the page itself**.
The only way to know that is to read the comments.
So I guess a manual sweep of the API might be necessary anyway to eliminate cases like this.

kisvegabor, 2020-09-01:
I see. Do you think it'd be worth the effort to create a build test with CI?

amirgon, 2020-09-01:
> I see. Do you think it'd be worth the effort to create a build test with CI?

Yes, it's definitely worth it, to catch at least some of the problems.
But it cannot replace the manual API review.

embeddedt, 2020-09-22:
Not stale.

amirgon, 2020-09-29:
Related: https://forum.lvgl.io/t/qrcode-with-lv-bindings-micropython/3369  

Apparently external components such as [lv_lib_qrcode](https://github.com/lvgl/lv_lib_qrcode) don't follow the conventions and can't be used with the Micropython bindings.  

I've updated the convention list above.

kisvegabor, 2020-09-29:
Why is it required to have only a parent and copy parameter? 
In v9 (becomes deprecated in v8) the copy parameter will be removed and for the new components, I'm planning to add more parameters to the create functions. E.g. `lv_tabview_create(parent, tab_pos, tab_height)`

If you identify create functions from the function name and parameters would it be possible to use only `lv_<name>_create` and `lv_obj_t * parent` as the first parameter?

amirgon, 2020-09-29:
> Why is it required to have only a parent and copy parameter?
> In v9 (becomes deprecated in v8) the copy parameter will be removed and for the new components, I'm planning to add more parameters to the create functions. E.g. `lv_tabview_create(parent, tab_pos, tab_height)`

In the binding script, object constructor requires special handling which is different than a regular functions.  
Currently I'm using a common function to handle the creation of all the objects, under the assumption that **constructors of all objects receive the same arguments**, which is the case today for all LVGL core objects.   

If you are planning to add more parameters, it would require a significant change to the way objects are constructed in the binding script. A separate construction code would need to be generated for each object type.  
When I find some time, I can check the details and estimate how much effort is required to support this. 

Perhaps it can be possible to support multiple constructors, like in C++?  
That way it would be possible to construct an object and set its parameters either directly by calling a verbose constructor, or indirectly by calling a simple constructor and setting the parameters with setter functions, like we do today.

kisvegabor, 2020-09-30:
I see the problem. So to confirm,  removing the copy parameter from all functions is not an issue but having arbitrary arguments besides the parent is problematic.

In the core lvgl repo all create functions will look like the same but in `lv_components` there will be extra arguments. The reason is to make some parameters creation time specific in order to remove some internal complexity and simplify the API.

By adding multiple types of create functions we would beat the purpose of simple components.

As you offered, please examine to the difficulty of the change in the binding generator.

amirgon, 2020-10-08:
> In the core lvgl repo all create functions will look like the same but in `lv_components` there will be extra arguments. The reason is to make some parameters creation time specific in order to remove some internal complexity and simplify the API.

@kisvegabor - 
- Can you confirm that all future `create` functions (also in `lv_components`) would always return an instance of `lv_obj_t *`, like today?
- Will `lv_components` be built as part of LVGL? Would we use `lv_conf.h` to mark which components should be included?  
If so, what is the motivation of separating `lv_components` from base components? Why not just disable them in `lv_conf.h`?

kisvegabor, 2020-10-08:
> Can you confirm that all future create functions (also in lv_components) would always return an instance of lv_obj_t *, like today?

Yes. And their first argument will be a `parent`.

> Will lv_components be built as part of LVGL? Would we use lv_conf.h to mark which components should be included?
If so, what is the motivation of separating lv_components from base components? Why not just disable them in lv_conf.h?

I imagined as an independent repo with its own config file. 

The motivation is to break complex widgets into small more specific ones. E.g. dedicated message box types (txt only, title + text, image + text, etc). I believe there will be a lot of simpler specific components there and the core widgets will be lean and simple.

amirgon, 2020-10-08:
> I imagined as an independent repo with its own config file.
>
> The motivation is to break complex widgets into small more specific ones. E.g. dedicated message box types (txt only, title + text, image [...]

There are disadvantage to independent component repo
- More complex build scripts for someone who would like to build LVGL + components
- Each repo would need to be versioned independently
- Would they be forward/backward compatible? New components with old LVGL or vice versa? You would need some way to mark which versions of components-repo are compatible with which versions of LVGL-repo.

I don't see much disadvantage to keeping both core LVGL and components in a single repo, especially if your goal is to keep things simple.
Just let the user select which of the specific components he is interested in using a config file.

kisvegabor, 2020-10-09:
> More complex build scripts for someone who would like to build LVGL + components

I think if someone can build one lvg repo he can build and other as well. I'd have an `mk` file similar to lvgl.

> Each repo would need to be versioned independently

It can strictly follow lvgl's version. Similarly to `lv_examples` and `lv_examples`

> Would they be forward/backward compatible? New components with old LVGL or vice versa? You would need some way to mark which versions of components-repo are compatible with which versions of LVGL-repo.

As described in the above point e.g. for lvgl v8.2.1 there will be a matching v8.2.1 lv_components.

> I don't see much disadvantage to keeping both core LVGL and components in a single repo, especially if your goal is to keep things simple.
Just let the user select which of the specific components he is interested in using a config file.

The value see in separations are:
- Do not bloat lvgl
- Welcome contribution more. Some widgets development contributions were rejected because they were too specific, not high quality enough, etc. In lv_components we can have looser rules
- In the future, it can evolve into a community store

puzrin, 2020-10-29:
> I don't see much disadvantage to keeping both core LVGL and components in a single repo, especially if your goal is to keep things simple.

It's still not evident, how much of widgets can be replaced with basic objects. This will be source of too much breaking changes, if been left in core.

Probably, after several major versions, when everything become really stable, it will be possible to return some widgets back. But right now it's very inconvenient for core development.

kisvegabor, 2020-12-21:
I got an idea about how to keep LVGL API lean without the `_` prefix.
Wrap the internal prototypes in the header files into `#ifndef LV_API_PUBLIC ... #endif` and define `LV_API_PUBLIC` in the beginning of `lvgl.h`. This way if you include (or parse) `lvgl.h` the internal function will be disabled becasue `LV_API_PUBLIC` is defined but if you include a header file directly, it's full content will be there. 

What do you think?

amirgon, 2020-12-21:
> I got an idea about how to keep LVGL API lean without the `_` prefix.

- Anyone who includes lvgl.h cannot use private functions. Is this ok?
- It won't work if you have a public inline function that calls private functions.
- It won't work for private structs that are passed (by pointer) to public functions.

Think about a struct who's content is opaque to the user (the user access it with getter/setter functions).
You would need to declare it twice, once as a forward declaration without implementation, and again with its implementation.

If you are thinking in this direction, why not separate public/private interfaces entirely into different header files? Why do you need ifdefs to make this distinction?
Create a public interface with the public functions (and forward declarations for private structs), and create one or more private (or "implementation dependent") API that includes it.

kisvegabor, 2020-12-22:
Probably you are right. I'll check how to separate headers in a more common way. 

Do you expect to see the drawing functions and similar lower-level API in MP? (e.g. to create a custom widget)

amirgon, 2020-12-22:
> Do you expect to see the drawing functions and similar lower-level API in MP? (e.g. to create a custom widget)

Yes, I think it can be very useful to create custom pure-Python widgets.  
The binding now [supports calling function pointers](https://github.com/lvgl/lv_binding_micropython/issues/110), so it's possible to call `get_design_cb` which returns a function pointer and then call this function pointer with arguments from Micropython.  

When you are separating the API into public/private API, please take into account these points:

- A new optimization allows the bindings to re-use wrapper functions for API functions with the same prototype. This is very significant for reducing the program size. So we really want to have as many functions with the same prototype on the public API.  
For that reason it's better to keep receiving `lv_obj_t*` argument than a pointer to a specific widget such that functions from different widgets could use the same wrapper.
- Consider converting public structs to opaque structs that are only accessible by functions but don't expose their fields. It's a good practice in general because it separates the API from the internal struct implementation, field names etc.  
The bindings know how to wrap structs for Micropython access, but it costs relatively large program RAM and cannot be re-used like function wrappers with the same prototype.

More details are on https://github.com/lvgl/lv_binding_micropython/issues/81#issuecomment-749373934

kisvegabor, 2020-12-27:
All clear, thanks!

kisvegabor, 2021-01-18:
@amirgon I'm thinking about a flexible and easy to use "drawer" extension. The idea is to have a `drawer_cb` style property which is a function pointer to a custom drawer. During rendering LVGL calls this callback in various stages of the drawing to allow customization. I'd like to verify if the followings really fulfill the callback conventions.

> Callback conventions must be followed:
>   - There's a struct that contains a field called void * user_data.
>   - A pointer to that struct is provided as the first argument of a callback registration [...]

Prototype of the drawer:
```c
lv_draw_res_t my_drawer(lv_obj_t * obj, lv_part_t part, lv_draw_mode_t mode, 
                        const lv_area_t * clip_area,  
                        void * ctx, void * user_data);
```
The interesting arguments are:
- `ctx` a pointer that is interpreted according to `mode`, `part` and type of the widget. E.g. `lv_slider_indicator_draw_ctx_t *`
- `user_data` to follow the convention. In C user can add any parameter to the drawer.

To assign a draw to a style:
```c
lv_style_set_drawer_cb(&style1, my_drawer, user_data);

static inline void lv_style_set_drawer_cb(style_t * style, lv_drawer_t * cb, void * user_data)
{
  lv_style_set(style, LV_STYLE_DRAWER_CB, cb);
  lv_style_set(style, LV_STYLE_DRAWER_USER_DATA, user_data);
}

``` 

Could it work?

Just out of curiosity, is there any chance to get rid of this convention somehow? If I understood correctly so far the user data is required to store a some MP data to call the callback later. Could these data be in a global dictionary for example? E.g. `obj1 : cb_data, style1: other_cb_data`

kisvegabor, 2021-01-18:
Another approach - maybe better - is to have an struct like:
```c
typedef struct {
  lv_drawer_cb_t cb;
  void  * user_data;
}lv_drawer_t;
```

and provider set/get functiosn for this struct and set the pointer of this struct in a style:
```c
static lv_drawer_t d;
lv_drawer_init(&d);
lv_drawer_set_cb(&d, my_drawer);
lv_drawer_set_user_data(&d, user_data);

lv_style_set_drawer(&style1, &d);
```

amirgon, 2021-01-18:
> The idea is to have a `drawer_cb` style property which is a function pointer to a custom drawer

@kisvegabor I would call it "draw_cb" instead of ["drawer_cb"](https://dictionary.cambridge.org/dictionary/english/drawer)... :wink:

> I'd like to verify if the followings really fulfill the callback conventions.

Yes, as long as both registration function and callback function receive `void * user_data` as their last argument.   

`void * ctx` is a bit confusing and can be a source of bugs if the user makes a mistake and casts it to the wrong type, no matter if on C or on Micropython.  
Is it possible to make it clearer? For example, a pointer to a union that contains all possible "context" types? Maybe even a different "draw_cb" prototype for each context type?

> Another approach - maybe better - is to have an struct like:

That can also work, but only if the "draw_cb" callback receives a pointer to `lv_draw_t` as its first argument.  
In your example it doesn't.

> Just out of curiosity, is there any chance to get rid of this convention somehow? If I understood correctly so far the user data is required to store a some MP data to call the callback later. 

Let me explain why we need these "callback conventions".  
The starting point is that on embedded platforms we cannot generate functions at runtime. In fact, in certain cases there is a way to do it with [libffi](https://en.wikipedia.org/wiki/Libffi) for example, but it's not always possible on embedded platforms for various reasons (it requires modifying the instruction RAM for example), so my assumption is that we can't use FFI.

If there is only a single callback then there is no problem.  For example, `lv_log_register_print_cb` receives `lv_log_print_g_cb_t` that doesn't follow the conventions above because it's set globally (we still need the "_g_cb" naming convention to claim that this is a global callback)  

But for most callbacks, we can use the same "register" function to register different callbacks in different contexts. In case of C we simply pass a different function each time. But in Micropython we decide about the callback functions to pass to the register function **on runtime**. Since we can't generate C functions on runtime, we must pass **the same C callback function** to the register function every time we call register, even if we want to register different callbacks to different entities (styles, objects etc.).  

The purpose of the `user_data` pointer on Micropython is to hold the Micropython callback that should be called.
The same C function is registered on different calls to the same "register_cb", but user_data is different each time and the C callback extracta the actual Micropython callback from it.  
For this to work, the same user data must be passed to the registration of the callback and to the callback function itself.

If you want to "get rid" of this convention, you would need to either
- Find a way to generate function on runtime each time a callback registration is called, with libffi for example, and limit yourself to targets that support that.
- Limit the registration function such that it would register only a single callback, like `lv_log_register_print_cb`. Registering a different callback would inevitably remove the previous callback, regardless of its context.

I hope this explanation was clear enough, please let me know if you need more clarification about this.

> Could these data be in a global dictionary for example? E.g. `obj1 : cb_data, style1: other_cb_data`

What would be the key of that dictionary? The same key would need to be passed to the registration and to the callback. You would still need a convention that says that the same key needs to be passed to both the registration and the callback functions (as first argument for example).  
If the key is "style" for example, then the same style pointer would need to be passed to both the registration function and the callback function. In that case, isn't it simpler just to add "user_data" member to style, like we do today, instead of this extra indirection through a dictionary?

kisvegabor, 2021-01-19:
> I would call it "draw_cb" instead of "drawer_cb"... wink
 
Thanks, I haven't noticed the coincidence. :laughing: 

Also thank you for the explanation, it's clear now. And knowing this I agree a global dictionary wouldn't be beneficial.

I'll polish the API, follow your instructions and show it when it's more stable.

kisvegabor, 2021-02-25:
I've ticked the points that are implemented. Maybe I missed them in a few places in LVGL but conceptually we can follow them and the incompatibilities can be easily addressed. 

See my comments on the unticked points:

> A function that returns a pointer to a specific widget should return a pointer to that widget and not to a generic lv_obj_t.
For example, lv_tabview_add_tab should return lv_page_t instead of lv_obj_t.

This way it wouldn't work:
```c
lv_btn_t * btn1 = lv_btn_create(lv_scr_act(), NULL);
lv_obj_set_width(btn1, 100);  // Should be (lv_obj_t *) btn1

lv_label_create(btn1, NULL);   // Should be (lv_obj_t *) btn1
```

> Widget constructor must be named lv_<widgetName>_create and receive lv_obj_t * par, const lv_obj_t * copy arguments. No additional arguments allowed. For example: lv_obj_t * lv_btn_create(lv_obj_t * par, const lv_obj_t * copy);

It'd be great to make this more flexible. E.g. `lv_<widgetName>_create`, receives `lv_obj_t * parent` as first argument and any other argument after the parent. I'm saying this because 
- `copy` will be removed in v9
- In extra (non-core) widgets by adding some parameters at creation time makes the API smaller and the implementation leaner.  E.g. do not allow changing the tabs' position after creation on tab view. See [here](https://github.com/lvgl/lvgl/blob/8fdefa850855596e76e8795487798db6e5dc903e/src/extra/widgets/tabview/lv_tabview.c#L48).

> Need to verify these conventions for both core and external widgets, such as lv_lib_qrcode

The libs still need to be reviewed.

> Avoid ambiguity in between struct members and struct member functions. (For example start in lv_anim_t)

Still needs to be checked.

>  lv_set_event_cb must be reimplemented and not inherited from lv_obj, if we want the callback to receive the component and not lv_obj as argument. See this post. (Can be implemented as static inline wrapper so no C overhead)

Is this still required? If I understood correctly it'd require functions like this
```c
void lv_btn_add_event(lv_btn_t * btn, lv_event_cb_t * cb, void * user_data); //User data is a new parameter
```

I'd like to keep the generic events unless it's inevitable. 

> Some structs should not be used directly by the user. Instead, LVGL provides many setter/getter functions. For example lv_anim_t has lv_anim_set_path to set the path instead of directly writing to path field of lv_anim_t.
In order to minimize the API size, it's better to mark such structs as "private" by prefixing them by underscore. This way the binding script would know not to wrap every field in them, and treat them as opaque objects instead.

If we named it `_lv_anim_path_t` the user also needed to use `_lv_anim_path_t path1` in C which is quite counterintuitive.

embeddedt, 2021-02-25:
If we had a more efficient way of checking an object's type than the current string solution, maybe the binding could check what type the object is and automatically cast it to the appropriate MicroPython structure.

kisvegabor, 2021-02-25:
Type checking became much simpler. E.g. 
```c
if(lv_obj_check_type(obj, &lv_dropdown_class)) ...
```

I added only this function but something like this is also possible:
```c
if(lv_obj_get_type(obj) == &lv_dropdown_class) ...
```

embeddedt, 2021-02-25:
Perfect!

amirgon, 2021-03-01:
> This way it wouldn't work:
> 
> ```c
> lv_btn_t * btn1 = lv_btn_create(lv_scr_act(), NULL);
> lv_obj_set_width(btn1, 100);  // Should be (lv_obj_t *) btn1
> 
> lv_label_create(btn1, NULL);   // Should be (lv_obj_t *) btn1
> ```

Actually, in this case it's not needed, because the bindings identifies `lv_btn_create` as btn constructor and will create btn object.  
The problem happens in cases where it's harder to guess the returned class from the API.  
For example `lv_tabview_add_tab` returns a page, but how could the binding guess that?  

In my opinion `lv_obj_get_type` is not a good solution for this, because this kind of type deduction should be resolved on compile time not on runtime. There is no sense in checking `lv_tabview_add_tab` return value on runtime because it would always return "page" type and nothing else.  
Instead, I suggest some hint or convention that would help.  

Here is an idea - what if we used typedef for mapping `lv_page_t` to `lv_obj_t`?
```c
typedef lv_page_t lv_obj_t;
lv_page_t * lv_tabview_add_tab(lv_obj_t * tabview, const char * name);
```
From C perspective that's the same as `lv_obj_t` return value, but the bindings sees typedefs and can automatically deduce that the returned value of `lv_tabview_add_tab` is a page object and not base obj.  
It actually provides a meaningful hint for the C user as well.

> Is this still required? If I understood correctly it'd require functions like this
>
> ```c
> void lv_btn_add_event(lv_btn_t * btn, lv_event_cb_t * cb, void * user_data); //User data is a new parameter [...]

Well, the problem here is that when the micropython callback is called, it would pass "obj" as its argument.  
So if the callback argument object is really a pointer to "btn", the user would have to explicitly cast it to "btn".  
It would be better to pass `lv_btn_t` instead of `lv_obj_t` such that the micropython callback would receive "btn" and no cast would be needed.
While casting is possible to work around this, I found out it confuses Python users who are not used to this concept and assume that when "obj" is passed it's never some obj subclass.

> If we named it `_lv_anim_path_t` the user also needed to use `_lv_anim_path_t path1` in C which is quite counterintuitive.

If you remember, that's why I suggested ["public" and "private" API](https://github.com/lvgl/lvgl/issues/1763#issuecomment-749058873-permalink:~:text=If%20you%20are%20thinking%20in%20this,%22implementation%20dependent%22%29%20API%20that%20includes%20it.).
On the public API the structs would be only declared (but not defined) therefore opaque to the user.
To access the struct fields, one would have to include the "private" API. Users would not need to do that since the public API would provide accessor functions.  
This has the advantage of hiding implementation details from a public API.

kisvegabor, 2021-03-02:
> There is no sense in checking lv_tabview_add_tab return value on runtime because it would always return "page" type and nothing else.

In v8 `lv_tabview_add_tab` returns return a plain `lv_obj` but I see the problem. E.g. `lv_list_add_btn` returns a button.  Can we use naming conventions here. E.g. 
`lv_<widget_name>_add_<return_type>(lv_obj_t * obj, <any other parameter>)` but it's too LVGL specific and doesn't work for 
`lv_chart_series_t * lv_chart_add_series(lv_obj_t * chart, lv_color_t color)` where not an `lv_obj_t` is returned.

Do you have any ideas about the naming convention way or is it a dead end in this case?

I think `typedef lv_obj_t lv_page_t;` would mislead users if they see the prototype of the function. What an `lv_page_t` is? What can I do with that?

> Well, the problem here is that when the micropython callback is called, it would pass "obj" as its argument.
So if the callback argument object is really a pointer to "btn", the user would have to explicitly cast it to "btn".
It would be better to pass lv_btn_t instead of lv_obj_t such that the micropython callback would receive "btn" and no cast would be needed.

I see the problem. Even a simple `label1.get_text()` is not working without a cast in the `event_cb`. Couldn't it be managed in the binding somehow? E.g. saving the original class of the widget in the `user_data` and automatically casting `obj` to `btn` before the event callback is called? Sorry if it doesn't make any sense, just asking to see the whole picture.

> If you remember, that's why I suggested "public" and "private" API.
On the public API the structs would be only declared (but not defined) therefore opaque to the user.
To access the struct fields, one would have to include the "private" API. Users would not need to do that since the public API would provide accessor functions.

The forward declaration can work if the struct is used as a pointer. But `lv_anim_path_t` is [directly embedded](https://github.com/lvgl/lvgl/blob/b8b66c6929bda8e19d4d4488326efd162ea72bfa/src/lv_misc/lv_anim.h#L83) into `lv_anim_t`. With forward declaration, the compiler couldn't calculate the size of the struct. But using pointers is doable. 
However, if the H file that defines `lv_anim_path_t` (not just the forward declaration) needs to be included in `lvgl.h` to let user creating new paths with `lv_anim_path_t` type instead of `struct _lv_anim_path_t`.

amirgon, 2021-03-02:
> Do you have any ideas about the naming convention way or is it a dead end in this case?

I'm not sure.
So far I don't have a good idea for naming conventions that would resolve this.

> I think `typedef lv_obj_t lv_page_t;` would mislead users if they see the prototype of the function. What an `lv_page_t` is? What can I do with that?

I see it the other way around - it tells the user that this `lv_page_t` can be used safely with any `lv_page_<somthing>(...)` function.  
It can be documented that it's just a name for `lv_obj_t`, and very easily understandable from the typedef itself.  
This is as close as we can get to returning `lv_page_t *` instead of `lv_obj_t *` which you prefer.  
But if you don't like it - I'm opened to other options. Currently I just don't have any.

We can, of course, revert to runtime type resolution with `lv_obj_get_type`.  
It's a possible solution but I don't like it for the following reasons:
- It's very LVGL specific. The bindings would need to know about `lv_obj_get_type` and how to use it. It's not easily scalable to other libraries.  
- It's inefficient from both code size and performance. It's already possible to deduce types on compile time in all these cases and return the right type directly. Why would we do that on runtime?

> I see the problem. Even a simple `label1.get_text()` is not working without a cast in the `event_cb`. Couldn't it be managed in the binding somehow? E.g. saving the original class of the widget in [...]

It might be possible but again suffers from the problems I mentioned above - it's very LVGL specific and requires adding program/cycles for something that can be done on compile time instead.

> The forward declaration can work if the struct is used as a pointer. But `lv_anim_path_t` is [directly [...]

It's still possible with the right accessor functions to read/write `lv_anim_path_t`.  
It won't work if you expect the user to directly create instances of these structs (on stack or statically). In these cases the struct would need to be in the public interface header and it makes sense because it really should be visible to the user who is allocating and initializing it.

I'm referring to all the structs that the user never creates instances of.
For example `lv_obj_t` or any of it's descendants such as `lv_chart_t` etc.

Another example is `lv_disp_t`. Is the user expected to create `lv_disp_t` instances directly? Is he expected to access `lv_disp_t` fields directly? If not, it's better to put it in a private header file and hide its implementation details (fields) from the user.  
By hiding implementation details of structs we save significant program size in the bindings because it won't create wrapper for each struct field.  
It's also good from pure C perspective as it allows you to change these structs without touching the public interface.

amirgon, 2021-03-04:
I've added [another unticked point above](https://github.com/lvgl/lvgl/issues/1763#:~:text=Function%20arguments%20that%20are%20pointers%20to,file_p%20and%20not%20void%20*%20file_p.) regarding function argument types. [Related forum post](https://forum.lvgl.io/t/micropython-examples-for-v7/3942/51#post_52:~:text=%40kisvegabor%2C%20%40embeddedt%20Do%20you%20know%20if,casts%20could%20be%20eliminated%20from%20Micropython.)

kisvegabor, 2021-03-08:
### lv_tabview_add_tab() return value

Introducing `typedef lv_obj_t lv_xxx_t` for every widget might work, but only for some of them would be messy IMO. You might ask "okay, then why don't we add it to every type?" Even it seems a minor change and fully backward compatible, it's a conceptual shift and needs some time to think about. If it workes so far this way, I suggest postponing it. We can even add it a minor release later.  

### Object type in event_cb 

Seemingly it also depends on `typedef lv_obj_t lv_xxx_t`. So let's wait a little with it as well. 

### Reduce API

> I'm referring to all the structs that the user never creates instances of, at least not directly.
For example lv_obj_t or any of it's descendants such as lv_chart_t etc.

they might be required if the user creates a new widget based on e.g. `lv_chart`. He needs to include `lv_chart_t` as the first element of `my_new_widget_t` to allow calling `lv_chart_...()` functions. 

> Another example is lv_disp_t. Is the user expected to create lv_disp_t instances directly? Is he expected to access lv_disp_t fields directly?

Yes, it's the return value of `lv_disp_drv_register` and `lv_disp_t` need to be used in a multi-display environment.

amirgon, 2021-03-08:
> Even it seems a minor change and fully backward compatible, it's a conceptual shift and needs some time to think about. If it works so far this way, I suggest postponing it.

So users would need to keep using these weird type-casting... I hoped we could find a way to overcome this.  
Until then I'll be spending my time in the forum explaining to users how to make these type-casts work :unamused:

> Yes, it's the return value of `lv_disp_drv_register` and `lv_disp_t` need to be used in a multi-display environment.

But the user is not creating it explicitly - he is receiving a pointer as a return value of  `lv_disp_drv_register`, right?
So my point stands that `lv_disp_t` can be defined in a private header.
I'm sure we can find more structs on the same status.

I've already spent significant effort bringing down the code size for the bindings, see https://github.com/lvgl/lv_binding_micropython/issues/81.  
Private headers sounds to me like a simple change that has the potential of improving program size with relatively small effort.

kisvegabor, 2021-03-09:
> So users would need to keep using these weird type-casting... I hoped we could find a way to overcome this.
Until then I'll be spending my time in the forum explaining to users how to make these type-casts work unamused

I wonder if it could help if split the API of LVGL in some. I don't have a clean and readable idea but maybe you have. 
E.g.
```c
#if LV_BUILD_MICROPYTHON
#define ...
#endif
```

Wrapping every create function in an `#if` is obviously not an option but we can create any kind of helper function. Even in a specific `c` file. That file can be even auto-generated.

> But the user is not creating it explicitly - he is receiving a pointer as a return value of lv_disp_drv_register, right?
So my point stands that lv_disp_t can be defined in a private header.
I'm sure we can find more structs on the same status.

But wouldn't it require writing this way?
```c
struct _lv_disp_t  * disp1 = lv_disp_drv_register(&drv);  //"struct _lv_disp_t" looks bad
```

amirgon, 2021-03-09:
> ```c
> #if LV_BUILD_MICROPYTHON
> #define ...
> #endif
> ```

I prefer avoiding this.  
The big problem is maintaining it - making sure these ifdefs always contain **only** what is needed and **all** what is needed, and reviewing that on every release.  
I prefer inferring everything from the same code base that is used for C applications, like we do today.

> But wouldn't it require writing this way?
> 
> ```c
> struct _lv_disp_t  * disp1 = lv_disp_drv_register(&drv);  //"struct _lv_disp_t" looks bad
> ```

No, you don't need the `struct` if you use typedef.
```c
typedef struct _lv_disp_t lv_disp_t;
```
This typedef can be in the public header.
```

kisvegabor, 2021-03-11:
*First of all, I'd like to ensure you that my intention is also to make the binding full-featured but I also need to be sure to not cause confusion in the C part meanwhile.  So I'm also interested in supporting Micropython as much as possible.*

---

I think there will be a moment when it won't be reasonable to apply further restrictions on C side but we should separate the API somehow. 

For example, now to set an animation path you have to create an `lv_anim_path_t` variable. Thinking only about the C code it wouldn't be required because the `path_cb` could be set directly on `lv_anim_t` (To be honest I don't remember why was it required to detach it. An `int32_t my_path_cb(const lv_anim_t * a)` seems working on both sides). Or we have `lv_anim_set_custom_exec_cb`. These are local changes and we can live with them. However, adding `define`s on similar cases would be too much trouble neither from readability nor from a maintenance perspective. 

Anyway, it doesn't solve our current issues of "create functions' return value" and "object type in event_cb".  

I believe for the events an auto generated file could work very well. It seems trivial to find all `lv_create_<name>` functions and add 
```c
void lv_<name>_add_event_cb(lv_<name>_t * obj, lv_event_t e, void * p)
```
 functions. 

For the "return value" question can we apply a naming convention like  `lv_tabview_create_tab`? (Drop the `add` keyword here) In LVGL we can ensure to have an `lv_tabview_tab_t` type. Similarly `lv_list_create_btn()` and `lv_list_btn_t`. These functions still return `lv_obj_t *` as e.g. `lv_btn_create()` does. 

Basically what I'd like to avoid is returning other types than `lv_obj_t *` in a few exceptional cases. Of course, it also has a maintenance cost, but all the already existing naming conventions have, 

I suggest finding a solution to these two issues first and then move to the "API size" question.

amirgon, 2021-03-12:
> I think there will be a moment when it won't be reasonable to apply further restrictions on C side but we should separate the API somehow.

The main problem with separate API is maintenance.
It's the same reason I wanted automatically generated bindings. If the bindings were manual we would have never reached that far. If we separate the API, the separated API would have to be maintained and it probably won't be maintained because it's hard to maintain two separate APIs that do the same thing and we are all already busy with a lot of other high priority tasks.  

So my opinion is that we must have a single unified API, and generate the bindings from it the best way we could.

I also agree that we need to keep the C API intuitive for the C users.

It's a trade-off:  
- If you make no compromise on the C API - the Micropython bindings suffer.
- If you make too many compromises on the C API - the C users can get confused.

So let's discuss how to design a C API that doesn't confuse the C users, but is descriptive enough for the Micropython bindings.
But not how to separate the API because that would kill Micropython bindings.

> Basically what I'd like to avoid is returning other types than `lv_obj_t *` in a few exceptional cases. Of course, it also has a maintenance cost, but all the already existing naming conventions have,

I didn't suggest returning a different type. I suggested returning the same type, but using a typedef such that the function return value is clear. Functionally it is exactly the same.  
Your opinion is that it is confusing for the C users, my opinion is this is actually clearer for the C users and useful for any bindings.

> Anyway, it doesn't solve our current issues of "create functions' return value" and "object type in event_cb".

Let's list the options we have so far:

- Don't change anything: the Micropython users would have to cast `lv_obj_t` explicitly to the specific widget class.
- Detect the real type of the returned `lv_obj_t` on runtime
- Use naming conventions on the C code to annotate the returned `lv_obj_t` type.

I'm willing to reconsider the runtime type detection.

Here is what I wrote about it the previous post:

> We can, of course, revert to runtime type resolution with `lv_obj_get_type`.
> It's a possible solution but I don't like it for the following reasons:
> 
> * It's very LVGL specific. The bindings would need to know about `lv_obj_get_type` and how to use it. It's not easily scalable to other libraries.
> * It's inefficient from both code size and performance. It's already possible to deduce types on compile time in all these cases and return the right type directly. Why would we do that on runtime?
> * If you think about bindings to C++ (or other statically typed language) in the future, you definitely would want to resolve these types on compile time and take advantage of the strong type system of the language.

- It's LVGL specific and not scalable to other libraries, but so are any naming conventions we would choose for annotating the returned type.
- It's less efficient, but if designed correctly would add very small penalty. For this to work efficiently we would need an efficient way to get the ID of the widget type on runtime so we could look it up in an array. `lv_obj_get_type` is not good enough because it requires many tests to find the right type.
- It's incompatible with imagined future bindings for statically typed languages such as C++, where dynamic type resolution is not an option. But do we really care about this now?

So what do you think? C naming conventions for returned types, or runtime type resolution?

> I suggest finding a solution to these two issues first and then move to the "API size" question.

Ok, but please let's discuss the "API size question" **before releasing v8**.
This is important because splitting the headers into private and public is a breaking change from user perspective so once you release v8 we would probably need to postpone this again to v9, and I prefer not to.

embeddedt, 2021-03-13:
> For this to work efficiently we would need an efficient way to get the ID of the widget type on runtime so we could look it up in an array. lv_obj_get_type is not good enough because it requires many tests to find the right type.

Which `lv_obj_get_type` are you referring to? The new one should be fast enough unless I've missed something.

amirgon, 2021-03-13:
> Which `lv_obj_get_type` are you referring to? The new one should be fast enough unless I've missed something.

What I meant is that a function that returns a pointer is not good enough because it requires comparing a type to all other types.  
Instead, we would need a function that returns an enum that can be used as an index to array.

kisvegabor, 2021-03-16:
So would you need something like this?
```c
uint32_t lv_obj_get_type_id(lv_obj_t * obj);

/*Or*/
uint32_t lv_obj_class_get_id(lv_obj_class_t * class)
```
If so 
1. all IDs need to be known in compile-time or can we assign them dynamically when a related widget is created?
2. IDs need to increasing "normally"? (E.g. 1,2,3,4,5 and not 12432, 56, 976234, 2342)

It's too obvious to be true, but why can't you use a dictionary to map `lv_obj_class`es to the types you need?

> Ok, but please let's discuss the "API size question" before releasing v8.

Okay.

embeddedt, 2021-03-16:
Why not just cast the class pointer to an unsigned integer and treat it as an ID?

amirgon, 2021-03-16:
> So would you need something like this?
> 
> ```c
> uint32_t lv_obj_get_type_id(lv_obj_t * obj);
> 
> /*Or*/
> uint32_t lv_obj_class_get_id(lv_obj_class_t * class)
> ```

Yes.

> 1. all IDs need to be known in compile-time or can we assign them dynamically when a related widget is created?

Compile time, at least for all widgets that are known by the bindings.  
We would also need an enum to list all widgets classes.  
There should be a certain number to represent the maximum widget class known on compile time, and everything above that would not be handled by the bindings. (could be the last element of the enum)

> 2\. IDs need to increasing "normally"? (E.g. 1,2,3,4,5 and not 12432, 56, 976234, 2342)

Yes because I want to use this number as an array index. And we must know the maximum number (array size)

> It's too obvious to be true, but why can't you use a dictionary to map `lv_obj_class`es to the types you need?

We can, but I want the code to be efficient. I want to avoid dictionary lookups on every access to `lv_obj_t`. But array lookup is much cheaper.

> Why not just cast the class pointer to an unsigned integer and treat it as an ID?

Because then we would need a more expensive dictionary lookup to find the class.
I want to use a simple array access, with a static array that contains all Micropython widget classes.
So accessing `arr[lv_obj_class_get_id(o)]` could return the Micropython  widget class directly.

We would also need an enum to list all widgets with strict naming that maps to each widget. This enum would be used by the binding script to create the static array that contains Micropython's widget classes.

---

There may be another way to do that.
If I could register the Micropython widget class in the C widget class, then I could ask for it directly from the `lv_obj_t`.  
But that would require some mechanism to register every C widget class by the bindings and a way to store the Micropython widget class in the C widget class (like `user_data` for a widget class).

kisvegabor, 2021-03-16:
> Because then we would need a more expensive dictionary lookup to find the class.

Is it really significant? Have you already had issues with it?
I would be very surprised if the difference between array vs dictionary would be measurable compared to cases when calculation needs to be performed on every pixel. 

> There may be another way to do that.
If I could register the Micropython widget class in the C widget class, then I could ask for it directly from the lv_obj_t.
But that would require some mechanism to register every C widget class by the bindings and a way to store the Micropython widget class in the C widget class (like user_data for a widget class).

Now C widget classes are `const` so we can't add `user_data`. We can make them non-const for 1-2k extra RAM usage.

amirgon, 2021-03-16:
>> Because then we would need a more expensive dictionary lookup to find the class.

> Is it really significant? Have you already had issues with it?

If we use a dictionary instead of a static array we would need an extra step of creating the dictionary and populating it on runtime.
Today there is no "init" function for the bindings. If we create one we would need LVGL to call it (probably from `lv_init`).  

Another option is to create a static array of pointers to all known C widget classes (on compile time), but then the bindings would have to scan it all over every time instead of using a more efficient data structure like a dictionary.
How many "classes" are we talking about?

> I would be very surprised if the difference between array vs dictionary would be measurable compared to cases when calculation needs to be performed on every pixel.

It's very different because of cache locality. When processing every pixel I assume (hope) everything is cached-in for the CPU.  
When accessing Micropython objects like dictionaries we can't guarantee cache locality.  
On ESP32 this is significant because RAM is often allocated on an external 40Mhz SPI-RAM so cache misses are very painful.

kisvegabor, 2021-03-17:
> How many "classes" are we talking about?

1 (very rarely 2 or 3) per widget type. Let's 1.2 per widget type. So for an average UI it's about 10 classes.

amirgon, 2021-03-17:
> So for an average UI it's about 10 classes.

What do you mean for the average UI?  
This is supposed to be an absolute number known on compile time, for the bindings to convert C classes to Micropython classes.

Classes are defined as initialized static globals, right?

If I define on compile time an array of pointers to all classes, can I assume its size is about the total number of widgets?

I would also need strict mapping between classes and widget names. Is there a naming convention for this?  (btn --> g_btn_class) can I assume every widget has a class that matches its name?

kisvegabor, 2021-03-17:
> What do you mean for the average UI?
This is supposed to be an absolute number known on compile time, for the bindings to be able to convert C classes to Micropython classes.

It depends on ​the enabled widgets. 
If a widget is enabled its class(es) are visible.

> Classes are defined as initialized static globals, right?

Yes. 

> If I define on compile time an array of pointers to all classes, can I assume its size is about the total number of widgets?

Almost. Some widget has more classes. E.g. the dropdown list has `lv_dropdown_class` (for the "button") and `lv_dropdown_list_class` (for the list that opens)

> I would also need strict mapping between classes and widget names. Is there a naming convention for this? (btn --> g_btn_class) can I assume every widget has a class that matches its name?

Yes, names are built like this (e.g. for checkbox)
- [lv_checkbox_t](https://github.com/lvgl/lvgl/blob/a7084509b51e8c39d4018144b38b5b6a4b5242f8/src/widgets/lv_checkbox.h#L29-L33) the data of the widget
- [lv_checkbox_class](https://github.com/lvgl/lvgl/blob/a7084509b51e8c39d4018144b38b5b6a4b5242f8/src/widgets/lv_checkbox.c#L37-L44) the class of the widget

amirgon, 2021-03-17:
> Almost. Some widget has more classes. E.g. the dropdown list has `lv_dropdown_class` (for the "button") and `lv_dropdown_list_class` (for the list that opens)

So according to strict naming, can I assume there are both `lv_dropdown_create(...)` and `lv_dropdown_list_create(...)` functions for creating `dropdown` and `dropdown_list` widgets?  

Or is `dropdown_list` widget actually a `list` and the naming convention is not strict?

kisvegabor, 2021-03-19:
No, `dropdown_list` is not a `list` it's just a coincidence. The whole name should match. `dropdown_list != list`

`dropdown_list` is used only internally by the dropdown widget and it doesn't expose the create function. `lv_dropdpwn_open/close` can be used instead.

Does the create functions matter from the type cast's point of view?

amirgon, 2021-03-19:
> `dropdown_list` is used only internally by the dropdown widget and it doesn't expose the create function. `lv_dropdpwn_open/close` can be used instead.

So something  is not clear to me.  
When I call `lv_dropdown_get_list` I get an object of class `lv_dropdown_list_t` right?
But there is no such widget. No `lv_dropdown_list_create`. No `lv_dropdown_list_*` members. How do we suppose to know which functions can be called with this object? It's impossible to know that from the function prototypes.

(btw. my typedef suggestion could have solved that)

Are there any other cases where the returned object is not really a widget?

> Does the create functions matter from the type cast's point of view?

In fact, the create functions matter.
The create functions are the way the bindings identify a widget. So the function name `lv_dropdown_create` tells the bindings that there is a widget called `dropdown`, there should be a class called `lv_dropdown_class` etc.

That's why I wanted to know if you keep strict naming conventions for the widget names, (eg. `dropdown`), public constructor name (eg. `lv_dropdown_create`), class name (eg. `lv_dropdown_class`), member names (`lv_dropdown_get_text`), etc.

If I understand correctly, the above example of `lv_dropdown_list` is an example where the conventions are not kept.

And another thing - a widget must not be a prefix of another widget. For example we can't have `dropdown` and `dropdown_list` widgets otherwise the member names are ambiguous.  
Are there any such cases? I'm adding this as another point to the list.

kisvegabor, 2021-03-20:
> When I call lv_dropdown_get_list I get an object of class lv_dropdown_list_t right?
But there is no such widget. No lv_dropdown_list_create. No lv_dropdown_list_* members. How do we suppose to know which functions can be called with this object? It's impossible to know that from the function prototypes.

Actually, `lv_dropdown_list` is still a widget. It's a normal object, and you set/get its size, add/remove styles. But it has no specific API functions, only `lv_obj_...`. `lv_btn` is the same in this regard because there are no `lv_btn_set/get` functions. It has a create function though.

So if there are no `lv_dropdown_list_set/get` function, `lv_obj_..` still should apply. 

> Are there any other cases where the returned object is not really a widget?

`lv_roller_class` and `lv_roller_label_class` are similar. 

> In fact, the create functions matter.
The create functions are the way the bindings identify a widget. So the function name lv_dropdown_create tells the bindings that there is a widget called dropdown, there should be a class called lv_dropdown_class etc.

I see.

> If I understand correctly, the above example of lv_dropdown_list is an example where the conventions are not kept.

Yes, it's a special case in which it was convenient to manage the "list" as a new widget. 

>And another thing - a widget must not be a prefix of another widget. For example we can't have dropdown and dropdown_list widgets otherwise the member names are ambiguous.

It might important in the future too: so widget names should be one word? E.g. `lv_my_magic_btn_create()` won't work? 

Would it help if we simply change it to `lv_dropdownlist_class`? (no `_`)

`lv_roller_label` can be hidden in the C file but `lv_dropdown_list` should be public to allow applying a theme based on its type. I.e. `lv_obj_get_type(obj)` returns `lv_dropdown_list_class` and the appropriate them can be applied.

amirgon, 2021-03-20:
> It might important in the future too: so widget names should be one word? E.g. `lv_my_magic_btn_create()` won't work?
> 
> Would it help if we simply change it to `lv_dropdownlist_class`? (no `_`)

Actually, there is no such limitation that widget names should be one word with no underscore.
What I ask is that a widget name would never be a prefix of another widget name.  
So you can have a `magic_btn` widget but not both `magic_btn` and `magic` widgets, for example.
Otherwise, when seeing `lv_magic_btn_set` we can't tell if it's a `set` member of `magic_btn` or a `btn_set` member of `magic`...

> `lv_roller_label` can be hidden in the C file but `lv_dropdown_list` should be public to allow applying a theme based on its type. I.e. `lv_obj_get_type(obj)` returns `lv_dropdown_list_class` and the appropriate them can be applied.

If `dropdown_list` has no member functions and only inherits from `obj` there is no problem.  
The bindings would not find an appropriate widget class (because of the missing `create` function) and leave it as the base `obj` without casting it to any other widget.

kisvegabor, 2021-03-21:
> Actually, there is no such limitation that widget names should be one word with no underscore.
What I ask is that a widget name would never be a prefix of another widget name.

Got it. `magic_btn` works but `btn_magic` doesn't. :slightly_smiling_face: 

> If dropdown_list has no member functions and only inherits from obj there is no problem.
The bindings would not find an appropriate widget class (because of the missing create function) and leave it as the base obj without casting it to any other widget.

Great! So finally is there anything to modify in lvgl?

amirgon, 2021-03-22:
> Great! So finally is there anything to modify in lvgl? (besides reducing the API)

I did not find an `lv_obj_get_class` function, only `lv_obj_check_type` and `lv_obj_has_class` which are less useful.  
Did I miss something?  

The reason I need `lv_obj_get_class` is that I need to associate a class with a widget on runtime according to the main class of `lv_obj_class_t`.
Current plan is to generate, on compile time, an array of pointers to the statically defined `lv_obj_class_t*` of all widget.  
Then, on runtime, when receiving `lv_obj_t` (in an event handler or as a returned value of some function) iterate the array and find its class. The index in the array where the class was found would be used to cast `obj` widget to the "real" widget.  
So the missing piece is some function that returns the widget class of an `lv_obj_t`.

After that, there are the [unticked points above](https://github.com/lvgl/lvgl/issues/1763#issue-689089014).
(assuming that all the points that are ticked are done)

Here is a summary of what I think is left after the discussion above:

- [ ] Public/Private interface H files, for minimizing the bindings program size
- [ ] Avoid ambiguity between struct members and struct member functions. (For example start in lv_anim_t)
- [ ] Function arguments that are pointers to struct_type should be declared in the function prototype as struct_type * and not void *.
For example, callbacks on lv_fs.h should receive lv_fs_file_t * file_p and not void * file_p.
- [ ] Widget name must not be a prefix of another widget name. For example, we can't have both dropdown and dropdown_list widgets.
- [ ] Checking the conventions on external widgets

kisvegabor, 2021-03-25:
> I did not find an lv_obj_get_class function, only lv_obj_check_type and lv_obj_has_class which are less useful.
Did I miss something?

I've just added it.

> (I'm assuming here that there is an extern for every lv_[widget]_class which is named exactly like this, so I can generate this array)

Yes.

> (assuming that all the points that are ticked are done)

Minor oversights are possible, but they should be done or can be fixed easily. 

>  Avoid ambiguity between struct members and struct member functions. (For example start in lv_anim_t)

I've updated `lv_anim_t` but there can be other ambiguous places too. 

> Function arguments that are pointers to struct_type should be declared in the function prototype as struct_type * and not void *.
For example, callbacks on lv_fs.h should receive lv_fs_file_t * file_p and not void * file_p.

`lv_fs_...` function do not recevie a pointer to `lv_fs_file_t` but a the pointer retunred by `lv_fs_open()`. `lv_fs_file_t` is used to store this retuned ppointer and other data required for LVGL. 

> Widget name must not be a prefix of another widget name. For example, we can't have both dropdown and dropdown_list widgets.

I've changed it to `lv_dropdownlist_class_t`.

amirgon, 2021-04-05:
I've just started looking at LVGL v8 (the new `master` branch) and discovered that **`user_data` is gone from `lv_obj_t`** .....
@kisvegabor - Is that intentional of an oversight? We use `user_data` extensively to bind Python objects to lv objects.

I also noticed that `lv_conf_template.h` changed significantly compared to v7, with some macros missing (`LV_ANTIALIAS` for example) and new macros born (`LV_MEM_CUSTOM_REALLOC` for example).  
I expected some changes, but there seem to be plenty of them.

embeddedt, 2021-04-05:
@amirgon Thanks for bringing this up; I also noticed this while playing around with ways to implement C++ inheritance. The current approach I've found is to manually extend the object's class at creation time:

```cpp
    lv_obj_class_t *cls = <whatever object is being created>;
    this->objClass.base_class = cls;
    this->objClass.instance_size = cls->instance_size + sizeof(widget *);
    this->objClass.signal_cb = widget::internal_signal_cb;
    this->handle = lv_obj_create_from_class(&this->objClass, lv_disp_get_scr_act(disp), nullptr);
```

After which, the `widget *` reference can be retrieved with this (very ugly) function:

```cpp
widget **get_widget_ref(lv_obj_t *obj) {
    return (widget **)(((char *)obj)+obj->class_p->base_class->instance_size);
}
```

Unfortunately, this doesn't help when working with an already-created object (like the active screen) as I don't think you can extend the class size post-creation anymore. I'm not pleased with this approach but haven't found a better one so far.

kisvegabor, 2021-04-06:
> I've just started looking at LVGL v8 (the new master branch) and discovered that user_data is gone from lv_obj_t .....
@kisvegabor - Is that intentional or an oversight? We use user_data extensively to bind Python objects to lv objects.

It was an oversight. I've added it back here 171b40a3

> I also noticed that lv_conf_template.h changed significantly compared to v7

Yes, I cleaned it up. Is it a problem with the extra work or do you miss something? If the former, I gladly help with migration. If the latter let me know about the issues you see.

> LV_ANTIALIAS for example

`LV_ANTIALIAS` and some similar configs that can be set in runtime in the display/indev driver were removed from `lv_conf.h`.

> LV_MEM_CUSTOM_REALLOC for example

I've added it for consistency because if a system provides `malloc/free` it certainly provides `realloc` as well, and we should use it. 

> I expected some changes, but there seem to be plenty of them.

Actually not that many. The majority of the changes come from the reorganization of the already existing configs. 
But again, if you need help let me know how can I help!

amirgon, 2021-04-07:
The LVGL bindings are generated by preprocessing `lvgl.h` and parsing it.  
The assumption is that `lvgl.h` represents the public API of the library.

Currently `lv_examples.h` is explicitly included in `lvgl.h`.  
Why is it?

https://github.com/lvgl/lvgl/blob/1ad767b152138e33a81e15c35efc3e93d660b5e0/lvgl.h#L75

It's a problem because we don't want to treat the examples as part of the API and generate Python wrappers for them.  
Are there any other headers included from `lvgl.h` that shouldn't be part of the API?

kisvegabor, 2021-04-08:
> The LVGL bindings are generated by preprocessing lvgl.h and parsing it.
The assumption is that lvgl.h represents the public API of the library.

Great!

> Currently lv_examples.h is explicitly included in lvgl.h.
Why is it?

It's for the users' convinence but I understand the problem in MP's case. Would you like to not generate binding for examples
 or generate a separate binding?

We have `LV_BUILD_EXAMPLES` in `lv_conf.h`. It can/should be used around `#include "examples/lv_examples.h"`.

> Are there any other headers included from lvgl.h that shouldn't be part of the API?

The others seem fine.

> The following functions have no implementation:

Removed here, thanks
25c464dc

[209 more replies]

closed 2022-05-28 by amirgon