lvgl/lvgl #1764: how to change the font of the label?

yangyongzhen, 2020-09-01:
# All enhancement requests or questions should be directed to the Forum.
I update the lvgl version form v5.3 to v7.4, 
change is a little big!
how to change the font of the label?
by v5.3 ,i can do this:
static lv_style_t label_style;
lv_style_copy(&label_style,&lv_style_plain_color);
label_style.text.font = &lv_font_roboto_16;
lv_label_set_style(label1,LV_LABEL_STYLE_MAIN,&label_style);
,but in v7.4, these apis are missing,
how can i change the label font by v7.4?
is this right?
   char *str="this is test!";
    lv_style_init(&my_style);
 
    /*Create a style and use the new font*/
    **lv_style_set_text_font(&my_style, LV_THEME_DEFAULT_FONT_NORMAL, &myFont);**

    /*Set a background color and a radius*/
    lv_style_set_radius(&my_style, LV_STATE_DEFAULT, 20);
    lv_style_set_bg_opa(&my_style, LV_STATE_DEFAULT, LV_OPA_COVER);
    lv_style_set_bg_color(&my_style, LV_STATE_DEFAULT, LV_COLOR_SILVER);
 
    /*Create a label and set new text*/
    lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
    lv_obj_set_pos(label, 100, 10);
    //lv_label_set_style(label, &my_style);
    lv_label_set_text(label, str );

kisvegabor, 2020-09-01:
If you are just updated to v7 I suggest reading this guide: https://docs.lvgl.io/latest/en/html/get-started/quick-overview.html#

---

We use GitHub Issues for development-related discussions.

Please ask your question in the [Forum](https://forum.lvgl.io) to make it visible for a larger community. You can simply log in with your GitHub account.

closed 2020-09-01 by kisvegabor

embeddedt, 2020-09-01:
```c
lv_style_set_text_font(&my_style, LV_STATE_DEFAULT, &myFont);
```

Or, if you want to change the default font, change `LV_THEME_DEFAULT_FONT_NORMAL` in `lv_conf.h`.