lvgl/lvgl #1765: Cursor for chart widget

ali-rostami, 2020-09-01:
Hi
Ads @kisvegabor suggested [here](https://github.com/lvgl/lvgl/issues/1338#issuecomment-679865013), I opened this issue to talk about implementing this feature for charts.

I want to implement the cursor as a dashed/solid line with the same length as the series_area. I think its the most simple form of the cursor that we can have.
It seems that for drawing it we need to add  `LV_CHART_PART_CURSOR` so we can then add style for it.
If I got it right, in its style we can define the dashed or solid, its color, transparency, and ...
We can enable or disable the cursor by making its style transparent or not.
So what do you think @kisvegabor ?

kisvegabor, 2020-09-02:
Hi,

Thanks for opening the issue.

> I want to implement the cursor as a dashed/solid line with the same length as the series_area. I think it's the most simple form of the cursor that we can have.

Optionally it should draw a horizontal or vertical line or both as well. 

> It seems that for drawing it we need to add LV_CHART_PART_CURSOR so we can then add style for it.
If I got it right, in its style we can define the dashed or solid, its color, transparency, and ...

Yes.

---

My initial view about the usage of this feature is this:
- Have two functions:
  - `lv_chart_get_coord_from_index(chart, ser, index, &point_out)` Gets where a point is. Useful to place a cursor directly to a point
  - `int index = lv_chart_get_index_from_coord(chart, ser, &point_in)` Gets the nearest point to a coordinates. Useful if you pressed the chart and want to place the cursor there.
- For best flexibility, I think the best would be to set the cursor manually in pixels. `lv_chart_get_coord_from_index` can be used to find the coordinates a point.  
  - `lv_chart_set_cursor_x(chart, 100);`
  - `lv_chart_set_cursor_y(chart, 100);`
- If a box should appear to tell what is on the current cursor position the user can create it manually on top of the chart. I think there are too many options regarding this box to create a general solution for it. 

What do you think?

ali-rostami, 2020-09-02:
So, if the user wants to display the cursor, He should call these:
```
lv_chart_get_index_from_coord //get the nearest index.
lv_chart_get_coord_from_index //get the coord of the found index.
lv_chart_set_cursor_x //set the cursor based on the coord.
```
in the click callback of the chart. right?

----------------

what am I doing wrong here? :

```
void lv_chart_set_cursor_x(lv_obj_t * chart, lv_coord_t coord_cursor_x)
{

	lv_area_t col_a;

	lv_area_t series_area;
	get_series_area(chart, &series_area);

	if(coord_cursor_x < series_area.x1 || coord_cursor_x > series_area.x2) return;

	lv_draw_rect_dsc_t col_dsc;
	lv_draw_rect_dsc_init(&col_dsc);
	lv_obj_init_draw_rect_dsc(chart, LV_CHART_PART_BG, &col_dsc);
	col_dsc.bg_grad_dir = LV_GRAD_DIR_NONE;
	col_dsc.bg_opa = LV_OPA_COVER;
	col_dsc.bg_color = LV_COLOR_WHITE;

	/*Make the cursor longer with `radius` to clip the rounding from the bottom*/
	col_a.y1 = series_area.y1;
	col_a.y2 = series_area.y2;

	col_a.x1 = coord_cursor_x;
	col_a.x2 = col_a.x1 + lv_obj_get_style_line_width(chart, LV_CHART_PART_BG);

	lv_draw_rect(&col_a, &series_area, &col_dsc);
}
```

kisvegabor, 2020-09-02:
> in the click callback of the chart. right?

Yes, I meant that.

> what am I doing wrong here? :

You can draw only in the design function of the chart.

ali-rostami, 2020-09-02:
Oh, what about this:
I will call this:
` draw_cursor(chart, &series_area, clip_area);`
In the `lv_chart_design` function, and if the cursor was transparent it will return.

we can add a `cursor_index` to lv_chart_ext_t structure. and based on it we draw the cursor. I like it more if you see so, I will implement it like this otherwise I will work on the way you said.

kisvegabor, 2020-09-03:
> draw_cursor(chart, &series_area, clip_area);
In the lv_chart_design function, and if the cursor was transparent it will return.

Perfect!

> we can add a cursor_index to lv_chart_ext_t structure. and based on it we draw the cursor

I see 2 issues with using point index instead of pixels:
1. pixel unit gives more freedom. E.g. you can indicate a max value with the cursor independently from the points.
2. What if you have 2 series and you want to add a horizontal cursor. The 3rd point's value is 30 on series 1 and 50 or series 2. To which should the horizontal cursor to be drawn? To the smallest, largest, or average? 

BTW, what do you think about having 2 cursors per axis? They could indicate a range, or people can use it to measure the difference between points.

kisvegabor, 2020-09-04:
> Forexample some one wants to draw 1 white ver line as cursor and 3 hor line in red and blue for max, min and white for cursor.
What do you think?

Sounds reasonable. How could its API look like?

What about something like this?
```c
lv_chart_cursor_t * cur1 = lv_chart_add_cursor_hor(chart, LV_COLOR_RED);
lv_cursor_set_pos(chart, cur1, 100);
```

ali-rostami, 2020-09-09:
> What about something like this?
> 
> ```c
> lv_chart_cursor_t * cur1 = lv_chart_add_cursor_hor(chart, LV_COLOR_RED);
> lv_cursor_set_pos(chart, cur1, 100);
> ```

Greate. So you mean instead of `lv_chart_set_cursor_x` or 'lv_chart_set_cursor_y' use `lv_cursor_set_pos`?

kisvegabor, 2020-09-09:
> So you mean instead of lv_chart_set_cursor_x or 'lv_chart_set_cursor_y' use lv_cursor_set_pos?

Yes, because `lv_chart_add_cursor_hor/ver` already expresses the direction, so the cursor can know its axis.

kisvegabor, 2020-10-02:
I close it as the discussion has continued in #1789 PR.

closed 2020-10-02 by kisvegabor