lvgl/lvgl #1767: Warn: Couldn't allocate memory

giuseppelafortezza, 2020-09-02:
After some time of execution the program stops with:
**Warn: Couldn't allocate memory**

I have build the app in debug and run it with gdb.

0x00005555555a2416 in lv_draw_map (map_area=map_area@entry=0x7fffffffd8b8, clip_area=clip_area@entry=0x7fffffffd7b0, map_p=<optimized out>, draw_dsc=draw_dsc@entry=0x7fffffffd8c0, chroma_key=chroma_key@entry=false, alpha_byte=alpha_byte@entry=true) at /home/sdk/workspace/test_app/build/../lvgl/src/lv_draw/lv_draw_img.c:419
419	                        map2[px_i].full =  *((uint32_t *)map_px);

lv_draw_map (map_area=map_area@entry=0x7fffffffd8b8, clip_area=clip_area@entry=0x7fffffffd7b0, map_p=<optimized out>, draw_dsc=draw_dsc@entry=0x7fffffffd8c0, chroma_key=chroma_key@entry=false, alpha_byte=alpha_byte@entry=true)
    at /home/sdk/workspace/test_app/build/../lvgl/src/lv_draw/lv_draw_img.c:556
556	                                map2[px_i].full = 0;

I have the same problem with version 7.3.1 and 7.4.0.

In my application I'm using labels, animations, masks and images.

I have tried to increase LV_MEM_SIZE but the behaviour is the same.

kisvegabor, 2020-09-02:
It can be a  memory corruption too. 
Enable `LV_USE_DEBUG` and `LV_USE_ASSERT_MEM_INTEGRITY` in `lv_conf.h`.

giuseppelafortezza, 2020-09-02:
This is the output:

Warn: Couldn't allocate memory 	(lv_mem.c #208 lv_mem_alloc())
Warn: Couldn't allocate memory 	(lv_mem.c #323 lv_mem_realloc())
Error: _lv_mem_buf_get 	(lv_mem.c #533 _lv_mem_buf_get())
Error: Out of memory, can't allocate a new  buffer (increase your LV_MEM_SIZE/heap size (0x0000000000000000) 	(lv_debug.c #127 lv_debug_log_error())

And this is the stacktrace:

0x000055555558b2d5 in _lv_mem_buf_get (size=size@entry=3200) at /home/sdk/workspace/test_app/build/../lvgl/src/lv_misc/lv_mem.c:533
533	                LV_DEBUG_ASSERT(false, "Out of memory, can't allocate a new  buffer (increase your LV_MEM_SIZE/heap size", 0x00);
(gdb) bt
#0  0x000055555558b2d5 in _lv_mem_buf_get (size=size@entry=3200) at /home/sdk/workspace/test_app/build/../lvgl/src/lv_misc/lv_mem.c:533
#1  0x00005555555a1441 in lv_draw_map (map_area=map_area@entry=0x7fffffffd898, clip_area=clip_area@entry=0x7fffffffd790, map_p=<optimized out>, draw_dsc=draw_dsc@entry=0x7fffffffd8a0, 
    chroma_key=chroma_key@entry=false, alpha_byte=alpha_byte@entry=true) at /home/sdk/workspace/test_app/build/../lvgl/src/lv_draw/lv_draw_img.c:453
#2  0x00005555555a2057 in lv_img_draw_core (draw_dsc=0x7fffffffd8a0, src=<optimized out>, clip_area=0x7fffffffd890, coords=0x7fffffffd898)
    at /home/sdk/workspace/test_app/build/../lvgl/src/lv_draw/lv_draw_img.c:272

I'm using versione 7.4.0 in this test.

kisvegabor, 2020-09-02:
Please add these line before the assert [here](https://github.com/lvgl/lvgl/blob/05960885440e7ac412ba7d4b5e7c46af9716134c/src/lv_misc/lv_mem.c#L533):
```c
printf("size: %d\n", size);
LV_ASSERT_MEM_INTEGRITY();

lv_mem_monitor_t mon;
lv_mem_monitor(&mon);
printf("used: %6d (%3d %%), frag: %3d %%, biggest free: %6d\n",
                       (int)mon.total_size - mon.free_size, mon.used_pct, mon.frag_pct,
                       (int)mon.free_biggest_size);

```

giuseppelafortezza, 2020-09-02:
This is the result:

Warn: Couldn't allocate memory 	(lv_mem.c #208 lv_mem_alloc())
Warn: Couldn't allocate memory 	(lv_mem.c #323 lv_mem_realloc())
size: 3200
used:  59424 ( 91 %), frag:  49 %, biggest free:   3136
Error: _lv_mem_buf_get 	(lv_mem.c #541 _lv_mem_buf_get())
Error: Out of memory, can't allocate a new  buffer (increase your LV_MEM_SIZE/heap size (0x0000000000000000) 	(lv_debug.c #127 lv_debug_log_error())

kisvegabor, 2020-09-03:
> used: 59424 ( 91 %),

It seems you are really out of memory.

giuseppelafortezza, 2020-09-09:
Thanks for your support. I solved it.

closed 2020-09-09 by giuseppelafortezza

zhaojiawen1996, 2021-04-22:
Hi,How did you solve this problem? thank you

famm2005, 2022-12-16:
Hi,How did you solve this problem? thank you

junqingqiao, 2023-03-26:
Hi,How did you solve this problem? thank you

codegrue, 2024-02-27:
@giuseppelafortezza  please post what your solution was.