lvgl/lvgl #1773: [suggesstion] refactor the name of lv_tabview_btns_pos_t

incity, 2020-09-07:
[image]

1. in the enum, use TABVIEW_TAB, but the type use name tabview_btns, i think this is not a good name.
2. position(POS) that means you can set the coordinates of x and y, but actually you can't.

DOCK is more meaningful.

enum {
    LV_TABVIEW_TABS_DOCK_TOP,
    LV_TABVIEW_TABS_DOCK_BOTTOM,
    LV_TABVIEW_TABS_DOCK_LEFT,
    LV_TABVIEW_TABS_DOCK_RIGHT
};
typedef uint8_t lv_tabview_tabs_dock_t;

GRAVITY is also a good alternative

enum {
    LV_TABVIEW_TABS_GRAVITY_TOP,
    LV_TABVIEW_TABS_GRAVITY_BOTTOM,
    LV_TABVIEW_TABS_GRAVITY_LEFT,
    LV_TABVIEW_TABS_GRAVITY_RIGHT
};
typedef uint8_t lv_tabview_tabs_gravity_t;

incity, 2020-09-08:
[image]

incity, 2020-09-08:
[image]

embeddedt, 2020-09-08:
I agree with changing the last two (please send a PR); @kisvegabor should comment on the first one.

kisvegabor, 2020-09-09:
Regarding the rename suggestion, the tabview will be reworked in v8. I'll consider it during this rework.

incity, 2020-09-10:
@kisvegabor will the tabview support any number of tabs in v8?

kisvegabor, 2020-09-11:
It supports any number of tabs in v7 too. Or what do you mean?

incity, 2020-09-11:
> It supports any number of tabs in v7 too. Or what do you mean?

It will cover each other when add too many tabs or tabs name too long

kisvegabor, 2020-09-11:
What would you expect instead? Multiple rows, or scrolling gon the header part?

incity, 2020-09-14:
it's more better to do a auto scroll as in android

kisvegabor, 2020-09-14:
So scrolling on the buttons, and snap the active button to center?

incity, 2020-09-15:
there is a demo here @kisvegabor 

https://material.io/components/tabs

[image]

kisvegabor, 2020-09-15:
Got it, thanks!

It can be demonstrated in a component.

incity, 2020-09-15:
@kisvegabor  i found you will remove lv_page and make all object scrollable(even the rotation), im curious that if it's possible to make all object also support scalable as Android's View(the base object of the widgets) does.
[image]
[image]

this can also help the canvas do scale transformation but not just do zoom transformation.

https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/view/View.java

kisvegabor, 2020-09-16:
I'm planning for a long time to make it possible to create screenshots from objects (with all children) to allow transformations (zoom, scale, blur, etc). It should be an internal mechanism, so if LVGL sees that a child has changed, it updates the screenshot.

In the future please use the [Future request category](https://forum.lvgl.io/c/feature-request) of the forum for similar topics.
If you have comments to the scaling topic, please post it in the forum and continue there.

closed 2020-09-16 by kisvegabor