lvgl/lvgl #1775: LV_THEME_DEFAULT_FONT_* cannot be used with custom fonts

upbeat27, 2020-09-08:
**Describe the bug**

The `LV_THEME_DEFAULT_FONT_*` defines in `lv_conf.h` can only be used with fonts included with LVGL, which greatly limits their usefulness.

**To Reproduce**

Try to use a custom font instead. It won't work, because (for example) `LV_THEME_DEFAULT_FONT_NORMAL` is used in a lot of places (`lv_draw_label.c`, `lv_draw_rect.c`, etc.) that have no visibility of the custom font. That is, there is no way to make it so these c files can see the custom font declaration.

There is `LV_THEME_DEFAULT_INCLUDE` but including the font declaration there does not work, as that is only included in `lv_obj.c`.

**Expected behavior**

It should be possible to use custom fonts as the default fonts.

**Screenshots**
N/A

**Additional context**

Suggestion: include `LV_THEME_DEFAULT_INCLUDE` everywhere that references any `LV_THEME_DEFAULT_FONT_*` so you can declare custom fonts in that file.

pete-pjb, 2020-09-09:
Hi @upbeat27 ,

I personally use custom fonts for the whole of my build, in my lv_conf.h file I have this:

```c
/* Optionally declare your custom fonts here.
 * You can use these fonts as default font too
 * and they will be available globally. E.g.
 * #define LV_FONT_CUSTOM_DECLARE LV_FONT_DECLARE(my_font_1) \
 *                                LV_FONT_DECLARE(my_font_2)
 *
 */
#define LV_FONT_CUSTOM_DECLARE			LV_FONT_DECLARE(lv_font_roboto_8_p)\
										LV_FONT_DECLARE(lv_font_roboto_12_p)\
										LV_FONT_DECLARE(lv_font_roboto_16_p)\
										LV_FONT_DECLARE(lv_font_roboto_16_bi)\
										LV_FONT_DECLARE(lv_font_roboto_22_bi)

/* A fast and impressive theme.
 * Flags:
 * LV_THEME_MATERIAL_FLAG_LIGHT: light theme
 * LV_THEME_MATERIAL_FLAG_DARK: dark theme
 * LV_THEME_MATERIAL_FLAG_NO_TRANSITION: disable transitions (state change animations)
 * LV_THEME_MATERIAL_FLAG_NO_FOCUS: disable indication of focused state)
 * */
 #define LV_USE_THEME_MATERIAL    1

/* Mono-color theme for monochrome displays.
 * If LV_THEME_DEFAULT_COLOR_PRIMARY is LV_COLOR_BLACK the
 * texts and borders will be black and the background will be
 * white. Else the colors are inverted.
 * No flags. Set LV_THEME_DEFAULT_FLAG 0 */
 #define LV_USE_THEME_MONO        0

 #define LV_THEME_DEFAULT_INCLUDE            <stdint.h>      /*Include a header for the init. function*/
 #define LV_THEME_DEFAULT_INIT               lv_theme_material_init
 #define LV_THEME_DEFAULT_COLOR_PRIMARY      LV_COLOR_RED
 #define LV_THEME_DEFAULT_COLOR_SECONDARY    LV_COLOR_BLUE
 #define LV_THEME_DEFAULT_FLAG               LV_THEME_MATERIAL_FLAG_LIGHT
 #define LV_THEME_DEFAULT_FONT_SMALL         &lv_font_roboto_12_p
 #define LV_THEME_DEFAULT_FONT_NORMAL        &lv_font_roboto_16_p
 #define LV_THEME_DEFAULT_FONT_SUBTITLE      &lv_font_roboto_16_bi
 #define LV_THEME_DEFAULT_FONT_TITLE         &lv_font_roboto_22_bi
```

This appears to work well for me, is this what you are trying to achieve?

Kind Regards,

Pete

upbeat27, 2020-09-09:
Holy moly I was so focused on the `LV_THEME_DEFAULT_INCLUDE` I didn't realize there was a specific spot for including custom fonts in `lv_conf.h`.

closed 2020-09-09 by upbeat27

pete-pjb, 2020-09-09:
No worries, it happens to all of us! 😃