lvgl/lvgl #1777: When clip corner is true, there is some leakage (not fully clipped)

upbeat27, 2020-09-09:
**Describe the bug**

Some leakage occurs outside the corners when it should be clipped.
It is a like a kid coloring outside the lines in their coloring book :)

**To Reproduce**

```c
	static const char *btn_map[] = { "1", "2", "" };
	lv_obj_t * obj = lv_btnmatrix_create(TUNER_BG, NULL);

	lv_obj_set_size(obj, TUNER_MENU_WIDTH, 50);
	lv_btnmatrix_set_map(obj, btn_map);
	lv_btnmatrix_set_btn_ctrl_all(obj, LV_BTNMATRIX_CTRL_CHECKABLE);
	lv_btnmatrix_set_one_check(obj, true);
	lv_obj_set_style_local_pad_all(obj, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, 0);
	lv_obj_set_style_local_pad_inner(obj, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, 0);
	lv_obj_set_style_local_clip_corner(obj, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, true);
	lv_obj_set_style_local_border_width(obj, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, 1);
	lv_obj_set_style_local_radius(obj, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, 20);
	lv_obj_set_style_local_border_side(obj, LV_BTNMATRIX_PART_BTN, LV_STATE_DEFAULT, LV_BORDER_SIDE_FULL | LV_BORDER_SIDE_INTERNAL);
	lv_obj_set_style_local_radius(obj, LV_BTNMATRIX_PART_BTN, LV_STATE_DEFAULT, 0);
```

**Expected behavior**

The clipping mask should prevent any leakage outside the border.

**Screenshots**

On the right side, you can see the orange outside the border:
[image]

**Additional context**

This is observable in other places too, like lists, etc. I think it is pretty wide spread in general. It is less obvious in certain circumstances, depending on the colors, radius, etc.

embeddedt, 2020-09-09:
Looks like an off-by-one error on the X axis. I'm not sure if the border is too short or the background is too long.

kisvegabor, 2020-09-10:
Thanks for the report. I've pushed a fix for it.

closed 2020-09-10 by kisvegabor

upbeat27, 2020-09-10:
This certainly fixes the leakage, but somehow this breaks others things.
125e6e95f60d8cf32040d73dbdd9554afea77581 opening a dropdown results in crash (try lv_demo_widgets)
c683a3612dfb1836a38d3bf9a04d2254e58ffce2 (the previous commit) dropdown opens fine

commit 125e6e95 (2020-09-10): fix color bleeding on corner clipping
Fixes #1777