lvgl/lvgl #1779: Missing GC roots on lv_font_fmt_txt.c

amirgon, 2020-09-10:
See the bug report here::  https://forum.lvgl.io/t/gc-error-and-reboot-with-last-version-of-lv-bindings/3222

Micropython crashes with memory corruption.
Problem was identified in `lv_font_fmt_txt.c` which allocates memory dynamically and stores it in global pointers like this:

https://github.com/lvgl/lvgl/blob/c683a3612dfb1836a38d3bf9a04d2254e58ffce2/src/lv_font/lv_font_fmt_txt.c#L50

https://github.com/lvgl/lvgl/blob/c683a3612dfb1836a38d3bf9a04d2254e58ffce2/src/lv_font/lv_font_fmt_txt.c#L113

So - need to declare `decompr_buf` as a gc-root.

kisvegabor, 2020-09-11:
Added here c55b1651.

amirgon, 2020-09-11:
@kisvegabor The Micropython CI fails with https://github.com/lvgl/lvgl/commit/c55b165150c97d48b9ca086b0f651d46cf2dffbb, did you notice?
 
```
../../lib/lv_bindings/lvgl/src/lv_font/lv_font_fmt_txt.c: In function ‘lv_font_get_bitmap_fmt_txt’:
../../lib/lv_bindings/lvgl/src/lv_font/lv_font_fmt_txt.c:116:29: error: ‘_lv_font_decompr_buf’ undeclared 
```
So the CI for Micropython proves itself useful!   

I think you should use `LV_GC_ROOT(_lv_font_decompr_buf)` instead of `_lv_font_decompr_buf`

kisvegabor, 2020-09-14:
Oh, I thought it's still the not found commit issue.

Fixed in master.

amirgon, 2020-09-14:
> Oh, I thought it's still the not found commit issue.

That was fixed on #1783, I hope the missing commits issue is now resolved.

kisvegabor, 2020-09-14:
Great! 

It seems the GC issue is also resolved, so I close it.

closed 2020-09-14 by kisvegabor