lvgl/lvgl #1782: LV_ASSERT_OBJ(obj, LV_OBJX_NAME) on latest LVGL

amirgon, 2020-09-10:
**Describe the bug**

When clicking on a dropdown list, `LV_ASSERT_OBJ(obj, LV_OBJX_NAME)` on `lv_obj.c`:`3098` is triggered

**To Reproduce**

- Clone Micropython
- Update LVGL to HEAD of master branch
- Build the unix port
- Run: `ports/unix/micropython -i lib/lv_bindings/examples/advanced_demo.py`
- Click on the dropdown list.

**Additional context**

Stack trace:
```
#0  0x00000000004ca86d in lv_obj_get_state (obj=0x0, part=0 '\000') at ../../lib/lv_bindings/lvgl/src/lv_core/lv_obj.c:3098
#1  0x00000000004912fa in lv_dropdown_signal (ddlist=0x7ffff17d0c60, sign=9 '\t', param=0x7fffffffd8c0) at ../../lib/lv_bindings/lvgl/src/lv_widgets/lv_dropdown.c:895
#2  0x00000000004c89e9 in lv_signal_send (obj=0x7ffff17d0c60, signal=9 '\t', param=0x7fffffffd8c0) at ../../lib/lv_bindings/lvgl/src/lv_core/lv_obj.c:2037
#3  0x00000000004ca89f in lv_obj_get_state (obj=0x7ffff17d0c60, part=64 '@') at ../../lib/lv_bindings/lvgl/src/lv_core/lv_obj.c:3108
#4  0x00000000004c9c89 in _lv_obj_get_style_int (obj=0x7ffff17d0c60, part=64 '@', prop=1) at ../../lib/lv_bindings/lvgl/src/lv_core/lv_obj.c:2655
#5  0x00000000004c3a6c in lv_obj_get_style_radius (obj=0x7ffff17d0c60, part=64 '@') at ../../lib/lv_bindings/lvgl/src/lv_core/lv_obj_style_dec.h:102
#6  0x00000000004caf9a in lv_obj_init_draw_rect_dsc (obj=0x7ffff17d0c60, part=64 '@', draw_dsc=0x7fffffffda20) at ../../lib/lv_bindings/lvgl/src/lv_core/lv_obj.c:3359
#7  0x00000000004ced5d in style_snapshot (obj=0x7ffff17d0c60, part=64 '@', shot=0x7fffffffda20) at ../../lib/lv_bindings/lvgl/src/lv_core/lv_obj.c:4765
#8  0x00000000004c7f7f in lv_obj_set_state (obj=0x7ffff17d0c60, new_state=16 '\020') at ../../lib/lv_bindings/lvgl/src/lv_core/lv_obj.c:1739
#9  0x00000000004c8439 in lv_obj_add_state (obj=0x7ffff17d0c60, state=16 '\020') at ../../lib/lv_bindings/lvgl/src/lv_core/lv_obj.c:1822
#10 0x00000000004ccb4f in lv_obj_signal (obj=0x7ffff17d0c60, sign=11 '\v', param=0x7ffff17ced00) at ../../lib/lv_bindings/lvgl/src/lv_core/lv_obj.c:4004
#11 0x0000000000491223 in lv_dropdown_signal (ddlist=0x7ffff17d0c60, sign=11 '\v', param=0x7ffff17ced00) at ../../lib/lv_bindings/lvgl/src/lv_widgets/lv_dropdown.c:878
#12 0x00000000004d40b7 in indev_proc_press (proc=0x7ffff17ced38) at ../../lib/lv_bindings/lvgl/src/lv_core/lv_indev.c:925
#13 0x00000000004d2aeb in indev_pointer_proc (i=0x7ffff17ced00, data=0x7fffffffddf0) at ../../lib/lv_bindings/lvgl/src/lv_core/lv_indev.c:414
#14 0x00000000004d2429 in _lv_indev_read_task (task=0x7ffff17cedc0) at ../../lib/lv_bindings/lvgl/src/lv_core/lv_indev.c:110
#15 0x00000000004615e5 in lv_task_exec (task=0x7ffff17cedc0) at ../../lib/lv_bindings/lvgl/src/lv_misc/lv_task.c:409
#16 0x0000000000460f1d in lv_task_handler () at ../../lib/lv_bindings/lvgl/src/lv_misc/lv_task.c:142
#17 0x00000000005467a5 in mp_lv_task_handler (arg=0x6) at ../../lib/lv_bindings/driver/SDL/modSDL.c:26
#18 0x00000000004330b3 in fun_builtin_1_call (self_in=0x59b200 <mp_lv_task_handler_obj>, n_args=1, n_kw=0, args=0x7fffffffdf30) at ../../py/objfun.c:71
#19 0x0000000000427b2f in mp_call_function_n_kw (fun_in=0x59b200 <mp_lv_task_handler_obj>, n_args=1, n_kw=0, args=0x7fffffffdf30) at ../../py/runtime.c:650
#20 0x0000000000427a78 in mp_call_function_1 (fun=0x59b200 <mp_lv_task_handler_obj>, arg=0x6) at ../../py/runtime.c:628
#21 0x0000000000429d6f in mp_call_function_1_protected (fun=0x59b200 <mp_lv_task_handler_obj>, arg=0x6) at ../../py/runtime_utils.c:33
#22 0x0000000000429f93 in mp_handle_pending_tail (atomic_state=0) at ../../py/scheduler.c:92
#23 0x0000000000429f03 in mp_handle_pending (raise_exc=true) at ../../py/scheduler.c:76
#24 0x000000000045c096 in mp_hal_stdin_rx_chr () at unix_mphal.c:170
#25 0x0000000000547567 in readline (line=0x7fffffffe0b0, prompt=0x56d85e ">>> ") at ../../lib/mp-readline/readline.c:512
#26 0x000000000045ab81 in do_repl () at main.c:194
#27 0x000000000045bc1d in main_ (argc=3, argv=0x7fffffffe348) at main.c:662
#28 0x000000000045b1a7 in main (argc=3, argv=0x7fffffffe348) at main.c:431
```

It looks like `ext->page` was somehow set to  NULL, on -

https://github.com/lvgl/lvgl/blob/3f36139469253e049f3be486f380b22606c462d7/src/lv_widgets/lv_dropdown.c#L895

@kisvegabor - did you change something related to dropdown or signals? 
I suspect (but not sure) this is not related to a missing gc_root, because a missing gc_root usually causes corrupted RAM and not in NULL pointer access.

upbeat27, 2020-09-10:
See latest comment in #1777, the offending commit has nothing to do with dropdown (using bisect, didn't have to go back far before it started working again).
It is indeed very strange, I couldn't figure how the changes to border clipping could cause this, but I am sure @kisvegabor can figure it out.

kisvegabor, 2020-09-11:
Seemingly I accidentally merged some code from the dev branch to master. I removed them and it works well for me now.

It should also fix #1777

closed 2020-09-21 by amirgon