lvgl/lvgl #1785: Screen size setting

zdp1994, 2020-09-11:
Now the screen size Settings are defined through macros, fixed size, how to achieve screen size adaptive, automatic screen size, set size.
thanks

kisvegabor, 2020-09-11:
Answered here: https://github.com/lvgl/lv_examples/issues/73

closed 2020-09-11 by kisvegabor

zdp1994, 2020-11-16:
Sorry to reply to you so late. I have been unable to log in because of a problem with my account some time ago, and it has just been resolved recently.
For the problem of screen size adaptation, I have my own solution. Here I define the macros LV_HOR_RES_MAX and LV_VER_RES_MAX as variables, and then assign them in lvdrv. Currently, it has been implemented in three screens of different sizes. After verification, it can be displayed in full screen