lvgl/lvgl #1787: lv_demo_widgets hit dropdown list, program will crash.

ftts12, 2020-09-11:
static lv_res_t lv_dropdown_signal(lv_obj_t * ddlist, lv_signal_t sign, void * param)
{  
  else if(sign == LV_SIGNAL_GET_STATE_DSC) {
        lv_get_state_info_t * info = param;
        if((info->part == LV_DROPDOWN_PART_LIST) ||
           (info->part == LV_DROPDOWN_PART_SCROLLBAR) ||
           (info->part == LV_DROPDOWN_PART_SELECTED) )
		{
			if (ext->page != NULL)//add check page 
			{
				info->result = lv_obj_get_state(ext->page, LV_PAGE_PART_BG);
			}
            
        }
    }

ftts12, 2020-09-11:
ext->page need check if it is NULL.

embeddedt, 2020-09-11:
Duplicate of #1782. Please try with the latest branch.

kisvegabor, 2020-09-14:
I close it as duplicated.

closed 2020-09-14 by kisvegabor