lvgl/lvgl #1790: textarea does get defocus event in group

liebman, 2020-09-12:
**Describe the bug**

On a touchscreen, with two textarea objects placed in a group, clicking on the non-focused textarea highlights it and does not un-highlight the other.  I debugger shows that no DEFOCUS event is sent.  It will un-highlight if you press on the container outside of the textarea, drag the pointer to inside of the non-focused one (you have to remember which that is as both are highlighted) and release.

**To Reproduce**

```
    lv_obj_t* cont = lv_cont_create(lv_scr_act(), nullptr);
    lv_cont_set_layout(cont, LV_LAYOUT_COLUMN_LEFT);
    lv_cont_set_fit(cont, LV_FIT_PARENT);
    lv_obj_align(cont, nullptr, LV_ALIGN_CENTER, 0,0 );

    lv_obj_t* ta1 = lv_textarea_create(cont, nullptr);
    lv_obj_set_width(ta1, 100);
    lv_textarea_set_one_line(ta1, true);
    lv_textarea_set_cursor_hidden(ta1, true);

    lv_obj_t* ta2 = lv_textarea_create(cont, nullptr);
    lv_obj_set_width(ta2, 100);
    lv_textarea_set_one_line(ta2, true);
    lv_textarea_set_cursor_hidden(ta2, true);

    lv_group_t* group = lv_group_create();
    lv_group_add_obj(group, ta1);
    lv_group_add_obj(group, ta2);
```
**Expected behavior**

Only one should be highlighted at a time.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

Add any other context about the problem here.

kisvegabor, 2020-09-14:
It works well for me with the latest master.

There were some related fixes in the near past, please update LVGL and try again.

kisvegabor, 2020-10-07:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2020-10-07 by kisvegabor