lvgl/lvgl #1793: lv_canvas with indexed color format has issues

limitz, 2020-09-15:
The current code on master (and dev) has numerous problems dealing with indexed formats (containing a palette), when the number of bits < 8 for instance, the following line in lv_canvas_copy_buf will produce 0 and copy nothing:

https://github.com/lvgl/lvgl/blob/b60714a710e811a815a73bdf72dcd31965d0e407/src/lv_widgets/lv_canvas.c#L251

Also, it seems that the sizes that are calculated for the indexed data is off by one, when the width can be divided by the number of pixels per byte. And the palette size is not taken into account.

The current documentation gives an example of how to calculate the size needed for the buffer to the canvas. This function is also incorrect for indexed formats.

I have pushed my changes here, i might put some more work into it and make a proper pull request. But at least it will show the issue: https://github.com/lvgl/lvgl/commit/3295ef1c1accdc44f7d707e7f37423e8b4683cb3

**To Reproduce**
Sorry, this code shows what works with the changes I made, as an indication what wouldn't work without them. Hope it is useful anyway.

```
 lv_coord_t w = 140;
        lv_coord_t h = 140;
        lv_coord_t wc = 80; 
        lv_coord_t hc = 80; 

        int type = LV_IMG_CF_INDEXED_4BIT;

        int bpp= lv_img_cf_get_px_size(type);
        int doc_nbytes = ((bpp * w) >> 3) * h;
        int act_nbytes = lv_img_buf_get_img_size(w,  h,  type); 
        int act_tocopy = lv_img_buf_get_img_size(wc, hc, type);
        int src_stride = lv_img_buf_get_img_stride(wc, type);
        int dst_stride = lv_img_buf_get_img_stride(w, type);
    
        printf("bpp %d - doc bytes %d - act nbytes %d - act tocopy %d\n",
                bpp,     doc_nbytes,    act_nbytes,     act_tocopy);

        void* pbuf = heap_caps_calloc(act_nbytes,1, MALLOC_CAP_8BIT);
        void* cbuf = heap_caps_calloc(act_tocopy, 1, MALLOC_CAP_8BIT);

        memset(pbuf, 0x11,  act_nbytes);
        for (int i=0; i<h; i++)
        {
                for (int j=0; j<w; j++)
                {
                        uint8_t* ptr = (uint8_t*) pbuf;
                        ptr += lv_img_buf_get_palette_size(type);
                        ptr[i*dst_stride + j * bpp / 8] = 0xFF;
                }
        }
    
        canvas = lv_canvas_create(lv_scr_act(), NULL);
        lv_canvas_set_buffer(canvas, pbuf, w, h, type);
    
        int pentries = lv_img_buf_get_palette_size(type) / sizeof(lv_color32_t);
        printf("Number of palette entries: %d\n", pentries);
lv_canvas_set_palette(canvas, 0x0, LV_COLOR_MAKE( 0x33, 0x00, 0xCC )); 
        lv_canvas_set_palette(canvas, 0x1, LV_COLOR_MAKE( 0x66, 0x00, 0x99 )); 
        lv_canvas_set_palette(canvas, 0x2, LV_COLOR_MAKE( 0x99, 0x00, 0x66 )); 
        lv_canvas_set_palette(canvas, 0x3, LV_COLOR_MAKE( 0xCC, 0x00, 0x33 )); 

        lv_canvas_set_palette(canvas, 0x4, LV_COLOR_MAKE( 0xFF, 0x33, 0x00 )); 
        lv_canvas_set_palette(canvas, 0x5, LV_COLOR_MAKE( 0xFF, 0x66, 0x00 )); 
        lv_canvas_set_palette(canvas, 0x6, LV_COLOR_MAKE( 0xFF, 0x99, 0x00 )); 
        lv_canvas_set_palette(canvas, 0x7, LV_COLOR_MAKE( 0xFF, 0xCC, 0x00 )); 

        lv_canvas_set_palette(canvas, 0x8, LV_COLOR_MAKE( 0xCC, 0xFF, 0x33 )); 
        lv_canvas_set_palette(canvas, 0x9, LV_COLOR_MAKE( 0x99, 0xFF, 0x66 )); 
        lv_canvas_set_palette(canvas, 0xA, LV_COLOR_MAKE( 0x66, 0xFF, 0x99 )); 
        lv_canvas_set_palette(canvas, 0xB, LV_COLOR_MAKE( 0x33, 0xFF, 0xCC )); 

        lv_canvas_set_palette(canvas, 0xC, LV_COLOR_MAKE( 0x00, 0xCC, 0xFF )); 
        lv_canvas_set_palette(canvas, 0xD, LV_COLOR_MAKE( 0x00, 0x99, 0xFF ));
        lv_canvas_set_palette(canvas, 0xE, LV_COLOR_MAKE( 0x00, 0x66, 0xFF ));
        lv_canvas_set_palette(canvas, 0xF, LV_COLOR_MAKE( 0x00, 0x44, 0xFF ));

        for (int pent = 0; pent < pentries; pent++)
        {
                printf("%X: %08X\n", pent, ((uint32_t*)pbuf)[pent]);
        }
        lv_obj_invalidate(canvas);

        int skip_palette = lv_img_buf_get_palette_size(type);
        for (int y=0; y<hc; y++)
        {
                for (int x=0; x < wc; x++)
                {
                        int py = (y / 2) * 0x11;
                        ((uint8_t*)cbuf)[skip_palette + y * src_stride + x * bpp / 8] = py;
                }
        }

        lv_canvas_copy_buf(canvas, cbuf, (w-wc)/2, (h-hc)/2, wc, hc); // center it
        lv_obj_set_pos(canvas, 120 - w / 2, 120 - w/2);
```
**Expected behavior**

A square should be copied to a canvas on the display.

limitz, 2020-09-15:
I think i will have some time this week to look into the issue of indexed color formats in general and also make sure the problems are identified, solved and any changes are also tested against non indexed color formats. 
There would also need to be a discussion on how edge cases would need to be solved, as in: if you want to copy a buffer to a location that doesn’t start on a byte boundary, would that be allowed? 

But before i spend too much time on it i would like to make sure no one is already doing something that might make this work redundant?

Let me know :)

kisvegabor, 2020-09-17:
Hi,

Thanks for reporting it. Happy to hear that you are also interested in sending fixes for this issue(s). :slightly_smiling_face: 

> There would also need to be a discussion on how edge cases would need to be solved, as in: if you want to copy a buffer to a location that doesn’t start on a byte boundary, would that be allowed?

I think it's required such cases. 

> But before i spend too much time on it i would like to make sure no one is already doing something that might make this work redundant?

I don't know anybody working this now.

closed 2020-10-17 by stale[bot]