lvgl/lvgl #1795: Switch widget

HX2003, 2020-09-16:
**Describe the bug**

When clicking the switch repeatedly, the state does not correspond to the visual appearance.

**To Reproduce**

Please provide a small, independent code sample that can be used to reproduce the issue. Ideally, this should work in the PC simulator unless the problem is specific to one platform.

**Expected behaviour**

A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Additional context**

Add any other context about the problem here.

embeddedt, 2020-09-16:
Confirmed using the demo on the website. I was not able to reproduce on STM32 - probably because the CPU takes so long to change themes that the extra clicks are not even registered.

hedyhefty, 2020-09-18:
I have encountered the same problem...my solution was to disable the animation of the switch, and the problem seems not to occur again.

kisvegabor, 2020-09-21:
I could reproduce the issue. 
I'm pretty sure it comes from the animation of the bar. I wonder if in the name of simplification we can go for a more simple bar animation handler (like previously). 

The current bar animation mechanism was invented to produce smooth animation even if the bar has a very small range (e.g. range = 0-3 but has width = 200 px).

So can we make the user to set a larger range (e.g. instead of 0-3 -> 0-300) and in 
- value change event scale the value
- release event snap to a valid value?

embeddedt, 2020-09-21:
I'd rather keep the current animation; I think it's useful and keeps things simple for the user.

The issue seems to happen specifically when a previous animation is still in progress and a new one is started. Otherwise, the code works without problems. I think we should be able to fix it or at least add a workaround for rapid changes.

kisvegabor, 2020-09-21:
I set the switch's anim time to 3000 in the widgets demo with `lv_switch_set_anim_time(sw, 3000);`.
With this, the issue is quite obvious. 

@embeddedt If I remember well this part was written by you. Could you take a look at it?

embeddedt, 2020-09-21:
Yes; I'm currently working on a fix.

kisvegabor, 2020-09-21:
Awesome :slightly_smiling_face:

PR #1805 (2020-09-23): Fix #1795: ensure switch/bar visual state matches logical state
@HX2003 @hedyhefty If you could also test this fix that would be great.

Closes #1795.
commit b2d201db (2020-09-23): Fix #1795: ensure switch/bar visual state matches logical state (#1805)
closed 2020-09-23 by embeddedt