lvgl/lvgl #1798: Use ST HAL function for enabling DMA2D clock

robekras, 2020-09-17:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

In src/lv_gpu/lv_gpu_stm32_dma2d.c (line 66) the DMA2D clock is enabled by direct write to register RCC->AHB1ENR.
This is working for the STM32F4/7 cores, but not for STM32H7 cores. On H7 cores the RCC registers has changed.
So please use the HAL function `__HAL_RCC_DMA2D_CLK_ENABLE ();`

**To Reproduce**

n/a

**Expected behavior**

n/a

**Screenshots**

n/a

**Additional context**

n/a

kisvegabor, 2020-09-17:
Hi,

we avoid using HAL to support cases where HAL is not used. 

Can we use a define or so from CMSIS to decide how to enable the clock?

robekras, 2020-09-17:
For direct setting the DMA2D clock on STM32H743

RCC->AHB3ENR |= RCC_AHB3ENR_DMA2DEN;

but you need another setting to switch between STM32F7/4 and STM32H7.

Direct using the registers just needs including stm32h743xx.h or equivalent.
Using the HAL functions needs to include all of the appropriate HAL files.
I don't know if there is anybody out there which uses an STM32F/H without the HAL files.
I think programming the STM32 family without any HAL function doesn't make any sense.
(Just my 2 cents).

kisvegabor, 2020-09-17:
Can we do something like this?
```c
#ifdef RCC_AHB3ENR_DMA2DEN
//H7 init
#else
//F4/7 init
#endif
```

Originally HAL was used, but it was changed to CMSIS by a contributor because he wasn't using HAL. 
So it seems there are people how don't use it.

robekras, 2020-09-17:
I think RCC_AHB3ENR_DMA2DEN isn't a good name for a switch define.
This would need some deeper knowledge of the registers and their meaning.

I would choose something like 
`#ifdef STM32H7 `

This define is set within the STM32H7xx.h (which is a CMSIS file), and which should be set within the lv_conf.h file when using DMA2D
(and in the hope that all STM32H7xx are the same in regard to the RCC registers).

Including the HAL files doesn't mean, that you have to use them in any regard.
I'm not always happy with the HAL functions either, but using at least the macros for setting the registers make things easier.
But I see, people are very different...

kisvegabor, 2020-09-17:
I like the `STM32H7` define if it comes from CMSIS. 

Can you send a pull request with this change?

I could imagine some like this:
```c
#if defined(STM32F4) || defined(STM32F7)
...
#elif  defined(STM32H7)
...
#else 
# warning "LVGL can't enable the clock of DMA2D"
#endif
```

I don't know what's the case with the L4 MCUs tough. Do you know it?

robekras, 2020-09-18:
I will try on weekend, and also will have a look about the different controller types.

kisvegabor, 2020-10-07:
@robekras I see that you added https://github.com/robekras/lvgl/commit/99af960ec57404257048f25471df1c7fb1a645c0 commit. 

If it works can you send a PR?

robekras, 2020-10-07:
I thought I have done so, but it didn't work.
I've to take a look again.

kisvegabor, 2020-10-07:
Great, thanks!

embeddedt, 2020-10-07:
@robekras Thanks for the report and fix!

commit 9ec4b614 (2020-10-07): Fix #1798: Use AHB3 enable register for STM32H7 (#1840)
closed 2020-10-07 by embeddedt