lvgl/lvgl #1801: Unused macro generates warning 'expression has no effect'

xennex22, 2020-09-17:
There are several debug macros which have a definition like this when unused:
```c
#define LV_ASSERT_OBJ(obj_p, obj_type) true
```
This generates a compiler warning under the Keil compiler I use.

Either using 
```c
#define LV_ASSERT_OBJ(obj_p, obj_type)
```
or
```c
#define LV_ASSERT_OBJ(obj_p, obj_type) (void)true
```
works, but I do not know which is more portable.

The macros are found in
lv_debug.h
```c
#define LV_ASSERT_NULL(p) true
#define LV_ASSERT_MEM(p) true
#define LV_ASSERT_MEM_INTEGRITY() true
#define LV_ASSERT_STR(p) true
#define LV_ASSERT_OBJ(obj, obj_type) true
```

lv_style.h
```c
# define LV_ASSERT_STYLE(p) true
# define LV_ASSERT_STYLE_LIST(p) true
```
lv_obj.h
```c
#    define LV_ASSERT_OBJ(obj_p, obj_type) true
...# define LV_ASSERT_OBJ(obj, obj_type) true
```

xennex22, 2020-09-17:
Also I believe the LV_LOG_XXXXX(...) macros leave the const strings in my code.
eg.
```c
LV_LOG_WARN("Can't draw sub-pixel rendered letter because LV_USE_FONT_SUBPX == 0 in lv_conf.h");
```
Leaves that string "Cant' draw..." in the binary even though LV_LOG_WARN is defined as
```c
#define LV_LOG_WARN(...) \
{                        \
    ;                    \
}
```
The following definition fixes it, but I suspect that not all compilers have the same behavior
```c
#define LV_LOG_WARN(...)
```
I suspect that we will need to add something to lv_conv.h like
```c
#define UNUSED_MACRO true
```
And then
```c
#define LV_ASSERT_OBJ(obj_p, obj_type) UNUSED_MACRO
```

kisvegabor, 2020-09-21:
Hi,

Thanks for the report. 

I don't see any issues with
```c
#define LV_LOG_WARN(...)
#define LV_ASSERT_OBJ(obj_p, obj_type)
``` 
so  I think `UNUSED_MACRO` is not required we can simply go for the empty macro if unused. 

If you agree can you send a PR?

xennex22, 2020-09-21:
I was hoping you weren't going to ask that.  My fork doesn't work very well - I get many 500 server backend errors for some reason, and have to delete and re-fork to do anything, which loses any modifications I have made.  But I'll try to get it sorted out as I would like to help.

For what branch?

embeddedt, 2020-09-21:
I think `master` is fine as it shouldn't change a thing in existing code.

kisvegabor, 2020-09-22:
Thanks for sending the PR regardless of these issues!

embeddedt, 2020-09-22:
Closed via #1808.

closed 2020-09-22 by embeddedt