lvgl/lvgl #1802: What data structure functions does lvgl have?

excitedbox, 2020-09-19:
I was wondering what kind of data structures such as lists. hash tables etc. lvgl already uses in the core? I found the linked list lib in the utils folder. Are there any other ones?

I saw a bsearch function to match strings with keys and some text functions but everything seems very specialized for it's purpose. Not really an easy way to reuse those without rewriting the whole function.

I am still working on API improvements and I figure it would be better if I try to reuse what is already there as much as possible to keep the size as small as possible.

I mainly need some kind of associative array or dictionary structure to bind API functions to lvgl functions. Kinda like a typedef for functions which is what an API really is for. Then changes on the backend don't affect the frontend code.

PS.  lv_area., and lv_obj.c use basically the same code for alignment. If you rename new_x and new_y or res->x and res->y you can trim about 100 lines of code.

kisvegabor, 2020-09-21:
Double linked list with node size set in runtime is the most important data type. It is used in a lot of places.

In some was the inheritance of objects also can be considered as a data structure type.

In "style property map" properties are stored tightly after each other with the following format:
`state (1 byte) + property type (1 byte) + property value (variable length)`

We have no FIFO,  associative array, dictionary or other common structures now.

kisvegabor, 2020-09-22:
I started to think about dictionaries. Actually thy "style map" is a dictionary where  `state + property type` is the `key` and `property value` is `value`.

Now there are only a very few essential properties in `lv_obj_t` (parent, coordinates, state, style_list). The others are optional even `child_ll` is optional as most of the objects have no children. If there is no `design_cb` and `signal_cb`  `lv_obj_design/signal_cb` can be called by deafult. But with a dictinary, we can add attributes dynamically. 

It would save a lot of RAM per object.

I wonder if it could replace the curernt inheritance system too. A decendant only needs to add his attributes and will see the alraedy exsisting ones.

Of course, this dicitionary needs to support variable types as value. E.g. `lv_coord_t`, `lv_signal_cb_t`, `lv_group_t *`, etc.

What do you think?

embeddedt, 2020-09-22:
> It would save a lot of RAM per object.

What would the performance impact be, though? Right now, most object properties can be looked up with one or two load instructions. A dictionary would need to be scanned quite frequently, unless it was also backed by a cache.

kisvegabor, 2020-09-22:
Yes, performance is one question. Readability is an other. E.g. debugging style properties is quite difficult.

Regarding performance. The question is how often these proeprties are set/get. Hard to estimate for all proepties but normally it shouldn't  be extramly high.

However, I'm afraid we are trying to invent a something C is not designed for. Maybe rather we should respect more what this language offers.

excitedbox, 2020-09-22:
Sorry been busy the last two days with work.

I was considering a hash table of some sort because it is faster than iterating through a list to find a key value pair. For the style types and attributes a dictionary is easy to do since they will always be the same but for the style definitions there are new styles added by the user so the keys can't be decided before hand. You don't want to iterate through lists of ~200 styles each time to convert "align-left" to a key to the function that does the alignment for every style and style property. It could be simplified if you parsed all the property names at once and iterate through the list once and pull all keys that match 1 of the properties in your list at once. This takes more memory but requires less time overall. 

Where a hash table is faster is if you for instance use the length of the property name plus some method to offset for properties with the same length you can use size of + offset and know your key without ANY looping. Especially since properties and named attributes (left, top, bottom) don't change it should be easier to come up with an algorithm that produces enough variance for the 200 or so properties you would need for most of the CSS properties, especially since we only need a sub set of them.

The DOM is best done with a N-ry tree from my research (although I have seen lists suggested) where you start from the bottom up. each branch of the tree is a different widget container and the children are the widgets. Each branch is sort of treated as it's own sub tree because the widgets on 1 container don't interact with another when you are drawing the display but can exist on the same layer. Also by working from the bottom up you can check if an area is already covered by another object and ignore it. When you think about it this is the same structure of how a display is setup. with the screen as the root, containers as the branches and the widgets as leaves on the top layer.

What I am working on right now is making create, edit, delete, and update APIs to abstract the LVGL function calls.
ex: create (type, owner, option, param)
type = event, widget, style, callback/trigger
owner = the widget it applies to
option = which event or widget or style you are creating
param = is any attributes that go along with it.

This is then used to create lvgl code such as "lv_obj_t * arc = lv_arc_create(lv_scr_act(), NULL);" The advantage is that if later we decide to change anything in lvgl we only need to change the API code or the function name and the user doesn't need to worry about their code not working. (As long as no function is removed)

I haven't decided yet if I want the first argument to be the actual widget or event such as "arc" or "button" and change option to be the parent of the widget you are creating or add the parent to the param argument. This is only really needed for the widget creation because in all other create functions the owner is the parent. When creating a widget the "copy" argument should go in "param" so I either need to have 2 things in there, add another optional argument for the parent, or like I described above.

I just wanted to know if there was something I overlooked in the core and before implementing another structure could reuse one that is already in place.

kisvegabor, 2020-09-23:
> The DOM is best done with a N-ry tree

I gound that N means the may number of "leafs" in a tree. But we don't know to the number of children in advance so we can't "set" N to a specifiv number. Anyway, yes, it's a tree with arbitrary number of nodes and subnodes.

> create (type, owner, option, param)

In the practive something like this?
```c
lv_obj_t * obj = create(ARC, parent, TO_COPY, arc1);
set(obj, ANGLE_START, 120);
```

---

Note that, we can use variable arguments (`...`) becasue the Micropython binding can't handle them.

excitedbox, 2020-09-24:
Ok that is good to know about python. The param field could be used to store multiple arguments with a separator and spilt into multiple items but the way you showed it is the way I think I am going to use.

In usage you would actually be able to leave off "lv_obj_t * obj ="You would be calling the function and not creating a new object. The object is created by the function. But I think you got the basics. The set might even be easier than having add and edit as 2 functions. Although it would use "arc1" going by your ex. instead of obj to update the style.

each time you create a widget you would JUST need to put create("arc", "my_new_arc", "parent", my_arc_style);
my_new_arc becomes the object name. The same function can also be used for events if instead of "arc" you put long_press for instance. 

In your style definition file you would then have

"my_arc_style {
start_angle : 120;
end_angle : 230;
bg_color : green;
}"

This way you separate your styles from your interface code. Although there will also be an add, edit, update function in case you need to change a widget during execution. But if you want to change the way your UI looks you will only need to modify your style sheet and it will update your widgets.

The reason I wanted to make the first arg "event,widget,etc." was so you could see what you are creating quickly. You wouldn't really need "to_copy" as an argument option because you can apply the same style to 2 widgets. That is why I am still testing to find all these little edge cases.

We have been moving servers at work so I have been held up on this but once I get all the tests done I will make a post for input for improvements. It is hard to explain on here without drawing lots of diagrams although I have several :D

I must say coming from a web dev background you come to appreciate all the data types you don't have working with C. You have to build everything from the ground up.

PS. The N-ry tree is just to show that each branch can have more than 2 children but is a Binary tree which has exactly 2. The number of children is controlled by the code in the data structure and not hard coded so you don't need to know what N is.

kisvegabor, 2020-09-28:
One thing is not clear: do you propose using strings to describe properties? E.g.
```c
"my_arc_style {
start_angle : 120;
end_angle : 230;
bg_color : green;
}"
```
Is it really a string or a struct in some form?

excitedbox, 2020-10-03:
In those cases it is strings in a style sheet file. I wrote a parser that reads the file and uses the { } as a separator to detect the styles opening and closing. The : is used to detect a property ending and the start of a value. While the ; is used to detect the end of a property value. Right now there is no error checking in my parser. It assumes any string between the first " and { is the name of your first style. The next string is a property that ends at : and the value comes after, ending at ; and so on until it reaches the } to end that style and moves to the next.

Right now while I am testing this style sheet is only used to convert to the lvgl style code before compiling but I am looking at how feasible it is to do it live. If I can make it work live it might even be possible to distribute precompiled versions of lvgl. Although I lack the experience to with LUA or other non compiled scripting implementations to allow people to interact with their callbacks without having to compile their code.

It is hard for me to give specifics right now because so much is still changing as I work on this. Right now the API uses 7 associative lists: Properties, attributes, events, callbacks, states, styles, widgets that are used to build 5 instance lists: Property, Style, State, Widget.

Each Property instance has a property and value/attribute.
Each Property belongs to a style
Each Style has a list of properties and a state.
Each Widget has a list of styles.

Then there is the DOM which just lists the widgets and their relationship.

As I have said before how to best store the State and Styles is where I run into issues. I don't know which is better, either store a States and the state determines the style list/array used or store the widget with many styles and each style has a state which determines if it is active or not. The first option makes more sense to me but it makes the parsing of a style sheet much more difficult but the second option would require more space.

This is a quick diagram of how data flows from Dictionary -> Instances ->DOM.

[image]

embeddedt, 2020-10-03:
I'm certainly not qualified to comment on DOM implementations, but have you tested how well this works on a weaker chip like STM32? I'd be curious to see how it performs, considering that the abstractions implemented for the new style system already caused a substantial regression to performance.

kisvegabor, 2020-10-05:
@excitedbox It sounds logical but I'm also concerned about the performance issues. As @embeddedt mentioned we got concerns about the performance of the new styles system as well (despite heavy optimizations and caching).

I could imagine this as an optional layer where people can describe the UI with "text" (as HTML) and a parser creates LVGL API calls from it. It can be even a real-time parser to redefine the UI runtime. 

It'd be great if you could get closer to HTML/CSS or some similar XML-like description and finally realize a simple browser.

> As I have said before how to best store the State and Styles is where I run into issues. I don't know which is better, either store a States and the state determines the style list/array used or [...]

I'm in favor of many styles with a  state in them. It's more flexible if we add more selectors besides states. See #1832.

excitedbox, 2020-10-10:
I just realized my previous response to @embeddedt didn't get posted. I must have forgotten to hit submit.

All my tests so far have assumed an optional abstraction layer that creates the lvgl calls for you. I am just trying to plan ahead to make future development as painless as possible. I run the parser as a translator that outputs the lvgl code that you then compile as part of your lvgl.

My API system actually works EXACTLY like your example in the [post](https://github.com/lvgl/lvgl/issues/1832#issuecomment-703481122) that you linked. It converts the CSS into the lvgl format you posted. I copied a list of CSS3 properties and removed the ones that don't apply so it is as close to CSS as can be.

This way adding a selector only requires adding it to the state dictionary definition and adding a switch/if statement for it.

I think the confusion is between storing and using styles. At runtime it is faster to check what state a widget is and then apply the right list of styles instead of getting a list of styles and then using the one with the right state for each property. This would require looping through the styles and all properties of that style instead of looping through the states and styles. This leads to less cycles through your loop and less comparisons. 

I define state on a class level and not individual property level as you described it in that post. I may not have made that clear in my previous ramblings. Since I come from a web dev background I basically implemented CSS although there is no inheritance support yet because lvgl doesn't support it.

As to the performance concerns I don't use list iteration in my dictionaries because I am using Hash tables which gives you MUCH faster pairing of key value pairs. O(1) to be exact. That means it could be implemented on top of lvgl with very little penalty. It may even be possible to use as a replacement for the current style system although I haven't looked into that. I did see a lot of repeat code in the way state checking is done.

The DOM is just a list of function calls used to draw the widgets and a redraw bit for widgets that have changed since the last draw call.

A direct runtime implementation would look like this:
Start program->parse style sheet to function calls-> Build DOM array ->Draw All Items - > Start Loop ->On Event update DOM item and redraw bit-> On Draw: loop through any function needed for screen updates -> Back to Start of Loop

I guess a DOM makes more sense on a PC where everything is redrawn instead of just the changed area since the DOM is arranged to allow efficient stepping through the array. Although iterating through the DOM allows for a quick check to see if an update bit is set. Lvgl does this with a list of items to redraw instead of a bit.

None of what I am doing now is in any way finalized and once I am done I will present it as an option and you can decide what to do with it. :D

A runtime implementation if done at all should come at a later time though because it would require stripping out or changing some of the current style system and I think there are more important things to do right now.

What do you mean by implementing a browser? If you are referring to my suggestion on the forum, that was just a suggestion. I didn't know I was supposed to be working on that :)

PS. What about setting up a TODO list task board to make it easier to see what people are working on? If you need somewhere to host it I can provide free webspace. Depending on the resource requirements I may even be able to offer free hosting for the entire website.

embeddedt, 2020-10-10:
Okay; no problem with that then. It had sounded like this was being done directly inside LVGL. If it's an optional abstraction layer we don't need to worry about performance nearly as much for now.

excitedbox, 2020-10-10:
Yea it wouldn't make much sense to have a style system on top of another style system which is why I think it would be better to see what I manage to bang together and then make a decision on which parts might be integratable without creating too much extra work and what would be better to keep as an external system. Most of what I am doing would only be used on start up even if it was to be integrated so replacing the style system only makes sense if the hash tables turn out to be faster than the style map that is used right now.

excitedbox, 2020-10-11:
I took a closer look at the style system and it your performance issues are coming from the iteration and looping needed to get the properties and style list. You are literally iterating through 10s of thousands of array items. That is where 100% of your performance hit is coming from. You can test that by logging the counter of how many loops it took to retrieve each property.

You need a faster way to retrieve the KEYs of styles and properties and use the keys instead of pointers in your getter and setter functions. Also when checking if a property exists you can skip the for loop because if you could already retrieve the key you can use an if statement to make sure the ptr matches the style and does not return NULL.

If you replaced that with hash tables you would see a speed up of at least 100x. Your hash function would then give you the key for your style or property and allow you to call the hash function and then use property_list[key] or style_list[key] any time you need the pointer.

I do like the way your styles are defined using macros in the style_decl.h. I think that would be the tie in point for the API I am working on.

kisvegabor, 2020-10-12:
> If you replaced that with hash tables you would see a speed up of at least 100x. Your hash function would then give you the key for your style or property and allow you to call the hash function and then use property_list[key] or style_list[key] any time you need the pointer.

Getting the style properties is really a bottleneck now. However, I don't really understand this concept. Could you describe it with an example. E.g. we have border color = RED in a style. How can we get it real fast?

excitedbox, 2020-10-12:
Data retrieval is measured in Big O notation. Usually O(n), where O is the function/process it takes to find an item and n is the number of items you are storing/or running that process.

In a linked list you loop through the list until you find the entry that matches your style so it is very slow and gets slower the more items you have. That is O(n) or O(log n) depending on worst or best case. but in a hash table you use a hash to convert your style which is a string to a number that is used as the key to an array. That is O(1) because no matter how many items it always takes the same amount of computation (1 cycle through the hash function). Not only the same computation but the same time as well

For instance if you use size_of (the_style_name) as the hash function it would return the size of the string contained in "the_style_name" in bytes which is a number small enough to use as the key for an array. So you use an array to store keys to an array or pointers to the struct that contains all the data you need to store for your style. And every time you need to match a style to the array or struct you repeat your hash function. Since it is faster to hash a string than to iterate through thousands of array items you get a huge increase in speed. 

You also need to decide how to deal with 2 items returning the same hash which is called a collision. In some cases it is enough to put them in a linked list and if the first item isn't the right one you check the next one until you find the one that matches your string. With only about 200 CSS properties you would most likely only need to iterate through 3 or 4 items instead of hundreds. There are other ways to deal with collisions as depending on your hash function and what system you use. You can use double hashing, adding offsets.

Hash functions are the number 1 most important data structure for storing data in a C program because it quickly lets you match a string and a value/key. If you need to look up an address in a data base for instance you could hash the name and know which item of the array stores the address without having to loop through every item until you find the one that is = to the name. 

Here is a better explanation with pictures: 
https://www.bogotobogo.com/Algorithms/hashmap_hashtable.php

This is the library I am using in current tests because it is very fast but it is easy to write your own. There are also some benchmarks comparing popular libraries.
http://www.tommyds.it/doc/index.html

Uthash is another but slower option:
https://troydhanson.github.io/uthash/

Google has also released a few hash table libraries but some of them use simd and are meant for huge server systems and I am not as familiar with their other ones.

kisvegabor, 2020-10-13:
Thank you for the detailed explanation. 

Now we have only the entries table and we search in it linearly.  This way an empty style's size is zero as the entry table is built dynamically. (There is a list head which is a pointer, 4 bytes)

With hashmap, we add an intermediate layer (buckets) that helps to directly map style properties to entries without searching.
Let's say we have 128 style properties in total therefore we need 128 buckets. Assuming that every bucket is 2 byte to store an index of the entries table (don't deal with collisions for now) the size of an empty style is ~256 bytes. Actually, we don't need hashing here because the style properties are unique so they can be directly used to index the buckets. More precisely now the style properties are stored with their state so there are many variations but in a future release we will remove property-based states, and the whole style will have a state similar to CSS so the properties will be really unique. 

Hashing sounds good but I'm a little bit concerned about the extra size. Declaring 20 styles means immediately 256*20 = 5 kB RAM usage. 

--- 

The other logical approach is the logarithmic search. To do this we need to have elements in the entry table with the same size.  Now it's not the case as pointers (4/8 byte), integers (2 byte), colors (1/2/4 byte), opacity (1 byte) are stored in the same list.

If we created 4 lists: opacity, color, pointer, integers and stored the values in an ordered way we could perform a logarithmic search.

This way instead of O(n) it would be O(log(n / 4)). ( / 4 becasue the values are split into 4 lists).

Practical example, 1 style with 32 properties:
- now: 
  - interations: O(32) -> 32
  - memory usage: 32 unit +  1 list "heads"  (pointer, so 4 byte)
- hash:
  - iterations: O(1) -> 1 + hashing
  - memory: 32 unit + 128 buckets (128/256 bytes ) +  1 list "heads" 
- log search
  - iterations: O(log(32 / 4)) = 3
  - memory:  32 unit + 4 lists "heads" 

Other example: search in 100 styles, 16 properties in each:
- now: 
  - interations: 100 * O(16) -> 1600
- hash:
  - iterations: 100 * O(1) -> 100 + 100 hashing
- log search
  - iterations: 100 O(log(16 / 4)) = 200

I think mor or less `O(1) + hashing ~= O(log(n / 4))`  but with log search we need much less memory.

What do you think?

excitedbox, 2020-10-16:
No it is much less. I think you are miss understanding how relational storage systems work. There is no need to store each property for each style that would be repetitive. This is why you have dictionary lists and instance lists. (defined below).

You have 1 list that contains ALL Properties that are possible and this list is reused for ALL styles system wide. (DICTIONARY)
1 List of all States there are (DICTIONARY) this contains "pressed" and a key and a function to set the state.
1 List of ALL States that are USED (INSTANCES) that contains the instance_key, dictionary_key and list of style_instance_keys that belong to this state.
1 List of all Styles (INSTANCES) this contains style1, mybtn, etc. and their style_instance_keys
1 list for EACH Style that contains a key to the properties that is USES. (INSTANCES)

If a style contains 10 properties x 1byte for the key (1 byte can store an int up to 255)+ 1-2 bytes for the style_instance_key depending on number of styles.

When you draw a widget you will need all the properties that apply for that widget at that moment, so it makes sense to have separate lists and iterate those. The way I explained my system to work above it only needs to store 1 copy of each property.

If you have 128 properties and 30 styles and each style has 3-5 properties you would have 30 lists that each contain 3-5 keys to the property list. So when you draw a widget you only iterate through the style list that contains the 3-5 property keys. 

Lets say widget has state of pressed and pressed has a list of 5 styles that each have 10 properties. I doubt a single widget state has 50 properties but lets assume.

to draw widget -> hash the state name and retrieve list of styles for that state -> iterate through 5 styles and list of properties that match style_instance_key -> iterate through 10 properties. 
50 iterations total. But ALL those items are used so it is ok.

A dictionary is used to look up the function or what something does. 
(ex. "align" has a key of 22 and your code has a switch that says if property = 22 use the align_widget(widget, align_type) function.) 
An Instance is a specific usage of a dictionary item. the property INSTANCE may contain prop=22 attribute=left

This is kind of hard to explain because I can't tell which parts I am not clarifying enough maybe it would be better to get on a skype call or whatever?

excitedbox, 2020-10-16:
Maybe it is easier to think of it like this

I have a list of 5 styles
style1 = a,g,d
style2= b,c,h
style3= d,b,a
style4= e,h,j,a,c
style5= c,d,f,e,g,h 

and a list of 10 properties
a,b,c,d,e,f,g,h,i,j

then I have a list of instances that contains
Style_inst1, style1, style3
Style_inst2, style2, style1
Style_inst3, style3

and a list of states
pressed = Style_inst3
default = Style_inst2
disabled = style_inst1

when I then say draw pressed I can quickly match style_inst1 to style3 and property d,b,a, a but only need to iterate through 3 properties not all 10. Now imagine all the instance lists only store array keys. Or I can combine styles by assigning 2 styles to 1 style_inst. And if they are weighted I could even use them for inheritance.

I hope this makes more sense even though I mixed commas and = signs.
The hash function saves time on matching strings to keys which is done on startup mostly and with the relational storage system each list already contains the keys it needs so you no longer need the hash function during runtime unless you are adding new styles. You could manually fill in the key values for each of the dictionary lists for styles, properties, states, widgets, events, callbacks, but that would take a long time and lots of work. Then you wouldn't need to worry about collisions or having a hash function that efficiently sets the keys. This makes future changes harder though because you would always need to make sure that the manually assigned numbers match in all locations throughout the code.

kisvegabor, 2020-10-19:
It makes sense and all clear. The only thing I'm not sure it's worth the complexity of "list of states". Normally an object has ~5 styles. It should be very quick to decide if a style is for the current state or not. I mean it's less computational overhead than memory and complexity overhead.  

---

Just make a big twist... (sorry)

In [this issue](https://github.com/lvgl/lvgl/issues/1832), we started to think about CSS selectors. E.g. how to apply a descendant selector. The conclusion was to add callbacks to styles that tell a style can be applied now or not. 

Here an example: let's realize `list btn` selector and apply `border-width:2px`. To do thins write a selector callback that checks the ancestors' type and the current object's type. If they are `list` and `btn` use the style. 

I see two ways to realize it:
1. Make a global list of styles and always search in this list to decide which style to apply. 
2. Search only in the ancestor objects (usually there are only 3-4 "layer" of ancestors)

The goal is to limit the style selector checks as they can be quite expensive. Or apply some kind of caching.  

Selectors are planned only for v9 so we have plenty of time to figure out something.

excitedbox, 2020-10-20:
All these features you want to implement really REQUIRE the kind of instance list system I am proposing. It trades computation for memory usage but when implemented efficiently the additional space required is very minimal. In the end it may even be smaller since it will remove inefficiencies with the current system.

implementing selectors could be done by adding them as a selector list or treating them as a state and during instance creation check the list against the widget list. If the widget type matches the selector, add the corresponding style. With the instance system you have all the information you need on hand for each step so it is as easy as adding 1 or 2 if statements in the right location and adding a key into an array.
_____________________________________________________________________________________________________________

1. Treating selectors as states and keeping them in the state list wouldn't require any changes to the proposed system. BUT the inheritance would require a minor change. Right now each widget assumes only 1 state being active at a time but if you add a field that holds another key you could check if that field = null and if not use that as the key for the next state to call. This way you can chain states together. There should be a check to make sure it doesn't create a never ending loop though.

_____________________________________________________________________________________________________________

2. Treating selectors separate to states would mean adding another instance list containing a widget_type_key and style_instance_key and on widget creation you run a check of the selector against the widget_type_key.
```
Ex: check if selector exists in widget list and if it does add the style to that widget instance

if (widget_type_list[selector_list[widget_type]] != NULL){ create_style_instance(widget_instance, style_instance_key); }
```

This would obviously need a little more code to insert into the proper instance lists and for all the widgets along with a loop but you get the idea.

_____________________________________________________________________________________________________________
3. Keep selectors in a list just as above and use if statement when drawing to add a style. This way you only have a list with 1 item for each selector you use that contains a widget_type_key and a style_instance_key. As you draw the widget you use the code if (selector_list[widget_key] != NULL){ draw_style(selector_list[widget_key]) } 

This saves you 1 item in each widgets style list that matches a selector so around 1-4 bytes per widget instance if you use selectors for all widgets but would execute that if statement and set style function during the draw cycle. In my opinion it is better to use a few extra bytes of memory and not have the extra compute overhead.

_____________________________________________________________________________________________________________

There are more options but I believe option 2 is sufficient although I have not addressed the inheritance issue. I do not believe that there is an easy option for solving inheritance without a DOM. Implementation. This comes back to inheritance lists being better on low resource systems due to the 1 time pre-computed nature. Maybe a compatibility tree? Look into how spell checking tree data structures work. I think that could be a good starting point to check style compatibility. There are a few graph algorithms that are used for this sort of thing and you will need to look for a low computation option.

There are almost no MCUs in use that have less resources than old gameboys and they kept entire games in sdram chips. If they could manage to store data efficiently so can we. Efficient resource usage is a lost art in my opinion. When I started in web design no image was larger than 12Kb and backgrounds were 1 pixel tall images repeated down the page. Now people add MB of js to libraries just to use 1 function.

PS. The dictionaries are only needed during startup so those can be unloaded from memory after all the instances have been created. This should remove 90% of the memory usage since you no longer need to store strings of names only a few single byte keys.

excitedbox, 2020-10-20:
The state list is not mandatory it could also be an enum. The benefit is being able to add multiple styles per state and it does cut a lot of compute overhead. Any time you don't have a precomputed list you have to iterate. There are ways around it but if each widget has 5 states and you use around 30 widgets that is trading ~300 bytes of memory overhead for more compute cycles on EVERY draw cycle x the number of widgets.

In my computer science classes one of the first semesters we were forced to take DB design and after spending 2 hours a day for 6 months doing JUST that you learn lots of things and reasons you can't easily explain in a github post but trust me there are good reasons for breaking everything down. :D

Having data stored can allow you to go both directions and merge data without recomputing the same values over and over. Sometimes forward is easier to compute than backwards in a join.

puzrin, 2020-10-21:
I suggested a more simple cache approach, based on how dom tree is traversed - cache parent nodes in active path.

Performance boost will be not ideal, but very good. Advantage is the simplicity. IMO, until real product & benchmarks are ready, it doesn't worth to dive into complicated optimizations. I'd prefer to see v9 in any working state asap, then 9.1 or 10 optimized.

kisvegabor, 2020-10-22:
@excitedbox 
Thank you for the detailed comment, that's very exciting! :heart_eyes: 

For first I'd like to highlight that selectors are not limited to select object types. They can be very abstract and ideally realized via callbacks. E.g. if a selector callback returns true that style can be used. 

It still makes it possible to group styles with the same selector and save/cache the reult. 

---

Let me summarize what's in my mind now based on the previous discussions. I can imagine 4 concepts, all without selectors, and not thinking about how to search in properties.

### 1. Current concept as a reference
```
- style 1: prop1 (state), prop2 (state)
- style 2: prop3 (state), prop4 (state)
- style 3: prop5 (state), prop6 (state)

- button
  - style 1, style 2, style 3
```

###  2. Style list for every state in the widget
```
- style 1: ... properties...
- style 2: ... properties...
- style 3: ... properties...

- button
  - pressed styles
     - style 1, style 2
  - focused styles
     - style 3
```

### 3. Styles are assigned to a state
```
- style 1 (pressed): ... properties...
- style 2 (pressed): ... properties...
- style 3 (focused): ... properties...

- button:
  - style 1, style 2, style 3
```

### 4. State lists in style, like in CSS
```
- style 1 
  - deafult: ... properties...
  - pressed: ... properties... `:active`
  - focused: ... properties... `:focus`

- style 2
  - pressed: ...properties...

- button:
  - style 1, style 2
```

To remain close to CSS I like 4). 
I suggest trying to choose one of these high-level concepts to narrow the scope of discussion. If I missed something or you have other concepts, please share them. 

---

If styles with selectors are added to the global list we can do the following optimization: 
1. Before starting to draw an object mark "selector validity" as unknown
2. When the first property is read all selectors will be evaluated and get a true/false "validity". 
3. For the rest of the property querries no selector evaluation is required, just check the "validity" flag.

puzrin, 2020-10-22:
> If styles with selectors are added to the global list we can do the following optimization:
>
>  1. Before starting to draw an object mark "selector validity" as unknown
>  2. When the first property is [...]

class (style) can be assigned to multiple nodes. So, it can not have embedded state. If class is stateless, you can not calculate and remember "validity flag" in separate pass. It's more simple  check it every time, that's not expensive.

IMO, you start falling into premature micro-opimizaions. There are different alternatives, and discussing only single one is pointless. A this stage, it's enough to know "transparent optimizations possible", freeze api and move next.

puzrin, 2020-10-22:
> To remain close to CSS I like 4).
> I suggest trying to choose one of these high-level concepts to narrow the scope of discussion. If I missed something or you have other concepts, please share them.

State LIST in style is possible, but not necessary grouping. Each pair of (style_name, state) is idepended and monolytic.

CSS notation is:

```css
.my_class:focused:checked {
  ...properies
}
```

kisvegabor, 2020-10-22:
> If class is stateless, you can not calculate and remember "validity flag" in separate pass (need to allocate additional store for each node). It's more simple check it every time, that's not expensive.

However, during drawing LVGL needs to get 10-20 properties from the same object. While getting these properties the state of the object doesn't change. I.e. if you get bg color and immediately after that border color, all selector should give the same result. 

> IMO, you start falling into premature micro-opimizaions. 

Yeah, maybe. Just want to see there is some room to optimize :slightly_frowning_face: 

> State LIST in style is possible, but not necessary grouping. Each pair of (style_name, state) is idepended and monolytic.

By "(style_name, state)" do you mean names like `style_1`, `style_1_pressed`, `style_1_focused` etc?
If so with this I'm afraid we will have too many styles and it will be difficult to mage them.
With state list in style I can imagine an API like this:
```c
static lv_style_t style1;
lv_style_init(&style1);

lv_style_set_bg_color(&style1, LV_COLOR_RED);

//Create :focus
lv_style_t * s = lv_style_add_state(&style1, LV_STATE_FOCUSED);
lv_style_set_bg_color(s, LV_COLOR_BLUE);

//Create :focus:checked
s = lv_style_add_state(&style1, LV_STATE_FOCUSED | LV_STATE_CHECKED);
lv_style_set_bg_color(s, LV_COLOR_GREEN);
```

puzrin, 2020-10-22:
> However, during drawing LVGL needs to get 10-20 properties from the same object. While getting these properties the state of the object doesn't change. I.e. if you get bg color and immediately after that border color, all selector should give the same result.

You CAN use cache (later), but it's better to consider cache as isolated concept, not affecting base data structs.

If you build cache for parents on each property request, that will give you gain for each "next sibling". This can be enabled or disabled, transparent for your api and your data.

> By "(style_name, state)" do you mean names like `style_1`, `style_1_pressed`, `style_1_focused` etc?

Yes, i've updated my post. I terms of HTML i mean `(class_name, :pseudo_class)` pair. I terms of LVGL that's `style_1`, `style_1_pressed`, `style_1_pressed_checked`

> If so with this I'm afraid we will have too many styles and it will be difficult to mage them.

We can do nothing with it. That's how CSS works and nobody died yet :). If you need alernative like sass/less/stylus - use marcos, helper functions, cpp wrapper and so on.

```c
lv_style_t style1 = lv_style_create(FOCUSED);
lv_style_set_bg_color(style1, LV_COLOR_BLUE);

// OR
lv_style_t style1 = lv_style_create();
lv_style_filter(FOCUSED);
lv_style_set_bg_color(style1, LV_COLOR_BLUE);
```

excitedbox, 2020-10-24:
I had written a long response but it got TOO long so I am gonna try again.

Lvgl is in a working state. What is being suggested here is not a small feat and requires many changes to the entire system controlling the individual widget draw code. It is the difference between a translator and changing the operating system.

The problem with a quick implementation would be the inheritance and checking compatibility. This style system so far is VERY simple. When you look at the memory usage of modern browsers you see that these things are resource intensive and we are slowly getting to a point that would exclude lower power MCUs. So if we put a bunch of time into implementing a simple system we either waste a lot of time because it will never be able to support the features we want without many changes or we give up those features forever.

The more we add now the more needs to be changed later. Kis and Embeddedt have been very busy adding features and every once in a while you need to do a refactor to keep things running smoothly. Which will also make future development faster and easier.

The way lvgl functions are done does not lend itself to polymorphism very well but there is a fix for it. Right now each widget has it's own functions as a kind of namespace which C does not really support. C is a very limited language so there are trade offs for everything. The current way is good for human readability but not great when trying to make an efficient style system.

The best way to make the style system is to setup function pointers which is not that great when you need a ton of them. If the functions were rewritten to have 1 create, set, edit, delete function we could get away with just 5-7 function pointers that use a pointer to a struct member to set or update the widget. This can be built on top of the current system without the rewrite but would require many more function pointers and inefficiencies. We would require huge switch trees with the current naming scheme. So fixing it now although a lot of work saves more work later.

In the end this change would save thousands of lines of code lvgl wide because as it is right now we would have to write hundreds of functions and switch trees to control which style is used where and everything needs to be done manually or with complicated macros. The setup and maintainability would be a nightmare. There are similar GUI systems that use 5000 LOC that are not quite as advanced as lvgl but I think we can get much closer to that.

The code savings comes from having for example 1 set function lv_set(*btn, *value) or lv_set(*bar,*value) or lv_set(*whatever,*text) instead of 1 function for value and 1 for text in each widget instead you set a ptr to the struct member you want to update. Since we will already have instance lists it will be easy to add a list of ptrs to the structs containing a widgets stored settings. I can try and write an example of one of these later. 
[Struct Pointers](http://www.mathcs.emory.edu/~cheung/Courses/255/Syllabus/2-C-adv-data/struct-pointer.html)

This is kind of a crappy tutorial but it explains what I mean and shows both cases. 23:40 shows the bad and the purple line in 34:26 shows what replaces the switch tree. (Ours would have several thousand cases without this change).
https://www.youtube.com/watch?v=bZO0A1tj2MI

This change requires rewriting pretty much everything that is not drawing a widget or widget feature. Most of it will only be cut and pasting once the basic system has been coded. I am going to try and get a rough example done in the next few days. I just want to make sure I get it right so we have the best possible system going forward.

In php I could use strings to build function calls and call those functions directly making the entire style system a few lines of code but this isn't possible in C so there needs to be much more care put in.

excitedbox, 2020-10-24:
I am contemplating how to best match styles to lvgl functions without having HUGE switch trees. Function pointers are the obvious solution but don't lend themselves well to the function scheme used in lvgl. For function pointers the parameters need to be the same and each widget has it's own set of functions for each setting so we need widgets x widget_functions + widgets x style_properties in our switch tree.

For the selectors I was thinking that pressed and focused and pressed:focused are really just 2 states applied at the same time and both need to be true so the parser can treat them as a new/3rd state. This would also solve compatibility check and inheritance issues because only compatible states would be in the state lists. 
How to store this in the dictionary list? maybe an array of compatible states for each state or multiple dictionaries.

Whichever syntax we end up using is a minor change to the parser. Up until now I had been using standard css syntax but reorganizing which separator is treated in which way shouldn't take long. Just switching a few lines around.

I think that some variation of this would be most readable:
```
Stylename {
    selector:selector
        property:attribute;
   selector
       property:attribute;     
    }
```

excitedbox, 2020-10-24:
As to the different style lists, they are there so that you can apply multiple styles to a state with multiple properties for each. You don't need multiple styles per state but there is no real overhead between having an array with 1 ptr or multiple ptrs so I used it for flexibility. Also CSS allows you to apply multiple styles to a item.

You might have 1 style that sets all your colors for that state and 1 style that sets alignment and if you have a selector it might turn off 1 style while activating another. It is more of a nice to have feature.

People may be getting confused by the lists names. Each list up the hierarchy is a container for those items below it. So having a state list that controls the styles is the mechanism that turns styles into classes. 

An easy way to implement having inheritance and multiple states/selectors would be to add a selector list above state. Which then allows you to use 1 value to assign multiple states to a widget. Then during widget creation you would apply a selector item for pressed:focused (which would just be a list containing the pressed and focused state ptrs) or a selector item for the pressed state.

This would build a chain Widget->selector->state->style->property. Each containing the list of settings for the item below it.

kisvegabor, 2020-10-26:
@puzrin 
Alright, let's consider optimization later. 

> We can do nothing with it. That's how CSS works and nobody died yet :).

In CSS you don't have to add `style1`, `style1_pr`, `style1_focus` as class, only `style1` because CSS "collects" all the related classes with the same name but with assigned "state selectors". So with NO state list in styles it'd look like this in CSS. `class="style1, style1_pr, style1_focus"`. It's still not deadly, but feels inconvinient. :slightly_smiling_face:   

---
@excitedbox 
If I understood correctly by "selector" you mean something like predefined options. For example `add_selector(style, slider_value_is_greater, 80)`?

I had something more custom in mind. This way we can let the user create any complex and custom selectors:
```c
add_selector(style, my_selector, 80);

bool my_selector(obj, style, param)
{ 
   if(lv_obj_slider_get_value(obj) > param) return true;
   return false;
}
```

> This would build a chain Widget->selector->state->style->property. Each containing the list of settings for the item below it.

So this way the selectors are bound to the widget and not the style. I wonder how it would look like in C. E.g. `lv_obj_add_style(obj, style, state, selector)`?

I'm not sure if it's a real problem (especially if the selectors can be really custom) but it allows only one selector per style. 

I'm thinking about if we can treat states as selectors as well. Ultimately there is no real difference and in CSS it's also not really separated.

I see some limitations of attaching the selector to the widget. E.g. we can't realize a simple `element-type` (e.g. `button`) selector without adding styles to the buttons. Seemingly we need a global list of styles-selectors too.

puzrin, 2020-10-26:
> In CSS you don't have to add `style1`, `style1_pr`, `style1_focus` as class, only `style1` because CSS "collects" all the related classes with the same name but with assigned "state selectors". So [...]

In CSS i have cases when states are declared later, after all. I suggest to implement CSS functionality as close as possible. Without "improvements".

All sugar can be added later, via macros, helpers an so on. But if you pin api with nails to certain programming patterns, that will be impossible to unroll back.

Also note:

- You will have to support multiple classes anyway
- Number of "stateful" classes is much less than normal ones. No need to create hack with small coverage.

> I see some limitations of attaching the selector to the widget. E.g. we can't realize a simple `element-type` (e.g. `button`) selector without adding styles to the buttons. Seemingly we need a global list of styles-selectors too.

If someone introduce composite widget (with base components inside), it's normal & right (according to BEM) to assign classes to internal elements. And to make those public. But problem may be with custom widgets if those need vendor-prefixd properties. If you define prop id-s via enum - it's no clear how to extend.

kisvegabor, 2020-10-26:
> If you see concrete problem with api methods - let's review, and i will try to suggest something accepable.

That's true that we can "stateless" styles are more "fundamental" and we can group them later with ease. 

> If someone introduce composite widget (with base components inside), it's normal & right (according to BEM) to assign classes to internal elements. And to make those public. But problem may be with [...]

I see this problem but can't see how it's related to the element type selectors.

We can reserve e.g. 64 IDs for custom properties. It doesn't allow clear separation between each custom widget type because the user needs to take care of the enumeration of the properties so that they don't collide.

puzrin, 2020-10-26:
Well, i see 2 directions to do in parallel:

- Data flow design - driven by requirement to be effetive & extendable by "cache".
- User api design - driven by requirement to be convenient & close to CSS.

Both goals are related, but requirements are different and should not be mixed. I think, when you publish preliminary api, that will help to understand what to do with data. And after several adjusting iterations we can get something final.

Under "api" i mean function signatures only, without implementation.

I can help with verification of correctness & suggestions how to follow CSS as close as possible.

kisvegabor, 2020-10-27:
Okay, here are the ultimate requirements I see. I'm not sure all are really reasonable so we can make compromises.
  - make the API lean, have short functions with few parameters
  - make it intuitive to use
  - make it work MicroPython too (no vararg, user_data field where required) 
  - allow attaching styles to objects similarly to classes
  - allow that a style can affect multiple objects even without attaching the style to them (e.g. an element type)
  - allow flexible selectors similarly to CSS (descendant selector, type selector, and any custom selector)
  - allow multiple selectors for a style in AND relation
  - handle specificity in some form (e.g. styles for 2 combined states are stronger than styles for 1 state)
  - target low memory usage
  - make it possible to add some kind of fast search (hash, log) or caching

I have two API ideas. Basically, the question is whether we want to support "global styles" (style the have effect but not directly assigned to the widget as a class, e.g. element-type selector)

### 1. Implement everything
It's basically like CSS
```c
lv_style_init(&style1);  //LVGL add this style toa global list or so. 
lv_style_add_selector_int(&style1, my_selector_cb, param_int);         //Assign the selector to the style
lv_style_add_selector_ptr(&style1, my_selector_cb, param_ptr);         //Assign an other selector too
lv_style_set_bg_color(&style1, LV_COLOR_RED);

lv_style_init(&style2);      
...

lv_obj_add_style(obj, LV_OBJ_PART_MAIN, &style2);  //style2 is assigned directly to a widget as a "class"

//Selector prototype
int selector (obj, part, style, param); //Returns the specificity

//Predefined selectors:
lv_style_select_if_state          //:focus  
lv_style_select_if_type           //button 
lv_style_select_if_parent_obj     //#parent-id > *
lv_style_select_if_parent_type    //button > *
lv_style_select_if_parent_class   //.class1 > *
lv_style_select_if_parent_state   //:focus > *
lv_style_select_if_ancestor_obj   //#ancestor-id  * 
lv_style_select_if_ancestor_type  //button  *
lv_style_select_if_ancestor_class //.class1  *
lv_style_select_if_ancestor_state //:focus  *

//E.g. to realize `list btn:pressed` use
lv_style_add_selector_ptr(style, lv_style_select_if_type, "lv_btn");
lv_style_add_selector_int(style, lv_style_select_if_state, LV_STATE_PRESSED);
lv_style_add_selector_ptr(style, lv_style_select_if_parent_type, "lv_list");
``` 

Advantages:
- Very flexible
- Convenient

### 2. No global styles
Styles affect only the object where it is added. The API is the same as in 1) but every style needs to be added to one or more widget(s).

Here selectors can be used only to enable disable certain styles added to the widget.

Two ways depending how states are handled:
```c
lv_style_init(&style1);
lv_style_set_state(&style2, state);
...
lv_obj_add_style(obj, LV_OBJ_PART_MAIN, &style2);  

//OR
lv_style_init(&style1);
...
lv_obj_add_style(obj, LV_OBJ_PART_MAIN, state, &style2);
```

Advantages:
- No specificity required because we have only classes
- Surely faster and more simple if there are no global styles

puzrin, 2020-10-27:
I'm for starting with minimal possible set.

I don't like idea of `> *` and `*`, because that's completely different complexity for calculations on the fly. As alternative i could suggest:

- Force (optionally) assign default class to each wiget type on create. For example, assign `_btn` to each btn. So, if user needs tweak all buttons, he just update global `_btn` class instead of ask for global "btn" selector.
- Provide DOM tree traverse iterator (instead of nested selectors).

This should be enough for embeded demands, keep api simple, and costs almost nothing.

excitedbox, 2020-10-28:
@kisvegabor So you are suggesting computed selectors if I understand you correctly? This would be very hard to implement. Not impossible but right now the biggest hold up we have is the C language itself.

The way lvgl functions are written makes it difficult to call a function dynamically so anything we want to do style wise will need a table of function pointers and a switch tree for EVERY widget and every widget function. This represents a giant overhead. If we rewrite lvgl to have the 5 or 6 constructor functions as I mentioned before we can remove 90% of this overhead. I mean things like lv_obj, lv_btn, lv_style etc. Just for the widgets we would need 30 switches as it stands and then switches for every function for those widgets and every style for adding and updating and deleting. This means for a 100 style properties x the number of widgets x 3 (for add, edit, remove) + all the widget add and update value functions.

If we use my previous suggestion of lv_create(what, parent, name, params) we would only need create, edit, delete functions and pass the rest as arguments. Then we use those to update the widget structs but still need switches for the draw functions and not the individual styles.

I am sort of stuck there because I can't find a way to call the widget functions as they are written, efficiently without that change.

Every option I look at comes back to this problem that we don't have good options for polymophism unless we use C++. Which is also not great.

kisvegabor, 2020-10-28:
@excitedbox 
I agree that these complex selectors can be too complex and performance and memory hungry in the end.  See my comments below to @puzrin's ideas. 

With this, we can back to something more simple, similar to what you suggested above. (object care only about styles that are added to them)

---

@puzrin 
> Force (optionally) assign default class to each wiget type on create. For example, assign _btn to each btn. So, if user needs tweak all buttons, he just update global _btn class instead of ask for global "btn" selector.

That's pretty much how themes are working now. Seemingly hierarchy-related selectors really quite similar to themes. 
So what if we change how thems work:
- Allow attaching theme callback(s) to the displays. Global element type styles can be handled here. 
- Allow attaching theme callback(s) to any object. Styles can be added to the children here when they are created.

So when an object is created the theme callbacks of each of its ancestors are called to style the new widget. (display, grandparent, parent) 
However, it doesn't allow run time style change (only creation time)

> Provide DOM tree traverse iterator (instead of nested selectors).

Combined with the theme callbacks it gives the run time styling ability too. 

I love your idea to replace descendant selectors with themes.

Here is an API draft:
```c
//Themes
lv_disp_add_theme_cb(disp, my_theme_cb);
lv_disp_add_theme_cb(disp, other_theme_cb); //Extend the previous theme

lv_obj_t * list = lv_list_create(lv_scr_act);
 //First  approach
lv_obj_add_theme_cb(list, list_btn_style_cb);
//Second approach 
//Add style1 to the new descendant if condition_cb returns true
lv_obj_add_theme_style(list, &style1, condition_cb, param);

//Change style of mass objects
//Apply exec_cb on objects where condition_cb is true. Start searching from start_obj in "deep" level 
lv_obj_select_and_execute(start_obj, deep, conditon_cb, exec_cb);
```

If we agree on this direction the selectors are much simplified and their purpose is only to enable/disable classes that are attached to an object.

@amirgon how do you like it? Do you think it'd serve the use cases well where you needed descendant selectors?

amirgon, 2020-10-28:
> @amirgon how do you like it? Do you think it'd serve the use cases well where you needed descendant selectors?

@kisvegabor So `lv_obj_add_theme_style(list, &style1, true_condition_cb, param)` would just apply the style to list and all its descendants, right?  So it looks like it could replace descendant selectors.  

However, it's not clear to me how you would do the equivalent of CSS classes/styles runtime manipulations.
Suppose that in CSS we have some predefined selectors that apply styles to some classes:
- How would you do the equivalent of adding/removing a class from an existing element? For example for marking which cells are selected in a table
- How would you do the equivalent of adding/removing a style from an existing class? For example for changing the way selected cells are highlighted

puzrin, 2020-10-28:
> ```c
> //Themes
> lv_disp_add_theme_cb(disp, my_theme_cb);
> lv_disp_add_theme_cb(disp, other_theme_cb); //Extend the previous theme
>
> lv_obj_t * list = lv_list_create(lv_scr_act);
>  //First  approach [...]

Sorry, i did not learned how themes work. Where can i read principle, how theme recognizes object type and apply modifications specific for this type?

> ```c
> //Change style of mass objects
> //Apply exec_cb on objects where condition_cb is true. Start searching from start_obj in "deep" level 
> lv_obj_select_and_execute(start_obj, deep, conditon_cb, exec_cb);
> ```

I have some comments:

- Method name is too specific. I'd prefer something more generic, like `lv_dom_traverse`.
- I don't understand why `conditon_cb` and `exec_cb` should be separate.
- `start_obj` => `root_obj`

Also, "true" iterator may be needed (for example, to search node by class):

```c
lv_dom_search_next(root_obj, deep, condition_cb, current_obj)
```

kisvegabor, 2020-10-28:
@amirgon 
> However, it's not clear to me how you would do the equivalent of CSS classes/styles runtime manipulations.
Suppose that in CSS we have some predefined selectors that apply styles to some classes:

Independently from this "theme concept" the normal `lv_obj_add_style()` still works. The discussed method can be used "only" to give a default appearance during creation.

In my vision, the theme callbacks simply add styles to the widgets (same as the themes now)

Selectors are used to disable a style (add to a widget) in certain conditions. 

@puzrin 
> Sorry, i did not learned how themes work. Where can i read principle, how theme recognizes object type and apply modifications specific for this type?

You can find it in the docs: https://docs.lvgl.io/latest/en/html/overview/style.html#themes

> Method name is too specific. I'd prefer something more generic, like lv_dom_traverse or lv_dom_walk.

True, but DOM is too specific to web. `lv_obj_tree_traverse`?

> I don't understand why conditon_cb and exec_cb should be separate.

It really can be the same.

> start_obj => root_obj

I agree.

> Also, "true" iterator may be needed (for example, to search node by class):

I don't get it. What's the difference between `obj_cb` and `condition_cb`?

puzrin, 2020-10-28:
> True, but DOM is too specific to web. `lv_obj_tree_traverse`?

Ok for now. I will review again when see all new methods on one page.

> I don't get it. What's the difference between `obj_cb` and `condition_cb`?

Not principle. But methods are different

- traverse  fn applies callback to each object and don't care about result.
- iterator `*_search_next` returns next object, satisfying condition (checks cb result). That's like "single step of traverse process".

> You can find it in the docs: [docs.lvgl.io/latest/en/html/overview/style.html#themes](https://docs.lvgl.io/latest/en/html/overview/style.html#themes)

https://github.com/lvgl/lvgl/blob/master/src/lv_widgets/lv_btn.c#L95

Got it. Now object type passed explicit from "constructor". I almost reinvented how themes work :). If `lv_theme_apply()` can just assign object_type-specific default global class(es), that's exactly what i suggested.

excitedbox, 2020-10-28:
A DOM is a coding Model. It is like object oriented but used for graphics and states. Not used just for web

If I interpreted the current style system correctly it already works kind of like that when you look at the lv_style.c that is basically an instance list system. I am suggesting using that for everything.

The DOM would be a list of ptrs to all instantiated objects and a bit to see if it should be redrawn or not. If yes then it gets the widget settings array/or struct and iterates through it. The order of items in the DOM list controls the order in which they are drawn. To speed things up you can have a second list of changed widgets so you don't need to iterate through all widgets for minor refreshes. This could be determined based on which Event is triggered. For instance a Tab change is a big refresh a state change is a small refresh like a check box being checked.

excitedbox, 2020-10-28:
I just came up with a fix for inheritance. By adding inheritance selectors during the building of the DOM instead of during style instancing. That would iterate the DOMs widgets in the right order anyway and add those style ptrs to each child widget's style list. That would allow applying inheritance with very few iterations. 

Since the DOM works kind of like a linked list (It is really a tree). Instead of tons of checking you only need to add a check to see if the widget already has that style and if not add it. That makes it the last style update to the widget and overwrite any direct widget styles. 

You would need an overwrite byte that contains the pointer to that style so if a callback attempts to change it you can stop it. But that only adds 1 byte to each widget instance.

A screen with a 3 row table and a button in each of those rows would need 6 iterations to add the Tables style to ALL 3 buttons.

puzrin, 2020-10-29:
There is still uncovered area, we discussed earlier as "class list manager", when i explained how to refer classes effectively and transparently for user-friendly api.

Imagine we have list of items and i wish to mark one as selected:

```
.list-container
  .item
  .item
  .item .item-selected
  .item
  .item
```

When i create `item` element, it's "natural" to assign the same shared class list to each. BUT, if class list is shared, attempt to add `.item-selected` will affect all items. Both straight alternatives are ugly:

- swapping class list is not "natural" for user.
- having independent class list will eat memory.

My suggestion was to use shadow pool of unique class, addressed by "bucket" (calculated on the fly):

```
[ classes_count, *class1_ptr, *class2_ptr, ... ]
```

When user add class or list of classes to lv_obj:

- bucket of new result is calculated.
- if already exists in pool - reuse it's pointer.
- if not exists - create and replace old pointer.

What to verify?

- When creating list, user may wish to create list of classes [template], and assign it to each item.
- Result should look independent. If user add ".item-selected" class, this shold not affect other items.
- If theme used, injected theme's classes must be first.

---

Search in pool happens only on [any] class-list modification - not too often. But, if extra-perf required, data can be organized as 2-dimention list:

```
  lists_with_1_class[...]
  lists_with_2_classes[...]
  lists_with_4_classes[...]
```

with sorted content for fast binary search.

puzrin, 2020-10-29:
@kisvegabor 

> So when an object is created the theme callbacks of each of its ancestors are called to style the new widget. (display, grandparent, parent)
> However, **it doesn't allow run time style change** (only creation time)

Could you clarify, what you mean?

Style lists and properties are "dynamic". User can patch anything & anytime, gloal and local. I see no limits.

embeddedt, 2020-10-29:
You can add/remove styles from an object as you wish, but theme callbacks are only called in the `lv_<type>_create` method.

puzrin, 2020-10-29:
You mean, difficult to replace theme on the fly? I think, this can be solved, if really needed. But, TBH, i've seen this feature only in demos.

embeddedt, 2020-10-29:
Themes are not a total replacement for selectors because they only allow styles to be assigned at creation time. With a selector, you could change a style at runtime and it would still be applied to many objects.

I think it could be useful for things like changing the font size, etc.

puzrin, 2020-10-29:
I don't undersand your case. If you have repeating objects, and know you need to modify those in bulk - add extra style to those in creating code. Then you can use it as direct selector when required.

Theme was discussed in context of global selector request (like `button`, `a`, `img` in HTML).

excitedbox, 2020-10-29:
With my suggestion you could apply a global inherited style called theme and it would work at runtime. I do see what you mean with theme though because the style might not contain every setting for every widget. How are themes done now? Don't they contain a implicitly defined value for every style property?

You would create a style called theme1 and define all properties in it for each widget, while using the widget type as a sub-selector in the style. Then set the global inherit selector on the screen instance and when the DOM is drawn/redrawn it will update all widgets below the screen with just the properties that apply to that widget. The global selector would count as a big refresh which iterates through all instanced widgets instead of just the ones in the refresh list. It would work kind of like setting the copy arg for a widget at draw time and copy the style.

________________________________________________________________________________________________________________________________

@puzrin your item example should work if each item in the list is an instance like a widget. I think currently the list widget sets a state on a certain row value though so it might need a slight mod to the widget. The "shadow pool" you suggested is EXACTLY what the instance system is meant to do. What I have been calling instance lists is what you are calling a bucket.

What about sub/restricted widgets that only work as child of certain widget type? Just like the <tr> element in HTML is restricted to the table. It is still an element but can only be used as a child. That would solve a lot of other issues as well.

____________________________________________________________________________________________________________________________

@kisvegabor Conditional selectors could be possible but would need to be a function for each condition and then we pass Selector parameters for the condition. You would pick from a list of predefined conditions or add your own and add it to the selector dictionary.

____________________________________________________________________________________________________________________________

I may have come up with a solution for the massive switch trees. I feel kind of stupid for not thinking of it before. I saw a multiplication table earlier and that made me think of it.

Using a multi-dimensional matrix for the function pointers should allow us to match them without or at least a lot fewer switches. If we have a function called function_to_execute and use that to call the function pointer from the matrix it eliminates the need for lots of switches. It may be beneficial to split it into multiple matrices to cut down on the dimensions. but it should work.

____________________________________________________________________________________________________________________________

Most of the discussed features are now supported by the instance system or could be made to work with minor additions as it is planned now. Including conditional selectors and inheritance :)

As it stands:
Part1: Dictionary arrays and hash functions are made for matching "CSS" to functions for setting Widget style properties.
Part 2: User writes Action functions for the widget to execute and adds them to an Action Dictionary. (automate with script?)
Part 3: User writes "CSS" code to define his widgets look and callbacks
Part 4: The user writes a list that defines the DOM (Think HTML template but function calls)
**The Program starts**
Part 5: The CSS is parsed to create instances of each widget, the look, the events, and what actions are triggered by those events
Part 6: The DOM list is created by the function calls and draws the widgets.
Part 7: A button is pushed and the Event is triggered which executes the Action causing any widget values to be updated and code executed.
Part 8: Each draw cycle the DOM can either be redrawn completely or just a list of changed widgets.
Part 9: If a inheritance is updated as the DOM iterates through it adds the inherited style to the widget before drawing it.

CSS Style Syntax:
Widget.Style*Selector{      *Widget and Selector optional
Widget.State*Selector [
property : attribute;
]
}

DOM Syntax:                     *instead of HTML
???????????????????? lv_dom(arg1,arg2,arg3,arg4);

puzrin, 2020-10-29:
> With my suggestion you could apply a global inherited style called theme and it would work at runtime.

IMO here is some confusion about terms. What is "inherited style"? In my mind, style === HTML class. But it has different name due context (embedded programming). No more differences.

Theme = magical black-box, which know how to assign extra styles to nodes (lv_obj-like).

So, i can't understand why style inheritance required. Duplication required for lists of styles, but that's another story, not inheritance.

> your item example should work if each item in the list is an instance like a widget. I think currently the list widget sets a state on a certain row value though so it might need a slight mod to the widget.

Current (v7) list widget is outdated concept. In v8 any lv_obj can behave like "list", because it behaves like `<div>` (supports scolling and so on).

> What about sub/restricted widgets that only work as child of certain widget type? Just like the element in HTML is restricted to the table. It is still an element but can only be used as a child. That would solve a lot of other issues as well.

I don't understand use-case. Looks like overkill "just for fun"

kisvegabor, 2020-10-30:
@puzrin 
> My suggestion was to use shadow pool of unique class, addressed by "bucket" (calculated on the fly):

I see your point but see the following advantages of local style list:
- local styles (`style="..."` in CSS) are unique to a given object. With the bucket concept, we need `local_style + bucket id` 
- transition style (anim. on state change) also unique to a given object. So finally `local_style + trans. style + bucket id`. 

So with buckets, we need 3 elements even if the object has no styles at all (12 byes).  

Now local and trans. style is added on demand so they only consume memory when used. Now it's a single pointer to the array of styles (4 bytes). When the first style is added a `malloc` called which has 4 byte overhead + 4 byte the pointer to the style. 

So memory usage:
| Num. of styles | Bucket | Current
|---------|----|----|
| 0 | 12 | 4   |
| 1 | 12 | 12 |
| 2 | 12 | 16 |
| 3 | 12 | 20 |
| 4 | 12 | 24 |
| 5 | 12 | 28 |

So if on average there are more than 1 style per widget, buckets are better in memory usage. So what is the average number of styles per widget? Let's say half of the widget are labels with no styles (all props are inherited), the others have 2-3 styles. So on average 1-2 styles per widget. It seems the gain wouldn't be that significant.

> I don't undersand your case. If you have repeating objects, and know you need to modify those in bulk - add extra style to those in creating code. Then you can use it as direct selector when required.
> Theme was discussed in context of global selector request (like button, a, img in HTML).

It possible, I just wanted to highlight that what @embeddedt already said: in contrast to CSS selectors it works only in creation time. There are methods to make runtime style change though.

---

@excitedbox 
>As it stands:
Part1: Dictionary arrays and hash functions are made for matching "CSS" to functions for setting Widget style properties.
...

It seems like a massive conceptual change. In an "upper scripting layer" it should be fine, but for the core LVGL I'd prefer to keep the current general concept. 

The main question here is to find the best way to organize styles. 

> Conditional selectors could be possible but would need to be a function for each condition and then we pass Selector parameters for the condition. You would pick from a list of predefined conditions or add your own and add it to the selector dictionary.

Yes, I see it in the same way.

---

Refining the requirements from [here](https://github.com/lvgl/lvgl/issues/1802#issuecomment-717200088):
- make the API lean, have short functions with few parameters
- make it intuitive to use
- make it work MicroPython too (no vararg, user_data field where required)
- target low memory usage
- make it possible to add some kind of fast search (hash, log) or caching
- allow attaching styles to objects similarly to classes
- allow enabling certain styles based on run-time conditions
- allow theme-like support for any object to affect their children in the creation time 

I think we should focus on finding a good solution for these.

puzrin, 2020-10-30:
> So if on average there are more than 1 style per widget, buckets are better in memory usage. So what is the average number of styles per widget? Let's say half of the widget are labels with no [...]

May be you are right, but i'd like to clarify some moments:

1. Current resricted CSS implemnetaion is done with BEM methodology in mind - it should compensate missed features. So, situation when no classes assigned - probably bad design. We should start measure from a least 1 class per node.
2. If theme used, two classes will be assigned - one from theme an another from user `[ .button, .my-list-button ]`.
3. First row of your table says "Num. of styles". Seems sould be (or mossed) "Number of duplicated DOM nodes". No sure.

If we take list with 2-lines items (container+header+descripion) - each node duplicates 3 elements.

So, checkpoint is: "make sure long lists with 2-class items will be ok"

> * local styles (`style="..."` in CSS) are unique to a given object. With the bucket concept, we need `local_style + bucket id`

- Using local styles in CSS is bad practice.
- Even if used, i'd suggest to drop it from competition as "hack". Because optimizing "solid mainstream solution" has more priority.

---

Summary:

- Probably, you are right, saving in real project will be not huge. But may be my notes will give you some ideas about estimates clarification.
- Described cache can be added transparently anytime. It's isolated and will not break anything. The only requirement is - don't write to class list manually (use `add_class` and other methods as abstraction).

embeddedt, 2020-10-30:
> Using local styles in CSS is bad practice.

I think there are legitimate cases for using local styles, particularly if the style is being driven by JavaScript as opposed to CSS. At least, I tend to use them for that purpose quite frequently. :slightly_smiling_face:

puzrin, 2020-10-30:
> I think there are legitimate cases for using local styles, particularly if the style is being driven by JavaScript as opposed to CSS. At least, I tend to use them for that purpose quite frequently. slightly_smiling_face

I'm not going to prohibit you use this pattern. I say, it not worth efforts to optimize or use as reference case.

- if we optimize it, that will not help to optimize BEM.
- if we are sure BEM consumption is ok, then your case will be ok automatically.

kisvegabor, 2020-10-30:
BEM should be great and praised by a lot of people but not a golden standard followed and accepted by everyone. Our design guide was to be close to CSS and not to BEM. And local styles are part of CSS. Personally, I also like them in LVGL and find it convenient quite often for really local changes. 

> Described cache can be added transparently anytime. It's isolated and will not break anything. The only requirement is - don't write to class list manually (use add_class and other methods as abstraction).

I the advantage that if we have 100 items with that same styles the caches need to recalculated 100 times. With buckets only once.

puzrin, 2020-10-30:
> I the advantage that if we have 100 items with that same styles the caches need to recalculated 100 times. With buckets only once.

- That happens "rare".
- Calculation consists of 2 binary searches only - very fast.

=> this is not critical chain, can not be advantage or disadvanage.

[99 more replies]

closed 2021-04-24 by kisvegabor