lvgl/lvgl #1806: Building with DMA2D support

KammutierSpule, 2020-09-21:
**Describe the bug**

I'm trying to build with DMA2D support
LV_USE_GPU_STM32_DMA2D 1
#define LV_GPU_DMA2D_CMSIS_INCLUDE "stm32f429xx.h"

on a stm32f429xx but it fails to build

**To Reproduce**

Build the v7.5.0 library version wtih 
#define LV_USE_GPU_STM32_DMA2D  1

**Additional context**

There are two issues,
The
#define LV_GPU_DMA2D_CMSIS_INCLUDE "stm32f429xx.h"
is looks it is not enough on my case,
It will not find the SCB_CleanInvalidateDCache()
and I need to include a second
#include "core_cm4.h"
in order to make it build.

It may be that the drivers I'm using (CMSIS and STM32F4xx_HAL_Driver) are old versions, but I cannot update at the moment as it is a stable project.

Is something I'm missing or is some issue with the old drivers version?

embeddedt, 2020-09-21:
You could create your own custom header that includes both `stm32f429xx.h` and `core_cm4.h`, and then pass that header to `LV_GPU_DMA2D_CMSIS_INCLUDE`.

KammutierSpule, 2020-09-21:
yes that is possible.
but what about SCB_CleanInvalidateDCache? it looks it is not available in cm4 but only on cm7 (at least on my library versions)

embeddedt, 2020-09-21:
That's correct. Try the latest `master` branch. https://github.com/lvgl/lvgl/commit/94a6dc08d838d78eaf96c91f380e5c0fed3c1484

KammutierSpule, 2020-09-21:
ah ok thanks!

kisvegabor, 2020-10-07:
As the issue seems to be solved I close it.

If it's still not working please reopen the issue and comment here.

closed 2020-10-07 by kisvegabor