lvgl/lvgl #1807: artifacts when using DMA2D on a STM32F429xx

KammutierSpule, 2020-09-21:
**Describe the bug**

I enabled DMA2D on a STM32F429xx.
It works.
but there are some random line artifacts.
I notice it running the benchmark demo, it will produce some lines, special on the sides (eg of squares, images).

**To Reproduce**

Running benchmark demo using v7.5.0

**Screenshots**
[image]

I also noticed this without using DMA2D but it was very minor or small issues.

What could be causing this artifacts that are very noticeable when enabling DMA2D ?

embeddedt, 2020-09-21:
If it appears without DMA2D, I think the issue is not related to DMA2D. Are you using the F429 example project's display driver, or your own?

KammutierSpule, 2020-09-21:
I guess its my own :) I will have a look on that display driver example.

KammutierSpule, 2020-09-21:
oks it may be a display driver issue, I will close this.

closed 2020-09-21 by KammutierSpule