lvgl/lvgl #1811: BIDI RTL: Text in not aligned correctly in DropDown

amirgon, 2020-09-22:
**Describe the bug**

On RTL DropDown, text is not aligned to the right as it should be.  
Also, the down-pointing-arrow should also be drawn on the left instead of on on the right.    

[image]
**To Reproduce**

- On lv_conf.h
  - Enable `LV_FONT_DEJAVU_16_PERSIAN_HEBREW` 
  - Enable `LV_USE_BIDI`
- Set BIDI to RTL
- Create a DropDown
- Apply font to dropdown

Example:

```python
scr = lv.obj()
scr.set_base_dir(lv.BIDI_DIR.RTL)

dropdown = lv.dropdown(scr)
dropdown.set_style_local_text_font(dropdown.PART.MAIN, lv.STATE.DEFAULT, lv.font_dejavu_16_persian_hebrew)
dropdown.set_style_local_text_font(dropdown.PART.LIST, lv.STATE.DEFAULT, lv.font_dejavu_16_persian_hebrew)
dropdown.set_style_local_text_font(dropdown.PART.LIST, lv.STATE.FOCUSED, lv.font_dejavu_16_persian_hebrew)
dropdown.set_options("1. זוהי שורה ראשונה\n2. זו שורה שניה\n3. וזוהי השלישית!")
dropdown.set_width(400)
dropdown.set_y(20)

lv.scr_load(scr)
```

**Related question**

Is there a way to apply a font to an object without iterating through all combinations of part/style of that object?

embeddedt, 2020-09-22:
> Is there a way to apply a font to an object without iterating through all combinations of part/style of that object?

You can create a single style and then add it to each part of the object. You still need to iterate through the parts though, as I don't think they can inherit from each other.

amirgon, 2020-09-22:
> You can create a single style and then add it to each part of the object. You still need to iterate through the parts though, as I don't think they can inherit from each other.

But not only through all parts, but also through all states, right?
That's not very scalable. A better approach could be a bitmap, so I could specify multipled (ORed) parts/states at once.

embeddedt, 2020-09-22:
States inherit from one another, so if you specify something on the default state, it also applies to the focused state unless you override it there.

amirgon, 2020-09-22:
> States inherit from one another, so if you specify something on the default state

Thanks. I didn't know that.

amirgon, 2020-09-22:
One more thing I've noticed - there is an `LV_OBJ_PART_ALL` constant, but it does not work.
If I'm trying to set it instead of MAIN part, the application just crashes.
```python
dropdown.set_style_local_text_font(lv.obj.PART.ALL, lv.STATE.DEFAULT, lv.font_dejavu_16_persian_hebrew)
```
It's not really documented so I'm not sure if it is supposed to work or not.

kisvegabor, 2020-09-24:
I'm sorry, somehow I missed this issue.

> On RTL DropDown, text is not aligned to the right as it should be.

This functionality was really lost. I will fix it on Monday.

>  One more thing I've noticed - there is an LV_OBJ_PART_ALL constant, but it does not work.

`LV_OBJ_PART_ALL` can't be used here, the part needs to be an exact part. `LV_OBJ_PART_ALL` should be marked with `_` because the user has nothing to do with it.

amirgon, 2020-09-24:
> `LV_OBJ_PART_ALL` can't be used here, the part needs to be an exact part. `LV_OBJ_PART_ALL` should be marked with `_` because the user has nothing to do with it.

Suppose I have a screen which is RTL and would like to apply an RTL font to all objects in it.
Possible solutions:
- Add a font-style to each part of each component
- Create a new theme that would add the font-style to each part of each component

But is there a simpler way to do it? A way that doesn't require iterating **every** part of **every** component?
(this brings us back to a discussion we had about applying styles to components hierarchies... https://github.com/lvgl/lvgl/issues/1551)

kisvegabor, 2020-09-27:
>> On RTL DropDown, text is not aligned to the right as it should be.

> This functionality was really lost. I will fix it on Monday.

I've just pushed a fix.

> Suppose I have a screen which is RTL and would like to apply an RTL font to all objects in it.

One more solution is to apply the style on the screen and let other objects inherit it. 

In the second half of [this comment](https://github.com/lvgl/lvgl/issues/1719#issuecomment-675935063), I proposed an extension for styles but we mainly started to talk about the widget simplification topic and this idea wasn't really discussed. I still see a big potential in it so I'm very open to warm it up. 

@puzrin suggested another thing [Document.querySelectorAll()](https://github.com/lvgl/lvgl/issues/1719#issuecomment-678274156). I'm not sure how can we make it easy to use in C but it's also worth discussing it.

amirgon, 2020-09-27:
> I've just pushed a fix.

@kisvegabor maybe I'm missing something, but ffd9b872d9dab7345c16cb5b36206809510296ee doesn't seem to solve the problem.

> One more solution is to apply the style on the screen and let other objects inherit it.

This didn't work for me.  
Setting the font style to screen didn't affect the font styles of new objects that were added to screen.

> In the second half of [this comment](https://github.com/lvgl/lvgl/issues/1719#issuecomment-675935063), I proposed an extension for styles but we mainly started to talk about the widget [...]

I think it would be a good idea to allow the user set the styles of multiple objects in a "batch" operation.  
Currently it seems that the only way is to set the style of each part of each object explicitly, which is tedious and not very scalable.

kisvegabor, 2020-09-28:
> maybe I'm missing something, but ffd9b87 doesn't seem to solve the problem.

Ah, I see. The issue that happened when the list was wider than the label. I also fixed the arrow-text order.

> Setting the font style to screen didn't affect the font styles of new objects that were added to screen.

It's because in the default themes the font is already set on almost every style so they overwrite the screen's font. It could work with a custom theme. 

> I think it would be a good idea to allow the user set the styles of multiple objects in a "batch" operation.
Currently it seems that the only way is to set the style of each part of each object explicitly, which is tedious and not very scalable.

Besides "batch applying" picking specific objects is also useful. (As CSS selectors do.) This way we could target "virtual parts" a well, e.g. a button in the button matrix or a cell in a table.

> By the way, if it was possible to apply a style to a component sub-tree, it would make sense to convert "bidi" into a style instead of a special object property.
I'm actually looking for a way to apply font style the same way Bidi is applied - to an entire component sub-tree.

Interesting idea. Do you know a CSS selector counterpart for it?

puzrin, 2020-09-28:
> Interesting idea. Do you know a CSS selector counterpart for it?

[direction](https://developer.mozilla.org/en-US/docs/Web/CSS/direction)?

puzrin, 2020-09-28:
> I think it would be a good idea to allow the user set the styles of multiple objects in a "batch" operation.
> Currently it seems that the only way is to set the style of each part of each object explicitly, which is tedious and not very scalable.

See [BEM](https://en.bem.info/methodology/css/).

1. Pre-assign modifier class to all desired objects
2. Update that class as you need.

embeddedt, 2020-09-28:
BEM is consistent but also painful to use for simple projects. It'd be better for objects to inherit their font from their parent, like HTML.

puzrin, 2020-09-28:
> BEM is consistent but also verbose, and painful to use for simple projects (in my opinion).

IMO that's view without big practice and without care about maintenance. For me, it's not difficult to use BEM on small projects too.

> It'd be better for objects to inherit their font from their parent, like HTML.

This is not related to BEM anyhow. There is simplified algorythm of classes merge. If nested object has class with property, that property will have priority over parent.

kisvegabor, 2020-09-28:
> See BEM.

I've read critics about BEM point out that BEM doesn't take advantage of "cascading". 

> It'd be better for objects to inherit their font from their parent, like HTML.

Fonts are inherited now as well. Or do you mean to design themes to not specify fonts for widgets, but let inheriting them from the screen?

embeddedt, 2020-09-28:
Fonts should inherit from the screen as well, IMO.

puzrin, 2020-09-28:
> I've read critics about BEM point out that BEM doesn't take advantage of "cascading".

That's universal methodology, matching well with lvgl's restriced styles. But anyone who knows something better can suggest it :)

amirgon, 2020-09-28:
> Ah, I see. The issue that happened when the list was wider than the label. I also fixed the arrow-text order.

@kisvegabor Looks good now! Thanks.

> It's because in the default themes the font is already set on almost every style so they overwrite the screen's font. It could work with a custom theme.

Why do the default themes set fonts to every component instead of setting the font only on scr and let objects inherit from it?  
If it did it that way, that would allow me to use the default theme but still override the font on scr and affect all descendant objects.  
Is that right?

> Interesting idea. Do you know a CSS selector counterpart for it?

How about Descendant Selectors? See https://github.com/lvgl/lvgl/issues/1551#issuecomment-637799157

kisvegabor, 2020-09-30:
> Looks good now! Thanks.

Great! :slightly_smiling_face: 

> Why do the default themes set fonts to every component instead of setting the font only on scr and let objects inherit from it?
If it did it that way, that would allow me to use the default theme but still override the font on scr and affect all descendant objects.
Is that right?

It wasn't a concern so far but it really sounds better and it seems doable too.

> How about Descendant Selectors? See #1551 (comment)

Just a minor thing to the linked comment. `>` select only direct children, ` ` selects descendants on any level.  

Let's open an issue for it and finally come to conclusion with this topic. Can you do this? If so can you also mention some use cases?

kisvegabor, 2020-10-02:
I"ve updated the material theme, Now you can change the font of the screen and it will affect all objects.

amirgon, 2020-10-02:
> I've updated the material theme, Now you can change the font of the screen and it will affect all objects.

Works well! Thanks.

> Let's open an issue for it and finally come to conclusion with this topic. Can you do this? If so can you also mention some use cases?

I'll open a new issue for this.

Closing this issue as all other issues are resolved!

closed 2020-10-02 by amirgon