lvgl/lvgl #1813: Checkbox tick broken

tgillbe, 2020-09-23:
**Describe the bug**

The checkbox tick no longer shows up when the checkbox is checked.

**To Reproduce**

```
lv_checkbox_create(bg, NULL);
```

**Expected behavior**

The checkbox tick to appear when the checkbox is checked.

**Screenshots**

Example is with material theme.

Before:
[image]

After:
[image]

**Additional context**

git bisect indicates the issue appeared in 81884401d236ff5cf47148ce307033e5d0c049ae

kisvegabor, 2020-09-23:
Hi,

I've noticed a realted issue a few hours ago and fixed it master. Does it work with the latest master?

tgillbe, 2020-09-23:
Yes, it's working. I was on v7.6.0, didn't think to try with master!

tgillbe, 2020-09-23:
Actually it's not entirely fixed - if a checkbox is programmatically set to checked then it shows as unchecked until you click it twice (once to uncheck, once to check again).

```
lv_obj_t *chk = lv_checkbox_create(background, NULL);
lv_checkbox_set_checked(chk, true);
```

kisvegabor, 2020-09-23:
Ah, really. 
I've just pushed fix for it too.

embeddedt, 2020-09-23:
@kisvegabor I don't see the fix. Was there a conflict? (I just updated the README to test a GitHub Action, so maybe your commit wasn't pushed.)

kisvegabor, 2020-09-23:
@embeddedt Oh, yes. Now it's there.

tgillbe, 2020-09-25:
It's now showing the correct checked state but the checked animation plays when the page loads. Not an issue for me because my theme doesn't use an animation but it looks bad on the material theme.

Unfortuantley there doesn't seem to be an lv_example with the checkbox checked by default, but lvgl/docs#154 would have been great for reporting this!

kisvegabor, 2020-09-28:
I've fixed it as well.

tgillbe, 2020-09-30:
All working now, thanks!

closed 2020-09-30 by tgillbe

kisvegabor, 2020-09-30:
Cool, thanks for the testing and feedback!